zephyrproject-rtos/zephyr #19545: usb: obtain configuration descriptor's bmAttributes and bMaxPower from DT

jfischer-phytec-iot, 2019-10-02:
Currently configuration descriptor's bmAttributes and bMaxPower is hardcoded to SelfPowered and MaxPower of 100mA. Where SelfPowered applies to most boards in the tree except boards like pca10059 for example. 

Both SelfPowerd and MaxPower should be derived from DT because they are hardware properties.

StefJar, 2019-10-21:
any news on that feature implementation?

StefJar, 2019-11-04:
@galak any news regarding that feature?

solsbarry, 2019-12-17:
I also would like this feature. I tired creating a second usb descriptor which is easy enough. But whether the Host even asks for anything but the first descriptor is host dependent. On my host, the first descriptor is used if it can be, so this is not a solution.
All you need to do I think, is make a new kconfig option called `CONFIG_USB_MAX_POWER `in `zephyr/subsys/usb/Kconfig`
The defualt could be 0x32
Then in usb_descriptor.c 
do `.bMaxPower = CONFIG_USB_MAX_POWER; `

and something similar for bmAttributes

solsbarry, 2019-12-17:
I provided a fix like I described above. I realize the original request asked for a way to configure this at runtime, and my fix is only at build time. But I am not sure there is a practical reason to reconfigure at runtime. But maybe there is.

jfischer-phytec-iot, 2019-12-17:
Where is this "original request" you talking about?

solsbarry, 2019-12-17:
@jfischer-phytec-iot I meant your request.

jfischer-phytec-iot, 2019-12-17:
This issue suggests to use device tree not Kconfig. But with #20375, Kconfig could probably be a good solution too.

StefJar, 2019-12-18:
I put a request for that some time ago
https://github.com/zephyrproject-rtos/zephyr/issues/19544

jfischer-phytec-iot, 2019-12-23:
resolved by #21461

closed 2019-12-23 by jfischer-phytec-iot