zephyrproject-rtos/zephyr #19549: kernel/mem_protection/stackprot fails on NXP RT series platforms on v1.14.1-rc3 release

hakehuang, 2019-10-02:
**Describe the bug**
kernel/mem_protection/stackprot fails on NXP all platform on v1.14.1-rc3 release

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1050_evk ..
3. make run

**Expected behavior**
test pass

**Impact**
low

**Screenshots or console output**
```
***** Booting Zephyr OS 1.14.1-rc3 *****

Running test suite stackprot

===================================================================

```
> starting test - Je榲V?AⅩ欳蹩?驾洗An賚? 觯$?薔:楷蹘O^晼え嶠cR萲啕$0?mY躌p衫p?i?齛
> 
> ***** USAGE FAULT *****
> 
>   Attempt to execute undefined instruction
> 
> ***** Hardware exception *****
> 
> Current thread ID = 0x8000022c
> 
> Faulting instruction address = 0xa9b0
> 
> Fatal fault in thread 0x8000022c! Aborting.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, ,)
 - Toolchain (Zephyr SDK)
 - v1.14.1-rc3

nashif, 2019-10-02:
cant reproduce: 

```
***** Booting Zephyr OS v1.14.1-rc3 *****
Running test suite stackprot
===================================================================
starting test - test_create_alt_thread
Starts alternate_thread
alternate_thread: Input string is too long and stack overflowed!

***** Stack Check Fail! *****
Current thread ID = 0x800000a8
Faulting instruction address = 0x6000b314
Fatal fault in thread 0x800000a8! Aborting.
PASS - test_create_alt_thread
===================================================================
starting test - test_stackprot
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
PASS - test_stackprot
===================================================================
Test suite stackprot succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

```

dleach02, 2019-10-02:
@hakehuang, I got the same result @nashif did. Tested on IMXRT1050-EVKB.

hakehuang, 2019-10-03:
@dleach02 , @nashif , loosk like my board falsh need erased for this case, sorry for this false positive.

closed 2019-10-03 by hakehuang