zephyrproject-rtos/zephyr #19552: [TOPIC-GPIO]: Support for legacy interrupt configuration breaks new API contract

pabigot, 2019-10-02:
In the original API development in #16648 and as merged to the topic branch `gpio_pin_configure()` was changed to not affect any interrupt configuration except `GPIO_INT_DEBOUNCE`.

In #19248 this decision was reverted, and `gpio_pin_configure()` was changed to include a call to `gpio_pin_interrupt_configure()`.  https://github.com/zephyrproject-rtos/zephyr/pull/19248#discussion_r326809961 questioned this and was answered with the point that this is necessary to support legacy API that configures both pin and interrupt in the same call.  It was described as to-be-removed-later, but there are no details/plans for how to accomplish that.

This breaks the promised new API: it is no longer possible to do things like reconfigure pull state without affecting the interrupt configuration, because presence of `GPIO_INT_DISABLED` cannot be distinguished from a set of flags that does not touch the interrupt configuration, so pin configuration while interrupts are enabled will disable (or reconfigure) the interrupt.

A potential solution is to change `GPIO_INT_DISABLED` to be a non-zero value, and so support a new `GPIO_INT_MODE_UNCHANGED` mode that has no effect.  This would not work on legacy code that intentionally disables interrupts by not providing any interrupt configuration flags in the call to `gpio_pin_configure()`.

galak, 2019-10-02:
So doing a survey of the drivers from code inspection, it looks like the behavior with regards to `GPIO_INT` not being set varies, here's what I could tell [for drivers that support interrupts]:

cc13xx_cc26xx - disable interrupts
cc2650 - disable interrupts
cc32xx - leaves interrupts alone
cmsdk - leaves interrupts alone
dw - leaves interrupts alone
esp32 - leaves interrupts alone
gecko - disable interrupts
imx - disables interrupts
intel-apl - disables interrupts (i think)
mchp-xec - disables interrupts
mcux - disables interrupts
mcux-igpio - disables interrupts
nrfx - disables interrupts
rv32m1 - disables interrupts
sam0 - disables interrupts (I think)
sam - disables interrupts
sifive - leaves interrupts alone [ uses enable/disable cb to enable/disable interrupts ]
stellaris - leaves interrupts alone
stm32 - leaves interrupts alone (I think)

galak, 2019-10-31:
@pabigot can we close this with the merge of #19553?

PR #19553 (2019-10-25): [TOPIC-GPIO] proposed workaround to distinguish interrupt disable from default no-change
Fixes #19552.
closed 2019-10-31 by pabigot