zephyrproject-rtos/zephyr #19560: Console on CDC USB  crashes when CONFIG_USB_COMPOSITE_DEVICE=y

lanvis, 2019-10-02:
**Describe the bug**
target: nrf52840
Our application uses a USB CDC serial port for proprietary communications and
we also need our console port to appear on the same physical port.
When  CONFIG_USB_COMPOSITE_DEVICE=y and CONFIG_USB_UART_CONSOLE=y
 and CONFIG_UART_CONSOLE_ON_DEV_NAME="CDC_ACM_2" or "CDC_ACM_1", the project builds but execution crashes before reaching main.

**To Reproduce**
Steps to reproduce the behavior:
CMake with:
CONFIG_USB_CDC_ACM=y

##Multiple USB CDC devices
CONFIG_USB_COMPOSITE_DEVICE=y
CONFIG_USB_CDC_ACM_DEVICE_COUNT=2
CONFIG_USB_CDC_ACM_RINGBUF_SIZE=512

CONFIG_USB_UART_CONSOLE=y
CONFIG_UART_CONSOLE_ON_DEV_NAME="CDC_ACM_2"
CONFIG_UART_CONSOLE=y

make and run.

**Expected behavior**
I would expect that LOG messages appear appear on virtual com port (usb cdc port 2) 

**Impact**
Our HW has only 1 USB port externally.  We will not be able to view console output.

**Screenshots or console output**
No console output :(

**Environment (please complete the following information):**
 - Linux
 - Zephyr SDC
 - 1.14.99

jfischer-phytec-iot, 2019-10-03:
okay, I do not know what `1.14.99` is referring to, but it works with the current master and the patch for configuration overlay below.

```diff
diff --git a/samples/subsys/usb/cdc_acm/overlay-composite-cdc-console.conf b/samples/subsys/usb/cdc_acm/overlay-composite-cdc-console.conf
new file mode 100644
index 0000000000..3d6cea51b9
--- /dev/null
+++ b/samples/subsys/usb/cdc_acm/overlay-composite-cdc-console.conf
@@ -0,0 +1,12 @@
+CONFIG_SERIAL=y
+
+CONFIG_USB_COMPOSITE_DEVICE=y
+CONFIG_USB_CDC_ACM_DEVICE_COUNT=2
+CONFIG_USB_CDC_ACM_RINGBUF_SIZE=512
+
+CONFIG_UART_INTERRUPT_DRIVEN=y
+CONFIG_UART_LINE_CTRL=y
+CONFIG_UART_CONSOLE_ON_DEV_NAME="CDC_ACM_1"
+
+CONFIG_USB_UART_CONSOLE=y
+CONFIG_USB_UART_DTR_WAIT=y
```

lanvis, 2019-10-03:
I did not see an overlay file for console in the master (samples/subsys/usb/cdc_acm) head or the 2.0.0 tag so I tried both using your overlay (provided above).  The sample/subsys/usb/2.0.0/cdc_acm does not build and fails to find nrfx.h.
This is with -DBOARD=nrf52840_pca10056

/home/lanvis/zeph20/zephyr-zephyr-v2.0.0/soc/arm/nordic_nrf/nrf52/soc.h:17:10: fatal error: nrf.h: No such file or directory
 #include <nrf.h>
          ^~~~~~~
compilation terminated.

So not sure what you used but please tell.

jfischer-phytec-iot, 2019-10-03:
@lanvis Please try `west update`

emob-nordic, 2019-10-04:
> CONFIG_UART_CONSOLE_ON_DEV_NAME 

For this config option uart device should be selected. Please refer to ./drivers/console/Kconfig l:15

emob-nordic, 2019-10-07:
@lanvis is it working now?

carlescufi, 2019-10-09:
@lanvis could you close the issue if this is now resolved?

lanvis, 2019-10-09:
Did not have a chance to try yet ... but will soon.

lanvis, 2019-10-30:
I pulled the master but the overlay-composite-cdc-console.conf is not in the samples/subsys/usb/cdc_acm directory so I made it from the contents you show above.  I changed no source so the example only opens the first port.  I would expect the console to open the 2nd port but I see no 2nd port being opened.  

prj.conf
CONFIG_STDOUT_CONSOLE=y
CONFIG_GPIO=y
CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_PRODUCT="Zephyr CDC ACM sample"
CONFIG_LOG=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_CDC_ACM=y
CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_UART_LINE_CTRL=y

CONFIG_SERIAL=y

CONFIG_USB_COMPOSITE_DEVICE=y
CONFIG_USB_CDC_ACM_DEVICE_COUNT=2
CONFIG_USB_CDC_ACM_RINGBUF_SIZE=512

CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_UART_LINE_CTRL=y
CONFIG_UART_CONSOLE_ON_DEV_NAME="CDC_ACM_1"

CONFIG_USB_UART_CONSOLE=y
#CONFIG_USB_UART_DTR_WAIT=y

lanvis, 2019-10-31:
OK .. found that it waits for DTR so need to comment out  the DTR_WAIT config and all is good.
If you don't do this, you need to have the serial port connection (e.g. term emulator) on the assigned port before execution continues.  All is good.
Thanks ... am closing.

closed 2019-10-31 by lanvis