zephyrproject-rtos/zephyr #19569: nRF RTC Counter with compile time decision about support of custom top value

nordic-krch, 2019-10-03:
**Is your enhancement proposal related to a problem? Please describe.**
Counter based on nRF RTC currently support custom top value. In order to do that one CC channel is used for wrapping which reduced number of available CC channels. #15510 changes that by adding flexible support of custom top value. If top value equals RTC max counter value then no channel is needed since RTC overflows and overflow interrupt is used for top value callback. If user changes top value to custom one then one of channels is used. However, this may lead to confusion because `counter_get_num_of_channels` will return wrong value if custom top value is used. Additionally, supporting custom top value adds more code because any arithmetical operations on ticks are more complex (performance penalty and code size).

**Describe the solution you'd like**
Adding `fixed_top` flag to static configuration of the instance (similar to `ppi_wrap`) would turn off custom top value support in the instance. Additionally, if no instance is using custom top value then code gets removed. It is quite likely that no instance will use custom top value.

pabigot, 2019-10-15:
Notes to myself for figuring out how to do this:

* See additional background at https://github.com/zephyrproject-rtos/zephyr/pull/15510#discussion_r330899406
* `CONFIG_COUNTER_RTC_CUSTOM_TOP_SUPPORT` currently enables this by default when `fixed-top` is selected in any RTC node.  This property is already defined in the devicetree yaml, and is not present in any non-test trees.
* `counter_nrfx_rtc` appears to already implement a filter that discards top-related code when support is not enabled.

@nordic-krch What is supposed to be done for this issue?

nordic-krch, 2019-10-17:
acually it can be closed as it got implemented within #15510.

closed 2019-10-17 by nordic-krch