zephyrproject-rtos/zephyr #19582: zephyr_library: missing kernel-mode vs app-mode documentation

marc-hb, 2019-10-03:
Not sure where this should be documented, probably as a comment in `cmake/extensions/cmake`?

From: `tests/subsys/settings/functional: make common code a zephyr_library()` #19531, see also references there.

> From: "Bøe, Sebastian" <Sebastian.Boe@nordicsemi.no>
> Date: Thursday, 3 October 2019 at 08:12
> To: @marc-hb , @mbolivar , @nvlsianpu 
> Subject: Re: [zephyrproject-rtos/zephyr] tests/subsys/settings/functional: make common code a zephyr_library() (#19531)
> 
> The build system runs in two "modes", or phases, kernel-mode,
> and app-mode. Scripts executed directly or in-directly by boilerplate.cmake
> is in kernel-mode, everything else is in app-mode.
> 
> It's not supported to create zephyr libraries in app-mode as we are doing here,
> but this is not documented anywhere AFAIK so therefore us needing
> to do workarounds like this is a "lack-of-documentation bug" and not a "design bug".
> 
> We use the API in this unsupported way in this case because it happens to do roughly
> (with the exception of needing to manually add_dependencies) what we need to do
> with minimal build script boilerplate.
> 
> The supported way of doing nearly the same thing would be for the app build script to
> add a zephyr_extra_module because west modules are run in kernel-mode and can use
> the zephyr_library API in a supported way. But I decided not to do it in this case because
> creating a zephyr module for this seemed like overkill.

marc-hb, 2019-10-03:
What I find concerning here is the **randomness** of build failures.

Quoting myself from: `tests/subsys/settings/functional: make common code a zephyr_library()` #19531

```
zephyr_library_named(  settings_func_test)
add_dependencies(      settings_func_test offsets_h)
```

`make -C build [settings_func_test]` always fails without `add_dependencies(      settings_func_test offsets_h)`
`make -j` sometimes succeeds in building even without this dependency thanks to a race condition. `make -k; make` obviously succeeds too.

`ninja -j1` doesn't fail, it seems "lucky" enough to build in some order that avoids the missing dependency issue. However `ninja settings_func_test` fails with the same error than `make settings_func_test` above.

nashif, 2021-01-30:
if you think this is not stale, then please do something about it, this is now the 3rd time being reopened.

marc-hb, 2021-01-31:
@tejlmand , @mbolivar-nordic , is Sebastian's email in the description above still valid today and is it still missing from https://docs.zephyrproject.org/latest/guides/zephyr_cmake_package.html ?

mbolivar-nordic, 2021-02-01:
I assume it is still valid. I think https://docs.zephyrproject.org/latest/application/index.html might be a better place to document this, but I defer to @tejlmand .

tejlmand, 2021-02-03:
This is still valid, but instead of just documenting this, I think it's better to issue a warning if the user tries to use `zephyr_library()` after `find_package(Zephyr)` completes, and then in future we can remove the possibility.

galak, 2021-06-01:
@tejlmand can you look at this for the 2.6 release

tejlmand, 2021-06-02:
> @tejlmand can you look at this for the 2.6 release

Done, fixed here: https://github.com/zephyrproject-rtos/zephyr/pull/35887

PR #35887 (2021-06-03): cmake: warn the user if a Zephyr library is created in app-mode
Fixes: #19582

When `find_package(Zephyr)` completes then all boilerplate code has been
processed and all Zephyr libraries has been placed inside the
whole archive flags.

Also all libs dependencies has been processed.

This is indicated by the presence of the zephyr_prebuilt target.

Thus, warn the user if `zephyr_library()` / `zephyr_library_named()`
is called in app-mode.
commit 86c92869 (2021-06-03): cmake: warn the user if a Zephyr library is created in app-mode

Fixes: #19582

When `find_package(Zephyr)` completes then all boilerplate code has been
processed and all Zephyr libraries has been placed inside the
whole archive flags.

Also all libs dependencies has been processed.

This is indicated by the presence of the zephyr_prebuilt target.

Thus, warn the user if `zephyr_library()` / `zephyr_library_named()`
is called in app-mode.
PR #42439 (2022-02-11): test: settings: functional: fix missing test_main
The functional tests for settings subsystem regressed with the
introduction of the weak test_main function. This was caused by
Issue #19582 since the common code was being linked as a library.

Update the logic of the common code to directly add the sources
to the `app` target. This resolved both Issue #19582 and #42396

Fixes #42723
Fixes #19582
closed 2021-06-03 by galak