zephyrproject-rtos/zephyr #19592: Request new repository to host the Eclipse plugin for building Zephyr applications

dcpleung, 2019-10-03:
The new repo will contain source code of an Eclipse plugin which extends Eclipse CDT to support application development on Zephyr (including building and target hardware debugging).

Currently:
* The plugin can create a new Hello World project. An empty project can also be created for use with existing code.
* The plugin has rudimentary code indexing support via the CDT indexer, allowing code references.
* The plugin builds applications by invoking cmake and make/ninja within Eclipse IDE.
* Running with emulator and flashing image to hardware can be done via "Run As" menu in Eclipse.
* Debugging can be started with "Debug As" menu in Eclipse. This starts the debug server and invokes GDB (in GDB/MI mode). Eclipse's debug perspective is used for debugging the application.
* On supported hardware, threads can be displayed in debug perspective after enabling OpenOCD support in application configuration.

nashif, 2019-10-14:
repo created and initial code pushed https://github.com/zephyrproject-rtos/eclipse-plugin

closed 2019-10-14 by nashif

zakariae87, 2019-10-23:
There is something wrong in tutorial, can't add pluging via "help" -> "install New Software". 
No URL available :(

dcpleung, 2019-10-23:
> There is something wrong in tutorial, can't add pluging via "help" -> "install New Software".
> No URL available :(

We are currently enabling CI, and find a home for it. We are going to make an announcement once these are done.