zephyrproject-rtos/zephyr #19629: tinycbor buffer overflow causing mcumgr image upload failure

jimparis, 2019-10-04:
Hi,
Reporting this here because [zephyrproject-rtos/tinycbor](https://github.com/zephyrproject-rtos/tinycbor) doesn't have an issue tracker.

There is a buffer overflow in zephyr's tinycbor, in [cborparser.c](https://github.com/zephyrproject-rtos/tinycbor/blob/31ae89e4b768612722620cb6cb173a0de4a19cc9/src/cborparser.c#L1307):
```c
CborError _cbor_value_copy_string(const CborValue *value, void *buffer,
                                 size_t *buflen, CborValue *next)
{
    //...
    CborError err = iterate_string_chunks(value, (char*)buffer, buflen, &copied_all, next,
                                          buffer ? (IterateFunction) value->parser->d->cpy : iterate_noop);
    //...
    if (buffer) {
        *((uint8_t *)buffer + *buflen) = '\0';
    }
```
If `iterate_string_chunks` fills the buffer completely, the byte past the buffer is zeroed.  This is in contrast to the function comment, which says "*If the buffer is large enough*, this function will insert a null byte after the last copied byte".

The bug is *not* present in the upstream [intel/tinycbor](https://github.com/intel/tinycbor/blob/master/src/cborparser.c), where they add termination inside `iterate_string_chunks` instead.

The bug was introduced by https://github.com/zephyrproject-rtos/tinycbor/pull/1, which appears intended to bring the code more in-line with the upstream mynewt tinycbor, but [their code](https://github.com/apache/mynewt-core/blob/060df6f9dfe3c22f8679e62a8515e8f73d830980/encoding/tinycbor/src/cborparser.c#L985) doesn't have the bug either.

---

The specific symptom I'm seeing is:
```
$ mcumgr --conntype serial --connstring dev=/dev/foo image upload foo.bin
Error: 3
```
and it's happening because `data_sha` overflows into `img_mgmt_data` in [`img_mgmt_upload`](https://github.com/zephyrproject-rtos/mcumgr/blob/84934959d2d1722a23b7e7e200191ae4a6f96168/cmd/img_mgmt/src/img_mgmt.c#L341), during a `cbor_read_object` call:
```c
static int
img_mgmt_upload(struct mgmt_ctxt *ctxt)
{
    uint8_t img_mgmt_data[IMG_MGMT_UL_CHUNK_SIZE];
    uint8_t data_sha[IMG_MGMT_DATA_SHA_LEN];
```
which clobbers the `IMAGE_MAGIC` checked by `img_mgmt_check_header`.

---

This same tinycbor bug was the root cause of an earlier mcumgr fix in https://github.com/zephyrproject-rtos/zephyr/issues/7924, which ended up in a patch to upstream's mcumgr https://github.com/apache/mynewt-mcumgr/pull/5.  There, it was [noted by @ccollins476ad](https://github.com/apache/mynewt-mcumgr/pull/5#issuecomment-401909191) that it was actually a zephyr tinycbor bug, but the workaround was merged to mynewt's mcumgr anyway (https://github.com/apache/mynewt-mcumgr/commit/c2da8ca1a4f512fc0f9a37916a647e10451bf2fc), and then also into zephyr's fork (https://github.com/zephyrproject-rtos/mcumgr/commit/d9b889ecd2382d924debdd139fd6b8243c040f9c).

I think both of those commits should be reverted in their repos once this is fixed.

dleach02, 2019-10-07:
@nvlsianpu, can you provide some guidance on priority level for this problem?

nvlsianpu, 2019-10-08:
@jimparis  https://github.com/zephyrproject-rtos/tinycbor/pull/7 Is this full fix for the issue?

jimparis, 2019-10-08:
As far as I'm aware yes, for tinycbor.  Then an update to zephyr's west.yml here to point to the new version.  Then the workarounds in mcumgr could be removed, but they are harmless.

de-nordic, 2019-10-16:
@carlescufi is already updating west.yml manifest here: https://github.com/zephyrproject-rtos/zephyr/pull/19838

PR #19838 (2019-10-16): manifest: Update the revision of tinycbor
Point to the current revision at the tip of the tinycbor repo after
merging https://github.com/zephyrproject-rtos/tinycbor/pull/7.

Fixes #19629.
commit b595087b (2019-10-16): manifest: Update the revision of tinycbor

Point to the current revision at the tip of the tinycbor repo after
merging https://github.com/zephyrproject-rtos/tinycbor/pull/7.

Fixes #19629.
closed 2019-10-16 by carlescufi