zephyrproject-rtos/zephyr #19643: samples/boards/arc_secure_services fails on nsim_sem

andrewboie, 2019-10-07:
```
***** Booting Zephyr OS build zephyr-v2.0.0-1024-g30fc3e9f8e2a *****
I am the main thread in secure world: 0
Go to normal application
***** Booting Zephyr OS build zephyr-v2.0.0-1024-g30fc3e9f8e2a *****
I am the main thread in secure world: 0
Go to normal application
***** Booting Zephyr OS build zephyr-v2.0.0-1024-g30fc3e9f8e2a *****
I am the main thread in secure world: 0
Go to normal application
***** Booting Zephyr OS build zephyr-v2.0.0-1024-g30fc3e9f8e2a *****
I am the main thread in secure world: 0
Go to normal application
***** Booting Zephyr OS build zephyr-v2.0.0-1024-g30fc3e9f8e2a *****
I am the main thread in secure world: 0
Go to normal application
***** Booting Zephyr OS build zephyr-v2.0.0-1024-g30fc3e9f8e2a *****
I am the main thread in secure world: 0
Go to normal application
***** Booting Zephyr OS build zephyr-v2.0.0-1024-g30fc3e9f8e2a *****
I am the main thread in secure world: 0
Go to normal application
***** Booting Zephyr OS build zephyr-v2.0.0-1024-g30fc3e9f8e2a *****
I am the main thread in secure world: 0
Go to normal application
***** Booting Zephyr OS build zephyr-v2.0.0-1024-g30fc3e9f8e2a *****
I am the main thread in secure world: 0
Go to normal application
***** Booting Zephyr OS build zephyr-v2.0.0-1024-g30fc3e9f8e2a *****
I am the main thread in secure world: 0
Go to normal application
***** Booting Zephyr OS build zephyr-v2.0.0-1024-g30fc3e9f8e2a *****
I am the main thread in secure world: 0
Go to normal application
***** Booting Zephyr OS build zephyr-v2.0.0-1024-g30fc3e9f8e2a *****
I am the main thread in secure world: 0
Go to normal application
***** Booting Zephyr OS build zephyr-v2.0.0-1024-g30fc3e9f8e2a *****
I am thzephyr/CMakeFiles/run.dir/build.make:57: recipe for target 'zephyr/CMakeFiles/run' failed
CMakeFiles/Makefile2:478: recipe for target 'zephyr/CMakeFiles/run.dir/all' failed
CMakeFiles/Makefile2:485: recipe for target 'zephyr/CMakeFiles/run.dir/rule' failed
Makefile:235: recipe for target 'run' failed
```

vonhust, 2019-10-15:
In fact, it's not a bug. This example cannot run alone, it requires a normal application. It's similar to ARM TF-M applicaiton or multicore AMP application, you need to build at leaset two zephyr imags. For details, pls go to README.rst of this example.

PR #13672 is trying to add multi-image build support. When it's ready,  I will modify this example to make it easier to use.

andrewboie, 2019-11-22:
> In fact, it's not a bug. This example cannot run alone, it requires a normal application. It's similar to ARM TF-M applicaiton or multicore AMP application, you need to build at leaset two zephyr imags. For details, pls go to README.rst of this example.

@vonhust this is **breaking** sanitycheck runs.
If you don't want this treated as a bug, get it out of the sanitycheck builds.

nashif, 2019-11-27:
filtering issue

PR #22766 (2020-02-13): samples: arc_secure_services: fix sample.yaml
This test can't be evaluated with sanitycheck, it
requires special set-up on multiple AMP cores to
function properly. Add build_only tag.

Fixes: #19643
Fixes: #22317
commit c6d37eba (2020-02-13): samples: arc_secure_services: fix sample.yaml

This test can't be evaluated with sanitycheck, it
requires special set-up on multiple AMP cores to
function properly. Add build_only tag.

Fixes: #19643
Fixes: #22317
closed 2020-02-13 by jhedberg