zephyrproject-rtos/zephyr #19649: [TOPIC-GPIO]: Replace GPIO_INT_DEBOUNCE with GPIO_DEBOUNCE

pabigot, 2019-10-07:
I propose that `GPIO_INT_DEBOUNCE` be deprecated and replaced by `GPIO_DEBOUNCE`.

This flag is documented as applying in `gpio_pin_configure()` and is the only `GPIO_INT_foo` that is intended to be used that way.  On SAM the use of debounce is not specific to interrupts: it also applies to the value read from `PIO_PDSR`.  (@vanti and @dcpleung please check whether this is consistent with use in cc* and dw GPIO hardware, the only other two groups that support this flag).

Functionally debounce seems closer to pull configuration than anything associated specifically with interrupts.  (At least on SAM it also appears to require out-of-driver configuration of other registers, so it's not necessarily useful.)

Finally the documentation for this flag should specify that it is ignored, not rejected, if passed to a driver that does not support hardware debounce.

pabigot, 2019-10-07:
From @dcpleung in https://github.com/zephyrproject-rtos/zephyr/pull/19631#discussion_r332138737:
> I don't think it's a good practice to ignore the debounce flag if hardware does not support it. Developers would expect this to be working since there is no error. They will be very confused when multiple interrupts are triggered in a very very short period of time for one, say, button down press.

The problem is that many sensor samples are passing `GPIO_INT_DEBOUNCE` when configuring data-ready interrupts on peripheral instances.  So if it's presence is diagnosed as `-ENOTSUP` on hardware where any debouncing is done by circuitry outside the GPIO peripheral then those samples won't work.

vanti, 2019-10-08:
I am fine with this. `GPIO_INT_DEBOUNCE` enables the hysteresis feature on the TI CC1352, which has more to do with the levels at which 0 and 1 are identified, and is not restricted to the notion of interrupts per se.

lawrencek52, 2019-10-08:
In my mind "Debouncing" is not the same as changing the pull up strength, or changing the level change thresholds, or adding addition hysteresis. If you want to change pull-up strength, then use the pull-up strength modifiers when configuring the GPIO (or interrupt input). If you want to change the hysteresis of level thresholds, then use an appropriate modifier. But debouncing shouldn't change levels, hysteresis, or pull-up strength.

Debouncing is really associated with the mechanical bounce of a snap action switch. When you press a snap action switch the moving contact hits the stationary contact and bounces up, then the spring forces the moving contact back down, eventually (possibly after several bounces) closing the contacts. This will generate several interrupts as the signal goes high-low-high-low over a extended period of time (up to a millisecond).

Debouncing should ignore all of the spurious interrupts and should have a time parameter for how long to ignore spurious interrupts.

There is and electronic feature on some GPIOs (an example is Qualcomm Snapdragon, which doesn't really apply to Zephyr) called "keeper", in which the GPIO will change from pull-up to pull-down on a transition. This would be a true "debounce" since the first impact or the contacts would change the pull-up to a pull down and then when the contacts were in the floating state (airborn) the input would stay low. Keeper does have the unfortunate side effect that you need a SPDT switch to use it because releasing the switch doesn't make the input go high, you have to actively drive the input high to get it back to the initial condition.

Just my two cents concerning "Debounce"....

-Lawrence-

pabigot, 2019-10-08:
API call decision: Do not change name, do document that drivers that don't support debounce should ignore the flag rather than reject it.

Future work will rework debounce to be compatible with Linux devicetree representations where time constants are provided.

pabigot, 2019-10-15:
Documentation updated in #19695.

closed 2019-10-15 by pabigot