zephyrproject-rtos/zephyr #19670: samples/drivers/spi_fujitsu_fram crashs due to uninitialized variables

karlp, 2019-10-08:
**Describe the bug**
A clear and concise description of what the bug is.
In https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/drivers/spi_fujitsu_fram/src/main.c#L153, the spi_cfg struct has not been properly initialized, and has whatever was on the stack for the struct.  _particularly_ this means taht the .cs property may contain garbage.

The following code in the driver gets triggered:

```
        data->ctx.config = config;
 8005a90:       9b0a            ldr     r3, [sp, #40]   ; 0x28
 8005a92:       601f            str     r7, [r3, #0]
        if (ctx->config->cs && ctx->config->cs->gpio_dev) {
 8005a94:       2900            cmp     r1, #0
 8005a96:       f000 8162       beq.w   8005d5e <spi_stm32_transceive+0x49a>

```
I get a bus fault on the attempt to deref ctx->config->cs->gpio_dev, as cs is garbage.

either explicitly setting .cs = 0, or intializing the struct with = {0} works to avoid the bus fault.

**Expected behavior**
working code on all platforms with spi.  Presumably will work on platforms or not, depending on how cs is being handled out of the box for each board.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
tedious inability to get viable spi demostrations, so... *shrugs* big for me.

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: linux
 - Toolchain gcc-arm-embedded
 - Commit SHA 178f16f9e9  (2.0 master)

dleach02, 2019-10-08:
@karlp the code snippet indicates an STM board. Can you confirm the board?

karlp, 2019-10-08:
sure, I was using my own board, (stm32l1 based) but it's a straight up uninitialized struct problem, the board is not exactly relevant.  from gcc notes, because I can't find the c99 notes as easily...

```
If you do not initialize a structure variable, the effect depends on whether it has static storage (see Storage Class Specifiers) or not. If it is, members with integral types are initialized with 0 and pointer members are initialized to NULL; otherwise, the value of the structure’s members is indeterminate.
```
https://www.gnu.org/software/gnu-c-manual/gnu-c-manual.html#Initializing-Structure-Members

It's not static, and it's not initalized at all, so it's contents are indeterminate.

dleach02, 2019-10-09:
Okay. Apologies for not looking closer at the ticket description. I see what you are saying about the data structure and you are correct that it will take any values that happen to be on the stack. I'm assuming in your test environment you have tried they assignment to zero? I also wonder if it is better to use memset() instead of the {0}? 

Can you supply a PR with this change?

karlp, 2019-10-09:
I didnt' send a PR because I don't know what sort of style on this the project wanted to have.  init the struct with {0} works, and is what I did.  Personally I feel that's better than explicitly memsetting, just leave it to the compiler, but I've no idea what's consistent with the project.

jfischer-phytec-iot, 2019-10-09:
This sample seems to be outdated and IMHO unnecessary, neither context->spi_cfg.slave nor context->spi_cs.gpio_dev will be initialized. Also a app.overlay should be added. I suggest to remove it. ping @tbursztyka

jfischer-phytec-iot, 2019-10-09:
@dleach02 changed to low, it is just a outdated sample.

karlp, 2019-10-09:
it might be "low" but it's the _only_ sample of the SPI API, so I think either a) not low, or b) provide some SPI examples.  (covering things like options for CS would be nice too)

jfischer-phytec-iot, 2019-10-09:
a)  drivers/ethernet/eth_enc424j600.c
b) drivers/ieee802154/ieee802154_mcr20a.c
c) drivers/can/can_mcp2515.c
d) drivers/display/display_ili9340.c
e) drivers/display/ssd16xx.c
f) drivers/sensor/lis2mdl/lis2mdl.c
g) use something like `git grep "struct spi_config" drivers`
h) https://docs.zephyrproject.org/latest/reference/peripherals/spi.html?highlight=spi
i) https://docs.zephyrproject.org/latest/guides/dts/index.html

karlp, 2019-10-09:
I'm just recounting my experience as a new user here.  I could find the sensor one, but I don't want to know how the sensor api worked, I wanted to know how the spi api worked.  Even more relevant for samples of displays and ethernet.  They add far too much overhead to what you're trying to learn.

Discoverabilty of them is ~zero anyway.  You have a nice web list of samples, but none of them mention spi, so  you're back to grepping code.  the "obvious" place to look was under samples/drivers/spi* (like samples/drivers/i2c*) as that's the _driver_ I was interested in using.

Way to go linking to the docs though, totally useful.  There's still nothing there about the CS modes.  you can provide a gpio yes, but nothing about gpios vs hardware chip selects for instance.

jfischer-phytec-iot, 2019-10-09:
You wanted to know how the spi api worked? Look at a)..f) all of them use spi. You could also take a look at `tests/drivers/spi`, but a)..f) show typical use case.

erwango, 2019-11-21:
ping @tbursztyka .
Any idea what to do with this sample? @jfischer-phytec-iot propose to remove it purely.
Assigning it to @dleach02 since there is not much I can do more.

dleach02, 2020-07-09:
@tbursztyka, @jfischer-phytec-iot, Do we want to remove this sample? I'm trying to help clean up low priority bugs for our next release.

dleach02, 2020-07-09:
Note that this sample relies on the spi-1 filter and that filter has virtually been removed (NXP boards have it but will likely get removed). I suspect that @jfischer-phytec-iot recommendation to remove the sample would be the best course unless someone wants to give this sample a work-over on a board that exists.

jfischer-phytec-iot, 2020-07-10:
People are still trying to use it, see  #26642. I think the mood was that the sampe should stay. cc @nashif

dleach02, 2020-07-21:
okay, I posted a fix  but I don't know how to test this.

PR #27034 (2020-07-22): samples: drivers: Fix unitialized spi_cfg in spi_fujitsu_fram sample
Fixes #19670
commit 1b192df0 (2020-07-22): samples: drivers: Fix unitialized spi_cfg in spi_fujitsu_fram sample

Fixes #19670
closed 2020-07-22 by ioannisg