zephyrproject-rtos/zephyr #19678: slowness networking stack - MXRT1050-EVK

alemoppo, 2019-10-08:
I'm trying the "samples/net/sockets/dumb_http_server", i just developed a little webserver based on this example.
The request always takes about 350ms, you can see it from the browser console (F12 -> network).
I think the problem is not related to the board or network because my bare-metal version is much faster (the HTTP respose is like 1ms).

I think the problem is in the "accept" call, but I don't know if is the context switch delay from the "idle" thread or other.

Can anyone reproduce the problem? I'm using the MIMXRT1050-EVK.

jukkar, 2019-10-08:
I created a PR that can be used to collect network timing statistics for both TX and RX paths. See #19690 for details. Can you try the PR and report what kind of numbers you see in that hw?
For example for sam-e70 board, I got these kind of results

Zephyr
```
uart:~$ net stats
...
Avg TX net_pkt (1330) time 33417 us
Avg RX net_pkt (220) time 221 us
```

The Apache Bench running in Linux host gave this
```
Server Software:        
Server Hostname:        2001:db8::1
Server Port:            8080

Document Path:          /
Document Length:        2122 bytes

Concurrency Level:      50
Time taken for tests:   4.123 seconds
Complete requests:      100
Failed requests:        0
Total transferred:      218100 bytes
HTML transferred:       212200 bytes
Requests per second:    24.25 [#/sec] (mean)
Time per request:       2061.597 [ms] (mean)
Time per request:       41.232 [ms] (mean, across all concurrent requests)
Transfer rate:          51.66 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        0   42 327.7      0    3112
Processing:   201  202   0.4    201     204
Waiting:        1    1   0.3      1       3
Total:        202  244 327.9    202    3314
ERROR: The median and mean for the processing time are more than twice the standard
       deviation apart. These results are NOT reliable.

Percentage of the requests served within a certain time (ms)
  50%    202
  66%    202
  75%    202
  80%    202
  90%    202
  95%    205
  98%   1265
  99%   3314
 100%   3314 (longest request)
```

I have also created a multi-threaded dummy http server sample, I will send it soon to for review.

jukkar, 2019-10-09:
I have been investigating this a bit with a proper hw (sam_e70_xplained) and qemu_x86.
Some of my results are described below:

Here I am running echo-server in sam-e70 board and connecting to it using echo-client running in Linux host.
Linux
```
$ ./echo-client -e -t 2001:db8::1
Average round trip 409 us
```

Zephyr
```
Interface 0x2040e920 (Ethernet) [1]
===================================
IPv6 recv      17696    sent    33134   drop    9       forwarded       0
IPv6 ND recv   3        sent    10      drop    0
IPv6 MLD recv  0        sent    3       drop    0
IPv4 recv      25       sent    2       drop    23      forwarded       0
IP vhlerr      0        hblener 0       lblener 0
IP fragerr     0        chkerr  0       protoer 0
ICMP recv      7        sent    7       drop    0
ICMP typeer    0        chkerr  0
UDP recv       0        sent    0       drop    27
UDP chkerr     0
TCP bytes recv 3369151  sent    3367919
TCP seg recv   17682    sent    33121   drop    0
TCP seg resent 0        chkerr  0       ackerr  0
TCP seg rsterr 0        rst     0       re-xmit 0
TCP conn drop  0        connrst 0
Bytes received 4683227
Bytes sent     5820281
Processing err 0
Avg TX net_pkt (33137) time 166 us
Avg RX net_pkt (17662) time 186 us
```
The round-trip time reported by Linux, is quite near the sum of TX and RX packet handling time. The result here look quite good.

Then I tried Apache Bench and the mt-http-server found in #19691, and got these results.

Linux
```
ab -n100 -c50 http://[2001:db8::1]:8080/

This is ApacheBench, Version 2.3 <$Revision: 1843412 $>
Copyright 1996 Adam Twiss, Zeus Technology Ltd, http://www.zeustech.net/
Licensed to The Apache Software Foundation, http://www.apache.org/

Benchmarking 2001:db8::1 (be patient).....done

Server Software:        
Server Hostname:        2001:db8::1
Server Port:            8080

Document Path:          /
Document Length:        2084 bytes

Concurrency Level:      50
Time taken for tests:   4.095 seconds
Complete requests:      100
Failed requests:        0
Total transferred:      214000 bytes
HTML transferred:       208400 bytes
Requests per second:    24.42 [#/sec] (mean)
Time per request:       2047.357 [ms] (mean)
Time per request:       40.947 [ms] (mean, across all concurrent requests)
Transfer rate:          51.04 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        0   42 324.1      0    3083
[... 122 lines cut ...]
Interface 0x140ce0 (Ethernet) [1]
===================================
IPv6 recv      1461	sent	2453	drop	174	forwarded	0
IPv6 ND recv   9	sent	13	drop	4
IPv6 MLD recv  0	sent	3	drop	0
IPv4 recv      1	sent	0	drop	1	forwarded	0
IP vhlerr      0	hblener	0	lblener	0
IP fragerr     0	chkerr	0	protoer	0
ICMP recv      12	sent	4	drop	2
ICMP typeer    0	chkerr	0
UDP recv       0	sent	0	drop	1
UDP chkerr     0
TCP bytes recv 243903	sent	243311
TCP seg recv   1281	sent	2439	drop	168
TCP seg resent 0	chkerr	0	ackerr	0
TCP seg rsterr 0	rst	1	re-xmit	1
TCP conn drop  0	connrst	0
Bytes received 448899
Bytes sent     425047
Processing err 0
Avg TX net_pkt (2453) time 175 us
Avg RX net_pkt (1278) time 183 us
```

It looks like things are much slower with Apache Bench than with echo-server/client, this could be because of how ab is sending data.

alemoppo, 2019-10-09:
Thanks for the tests jukkar.

I didn't find the download of the modified files #19690 , i could modify them manually but i'm afraid to make some mistake.

With your sample (mt_http_server, i just change the ip) i have this result:

`$ ab -n100 http://192.168.0.220:8080/
This is ApacheBench, Version 2.3 <$Revision: 1826891 $>
Copyright 1996 Adam Twiss, Zeus Technology Ltd, http://www.zeustech.net/
Licensed to The Apache Software Foundation, http://www.apache.org/

Benchmarking 192.168.0.220 (be patient).....done

Server Software:        
Server Hostname:        192.168.0.220
Server Port:            8080

Document Path:          /
Document Length:        2122 bytes

Concurrency Level:      1
Time taken for tests:   31.023 seconds
Complete requests:      100
Failed requests:        0
Total transferred:      218100 bytes
HTML transferred:       212200 bytes
Requests per second:    3.22 [#/sec] (mean)
Time per request:       310.227 [ms] (mean)
Time per request:       310.227 [ms] (mean, across all concurrent requests)
Transfer rate:          6.87 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        1    2   1.0      1       7
Processing:   303  308   0.9    308     309
Waiting:      102  108   0.9    108     108
Total:        309  310   0.4    310     312
WARNING: The median and mean for the initial connection time are not within a normal deviation
        These results are probably not that reliable.

Percentage of the requests served within a certain time (ms)
  50%    310
  66%    310
  75%    310
  80%    310
  90%    310
  95%    310
  98%    312
  99%    312
 100%    312 (longest request)`

The hardware could do better: with my bare-metal webserver:

`ab -n100 http://192.168.0.220/var_len.html?len=2048
This is ApacheBench, Version 2.3 <$Revision: 1826891 $>
Copyright 1996 Adam Twiss, Zeus Technology Ltd, http://www.zeustech.net/
Licensed to The Apache Software Foundation, http://www.apache.org/

Benchmarking 192.168.0.220 (be patient).....done

Server Software:        Webserver
Server Hostname:        192.168.0.220
Server Port:            80

Document Path:          /var_len.html?len=2048
Document Length:        1918 bytes

Concurrency Level:      1
Time taken for tests:   1.062 seconds
Complete requests:      100
Failed requests:        0
Total transferred:      204900 bytes
HTML transferred:       191800 bytes
Requests per second:    94.16 [#/sec] (mean)
Time per request:       10.620 [ms] (mean)
Time per request:       10.620 [ms] (mean, across all concurrent requests)
Transfer rate:          188.42 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        1    2   1.6      2      14
Processing:     7    8   1.6      8      21
Waiting:        7    8   1.6      7      21
Total:          9   10   2.3     10      23

Percentage of the requests served within a certain time (ms)
  50%     10
  66%     10
  75%     10
  80%     11
  90%     12
  95%     13
  98%     23
  99%     23
 100%     23 (longest request)
`

I just run it in 2 boards: the mxrt1050-evk and the new mxrt1050-evk**b**. 
Probably is someting at low level code with the 1050 mcu? Should i open a ticket with nxp to make test them?

jukkar, 2019-10-09:
If possible, apply both #19690 and #19691, and re-run mt-http-server against ab again. I would be interested in seeing what are the TX/RX numbers in Zephyr side. The compilation could be done for example like this:
`cmake -Bbuild -DBOARD=mxrt1050-evk . -DCONFIG_NET_PKT_RXTIME_STATS=y -DCONFIG_NET_PKT_TXTIME_STATS=y`
After the ab test run is finished, just do `net stats` in zephyr side. It will print out the avg processing times for TX and RX packets.

I also noticed that you used ab without `-c` option. Basically that means that ab sends one request at a time, that is why I had `-c50` in my tests. That gives a bit better throughput numbers.

alemoppo, 2019-10-09:
I'm sorry but Avg TX and Avg RX didn't appear. Probably i doing someting wrong.

I changed all 9 files and add to "prj.conf": `CONFIG_NET_PKT_RXTIME_STATS=y` and `CONFIG_NET_PKT_TXTIME_STATS=y`.

```
$ ab -n100 -c50 http://192.168.0.220:8080/
This is ApacheBench, Version 2.3 <$Revision: 1826891 $>
Copyright 1996 Adam Twiss, Zeus Technology Ltd, http://www.zeustech.net/
Licensed to The Apache Software Foundation, http://www.apache.org/

Benchmarking 192.168.0.220 (be patient).....done

Server Software:        
Server Hostname:        192.168.0.220
Server Port:            8080

Document Path:          /
Document Length:        2122 bytes

Concurrency Level:      50
Time taken for tests:   48.999 seconds
Complete requests:      100
Failed requests:        3
   (Connect: 0, Receive: 0, Length: 3, Exceptions: 0)
Total transferred:      218100 bytes
HTML transferred:       212200 bytes
Requests per second:    2.04 [#/sec] (mean)
Time per request:       24499.563 [ms] (mean)
Time per request:       489.991 [ms] (mean, across all concurrent requests)
Transfer rate:          4.35 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        0  295 1708.5     13   15393
Processing:   296 1772 7423.1    308   48888
Waiting:        0  166 258.1    108    1818
Total:        309 2068 7610.7    332   48888

Percentage of the requests served within a certain time (ms)
  50%    332
  66%    428
  75%    575
  80%    576
  90%   1993
  95%   9186
  98%  41067
  99%  48888
 100%  48888 (longest request)
```

```
uart:~$ net stats

Interface 0x80013500 (Ethernet) [1]
===================================
IPv6 recv      1        sent    9       drop    1       forwarded       0
IPv6 ND recv   1        sent    5       drop    1
IPv6 MLD recv  0        sent    2       drop    2
IPv4 recv      2288     sent    1426    drop    982     forwarded       0
IP vhlerr      0        hblener 0       lblener 0
IP fragerr     0        chkerr  0       protoer 0
ICMP recv      1        sent    2       drop    2
ICMP typeer    0        chkerr  0
TCP bytes recv 19864    sent    449286
TCP seg recv   1306     sent    1426    drop    889
TCP seg resent 0        chkerr  0       ackerr  0
TCP seg rsterr 23       rst     0       re-xmit 213
TCP conn drop  149      connrst 0
Bytes received 166297
Bytes sent     527010
Processing err 3
```
I'll try again tomorrow

Thanks.

jukkar, 2019-10-10:
> I'm sorry but Avg TX and Avg RX didn't appear. Probably i doing someting wrong.

You need to apply both #19690 and #19691, one easy option is checkout #19690 and then cherry-pick one (and only) commit from #19691. If you do no have both PRs in your test, then the statistics are not calculated / shown in zephyr.

jukkar, 2019-10-10:
Found the issue, this is actually a bug. I will send a fix soon.

jukkar, 2019-10-10:
@alemoppo can you try the proposed fix and report the performance numbers?

alemoppo, 2019-10-10:
I confirm that this change improves speed.

I updated all the zephyr files and applied #19690.
After that, I was no longer able to view the http page with your webserver (mt), but i appled the console in my webserver.

Before the #19747 :

```
Interface 0x801ccea0 (Ethernet) [1]
===================================
IPv4 recv      1274     sent    1074    drop    58      forwarded       0
IP vhlerr      0        hblener 0       lblener 0
IP fragerr     0        chkerr  0       protoer 0
ICMP recv      0        sent    50      drop    0
ICMP typeer    0        chkerr  0
UDP recv       0        sent    0       drop    45
UDP chkerr     0
TCP bytes recv 76323    sent    108198
TCP seg recv   1216     sent    1024    drop    13
TCP seg resent 0        chkerr  0       ackerr  0
TCP seg rsterr 0        rst     0       re-xmit 202
TCP conn drop  0        connrst 5
Bytes received 154863
Bytes sent     167040
Processing err 3
Avg TX net_pkt (1075) time 45321 us
Avg RX net_pkt (202) time 134401 us
```

Before:

```
Interface 0x801ccea0 (Ethernet) [1]
===================================
IPv4 recv      2374     sent    2369    drop    44      forwarded       0
IP vhlerr      0        hblener 0       lblener 0
IP fragerr     0        chkerr  0       protoer 0
ICMP recv      0        sent    39      drop    0
ICMP typeer    0        chkerr  0
UDP recv       0        sent    0       drop    39
UDP chkerr     0
TCP bytes recv 178644   sent    233778
TCP seg recv   2330     sent    2330    drop    5
TCP seg resent 0        chkerr  0       ackerr  0
TCP seg rsterr 0        rst     0       re-xmit 0
TCP conn drop  0        connrst 0
Bytes received 323004
Bytes sent     362230
Processing err 2
Avg TX net_pkt (2370) time 104 us
Avg RX net_pkt (466) time 134439 us
```

I know this is not the best test, I'll test more now. This is just a "fast" feedback :), thanks!

jukkar, 2019-10-10:
> I updated all the zephyr files and applied #19690.
> After that, I was no longer able to view the http page with your webserver (mt), but i appled the console in my webserver.

Please try latest version of #19747 as I fixed couple of issues there, the http page shows ok with the test webserver (mt) with firefox.

alemoppo, 2019-10-11:
Here is the test with your mt webserver:

```
This is ApacheBench, Version 2.3 <$Revision: 1843412 $>
Copyright 1996 Adam Twiss, Zeus Technology Ltd, http://www.zeustech.net/
Licensed to The Apache Software Foundation, http://www.apache.org/

Benchmarking 192.168.0.220 (be patient).....done

Server Software:        
Server Hostname:        192.168.0.220
Server Port:            8080

Document Path:          /
Document Length:        2122 bytes

Concurrency Level:      50
Time taken for tests:   28.826 seconds
Complete requests:      100
Failed requests:        0
Total transferred:      218100 bytes
HTML transferred:       212200 bytes
Requests per second:    3.47 [#/sec] (mean)
Time per request:       14413.121 [ms] (mean)
Time per request:       288.262 [ms] (mean, across all concurrent requests)
Transfer rate:          7.39 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        3  218 414.5    160    3419
Processing:   108  356 961.3    161    7220
Waiting:      107  251 598.6    159    4546
Total:        120  574 1290.6    321   10639

Percentage of the requests served within a certain time (ms)
  50%    321
  66%    321
  75%    321
  80%    321
  90%    322
  95%   2886
  98%   5506
  99%  10639
 100%  10639 (longest request)
```
```
Interface 0x80019160 (Ethernet) [1]
===================================
IPv6 recv      0        sent    9       drop    0       forwarded       0
IPv6 ND recv   0        sent    5       drop    0
IPv6 MLD recv  0        sent    2       drop    2
IPv4 recv      897      sent    612     drop    294     forwarded       0
IP vhlerr      0        hblener 0       lblener 0
IP fragerr     0        chkerr  0       protoer 0
ICMP recv      0        sent    2       drop    2
ICMP typeer    0        chkerr  0
TCP bytes recv 8600     sent    218100
TCP seg recv   603      sent    612     drop    252
TCP seg resent 0        chkerr  0       ackerr  0
TCP seg rsterr 0        rst     0       re-xmit 0
TCP conn drop  0        connrst 0
Bytes received 66080
Bytes sent     251760
Processing err 2
Avg TX net_pkt (620) time 63322 us
Avg RX net_pkt (100) time 222329 us
```

Now a request is done in like 160ms (read in browser's console).

jukkar, 2019-10-11:
I did a quick test with sam-e70

```
ab -n10 http://192.0.2.1:8080/
This is ApacheBench, Version 2.3 <$Revision: 1843412 $>
Copyright 1996 Adam Twiss, Zeus Technology Ltd, http://www.zeustech.net/
Licensed to The Apache Software Foundation, http://www.apache.org/

Benchmarking 192.0.2.1 (be patient).....done

Server Software:        
Server Hostname:        192.0.2.1
Server Port:            8080

Document Path:          /
Document Length:        2084 bytes

Concurrency Level:      1
Time taken for tests:   0.016 seconds
Complete requests:      10
Failed requests:        0
Total transferred:      21400 bytes
HTML transferred:       20840 bytes
Requests per second:    642.10 [#/sec] (mean)
Time per request:       1.557 [ms] (mean)
Time per request:       1.557 [ms] (mean, across all concurrent requests)
Transfer rate:          1341.88 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        0    0   0.1      0       0
Processing:     1    1   0.0      1       1
Waiting:        1    1   0.1      1       1
Total:          1    2   0.1      1       2
ERROR: The median and mean for the total time are more than twice the standard
       deviation apart. These results are NOT reliable.

Percentage of the requests served within a certain time (ms)
  50%      1
  66%      2
  75%      2
  80%      2
  90%      2
  95%      2
  98%      2
  99%      2
 100%      2 (longest request)
```

Then with 100 reqs, note the -r option which was needed in my testing.
```
ab -n100 -r http://192.0.2.1:8080/
This is ApacheBench, Version 2.3 <$Revision: 1843412 $>
Copyright 1996 Adam Twiss, Zeus Technology Ltd, http://www.zeustech.net/
Licensed to The Apache Software Foundation, http://www.apache.org/

Benchmarking 192.0.2.1 (be patient).....done

Server Software:        
Server Hostname:        192.0.2.1
Server Port:            8080

Document Path:          /
Document Length:        2084 bytes

Concurrency Level:      1
Time taken for tests:   5.423 seconds
Complete requests:      100
Failed requests:        15
   (Connect: 0, Receive: 5, Length: 5, Exceptions: 5)
Total transferred:      203300 bytes
HTML transferred:       197980 bytes
Requests per second:    18.44 [#/sec] (mean)
Time per request:       54.226 [ms] (mean)
Time per request:       54.226 [ms] (mean, across all concurrent requests)
Transfer rate:          36.61 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        0   53 230.8      0    1060
Processing:     0    1   0.2      1       1
Waiting:        0    1   0.2      1       1
Total:          1   54 230.8      2    1061

Percentage of the requests served within a certain time (ms)
  50%      2
  66%      2
  75%      2
  80%      2
  90%      2
  95%   1037
  98%   1060
  99%   1061
 100%   1061 (longest request)
```
It is a bit mystery why some requests fail.

I also tried with curl
```
$ curl -so /dev/null -w '%{time_total}\n' http://192.0.2.1:8080
0,001493
```
and the results are typically under 2ms
```
while [ true ]; do curl -so /dev/null -w '%{time_total}\n' http://192.0.2.1:8080; done
```

pfalcon, 2019-10-11:
This is pretty old issue, as old as https://github.com/zephyrproject-rtos/zephyr/issues/13943 , and occurrence coincides with it. I definitely reported it to the core team, but can't find a separate ticket, perhaps it was done in a comment to another issue, or on Slack. I've updated title of this ticket to hopefully better describe the situation.

Before I proceed to test https://github.com/zephyrproject-rtos/zephyr/pull/19747 , I'd like to record my results:

qemu_x86+e1000, `ab -n100` (I usually use -n1000, and recommend doing that, as lower number of requests is not conclusive. But qemu_x86+e1000 combo is not stable enough for -n1000):

~~~
$ ab -n100 http://192.0.2.1:8080/
[]
Server Hostname:        192.0.2.1
Server Port:            8080

Document Path:          /
Document Length:        2122 bytes

Concurrency Level:      1
Time taken for tests:   51.542 seconds
Complete requests:      100
Failed requests:        0
Total transferred:      218100 bytes
HTML transferred:       212200 bytes
Requests per second:    1.94 [#/sec] (mean)
Time per request:       515.420 [ms] (mean)
Time per request:       515.420 [ms] (mean, across all concurrent requests)
Transfer rate:          4.13 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        0    0   0.4      0       4
Processing:   209  515 253.5    410    1168
Waiting:        1  311 252.9    208     959
Total:        210  515 253.6    410    1172

Percentage of the requests served within a certain time (ms)
  50%    410
  66%    410
  75%    410
  80%    840
  90%   1020
  95%   1030
  98%   1030
  99%   1172
 100%   1172 (longest request)
~~~

Quite blurred picture actually, issues with e1000 (#13943) definitely come into play.

qemu_x86+SLIP (with -n1000):

~~~
$ ab -n1000 http://192.0.2.1:8080/
[]
Server Hostname:        192.0.2.1
Server Port:            8080

Document Path:          /
Document Length:        2122 bytes

Concurrency Level:      1
Time taken for tests:   224.462 seconds
Complete requests:      1000
Failed requests:        0
Total transferred:      2181000 bytes
HTML transferred:       2122000 bytes
Requests per second:    4.46 [#/sec] (mean)
Time per request:       224.462 [ms] (mean)
Time per request:       224.462 [ms] (mean, across all concurrent requests)
Transfer rate:          9.49 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        2    4   1.2      3      26
Processing:   214  221   6.0    217     278
Waiting:        5   10   2.6     10      48
Total:        219  224   6.5    220     290

Percentage of the requests served within a certain time (ms)
  50%    220
  66%    230
  75%    230
  80%    230
  90%    230
  95%    230
  98%    240
  99%    240
 100%    290 (longest request)
~~~

This is the familiar picture for since the end of last year (but not before, when we had 20ms).

pfalcon, 2019-10-11:
So, with https://github.com/zephyrproject-rtos/zephyr/pull/19747 and qemu_x86+SLIP:

~~~
$ ab -n1000 http://192.0.2.1:8080/
[]
Server Hostname:        192.0.2.1
Server Port:            8080

Document Path:          /
Document Length:        2122 bytes

Concurrency Level:      1
Time taken for tests:   155.758 seconds
Complete requests:      1000
Failed requests:        0
Total transferred:      2181000 bytes
HTML transferred:       2122000 bytes
Requests per second:    6.42 [#/sec] (mean)
Time per request:       155.758 [ms] (mean)
Time per request:       155.758 [ms] (mean, across all concurrent requests)
Transfer rate:          13.67 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        2  137 341.8      4    1036
Processing:    11   18   6.1     17      58
Waiting:        6   12   4.5     10      48
Total:         15  156 341.4     22    1069

Percentage of the requests served within a certain time (ms)
  50%     22
  66%     25
  75%     29
  80%     33
  90%   1026
  95%   1035
  98%   1047
  99%   1051
 100%   1069 (longest request)
~~~

So, it gets to classical figures for 80% of connections, but then either a retransmission, or unknown delay kicks in. This proves that https://github.com/zephyrproject-rtos/zephyr/pull/19747 (in its current, 2nd by now version) is just a workaround, not a solution. And IMHO, it makes the situation only worse: whereas before we at least had a way of reproducing a deterministic behavior with uniform delay distribution, that patch adds high delay dispersion, and throws *everything* into the situation when we don't know the exact cause(s) of such dispersion.

jukkar, 2019-10-11:
> This proves that #19747 (in its current, 2nd by now version) is just a workaround, not a solution.

Those numbers you are showing here prove nothing of that kind, the fix is real and working. It seems we hit some limit and incoming connection is denied (verified it from wireshark), after this ab just prints the statistics weirdly. I will try to figure out what is the root cause for those resets.

pfalcon, 2019-10-11:
> It seems we hit some limit and incoming connection is denied (verified it from wireshark), after this ab just prints the statistics weirdly.

In this case, `ab` should print an error. It's otherwise as standard, decade-proven load testing tool, and I've been using it against Zephyr servers for quite a while (well, dumb_http_server was developed specifically for it, to avoid using adhoc tools for such testing, as then results look questionable/artificial).

 > I will try to figure out what is the root cause for those resets.

Thanks, looking forward to it.

pfalcon, 2019-10-11:
> So, it gets to classical figures for 80% of connections

To prove that these classical figures existed: I took https://github.com/zephyrproject-rtos/zephyr/commit/338dc8a9520ea7b3c60503e35daba085073f4a1d, as suggested by @jukkar in https://github.com/zephyrproject-rtos/zephyr/pull/19747#issuecomment-540646621 and run it now. That's not so easy, as that version no longer works with SDK 0.10.x, so you have to go back to SDK 0.9.5 (I naively killed it already and had to re-download). But otherwise, the situation is:

~~~
Time taken for tests:   22.476 seconds
Complete requests:      1000
Failed requests:        0
Total transferred:      2181000 bytes
HTML transferred:       2122000 bytes
Requests per second:    44.49 [#/sec] (mean)
Time per request:       22.476 [ms] (mean)
Time per request:       22.476 [ms] (mean, across all concurrent requests)
Transfer rate:          94.76 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        2    6  32.6      4    1031
Processing:    11   17   4.5     16     102
Waiting:        7   11   3.4     10      73
Total:         14   22  33.2     20    1056

Percentage of the requests served within a certain time (ms)
  50%     20
  66%     21
  75%     23
  80%     23
  90%     27
  95%     30
  98%     35
  99%     42
 100%   1056 (longest request)
~~~

As you can see, there're still can be O(1) cases of retransmissions or something, but otherwise it was within range of 20ms - 2*20ms.

pfalcon, 2019-10-11:
Ok, so I'm bisecting this to see if it can help to understand the delay cause. As we know, 338dc8a9520ea7b3c60503e35daba085073f4a1d works w/o delay. The last commit of that "streak" which works is 3215dd862f93c68a0a96ad56e42ae4edb80b5253.

In the next commit, f484bbaa264f3a9640ca52477e3038f912ea5642, I myself broke the build of dumb_http_server (`fdtable.c:80:21: error: ‘FD_OBJ_RESERVED’ undeclared (first use in this fu)
   fdtable[fd].obj = FD_OBJ_RESERVED;`).

This error happens until (and including) 9aafe388280f5920b9400cc4c3b62fc548a2eeb6. In the next commit, 65a33bd0c07a29e71630f910d6304f2a0468f775, the build is fixed, but runtime of the example is broken (`Error in accept: 23 - continuing`). Last commit with that error is aa5bbd5dfe9930d26555d0db95df0291386a4129.

Then it recovered in 2e21a95fd470e9f244e7532dae9b69f181f9ceca and serves w/o delay again! It works until and including df07e093ce5e7696705f1408e315432fb9275a58. Next commit, 94e89a11b7151c9f8dd8c22007e64bd30ee8b809, is beginning of @tbursztyka's last refactor (it was split after a number of PRs after. e.g. https://github.com/zephyrproject-rtos/zephyr/pull/10547 is one of them). The 94e89a11b7151c9f8dd8c22007e64bd30ee8b809 still works for dumb_http_server. Then with next commit, dcc6eddfefe51eacd2ecd60d1c2596b1ceea338a, packets don't go thru the stack. dump_http_server again works w/o delay at 10c4841835249bfc61d47f18f1908e77fca77377, the end of this streak of @tbursztyka's changes!

(to be continued)

pfalcon, 2019-10-11:
Braindump in case of need for further reference, feel free to skip.

~~~
last good 8a1481735b5a09a6b1024184f7ad7f076ab3e786

First bad - 76b3518ce6c0da6c4f3a5f688ea4e5beb827ac2c 

a26fe1e2f4d7fbdee6d93216e831e866c00ba62b - bad, deadlocks on <100 requests

513cceb5d1c23b0e062d77823cc1cbd0ec782036 - bug was fixed, works again
4f911e192fc24dc79268b8b78dfeff36c5403dd7
c885cb533e70e289d1fafe13eb1fb27bd90044cb
12995b95c25200a14e12be802de46b088fa6223d

next tbursztyka refactor:
8eb9ff72f69c94e12e114184ea0fc2d305abf370
999b3740b8f14822b82ecc61156442782aad51de
1b6a500067e58ac0e5768335924a78caf09ddb95
f9315f80970fe22d74f9bbe25f18c5f23908c4c4
0d519f7bcf76e2953f668cd28fb7639a0ac01bfb - last working

083470a14a52234f494ad2a302497d309657ac79 - Error sending data to peer
6ea5a12f4cb448c7b480d4879bfa8fd37a98d5f3 -"-
40ad4b7e3e0827a48b763ec9cee9e94207e37505 -"-
8d21c6871885e51c4134484e0b863a1f3ba41d3a -"-
6a5f12b7f819f72f42e7bf0c220c987c23a10ee5 -"-
55e02d2b99ccc8038b5804a6f31911a35530147c -"-
28c01b6219b4c7d6012d9c7c71471a738f86acb3
2168d809871695b42443fa06a1d230413cc237d5
a46aadeff25d65852941f99ab710aa2c647cf791

16edbbad6387bdec50f1c1cf7aaa6568ebea5c33 - recovered
0bf94333448e54dc2ef0073b52e9320883f80dec - ok, no delays
024a7e0502aebfa5676bf97fc6fa01c06eaedf2e

8ceedafdc496927d3e805f991891f988bb5e33bf
34fe5545c91dbd9e0bc4d48e02f751acf68a6383
76553032b9c72ca0be18f2176d232080e185818b
d33641f39a85ca8fa06f915952ea9b3e28875aa1
dad4b297f30e655f619ecc085b899d7abfc988ae
bf855c621bb5d2facddb53f7c06e2a0a93276823
46a02322ec638ec4831257c80e075d2a548125d4

12c98eceb1a728f01edc1eebde88785de9fba941 - ns16550.yaml parse issue
6fabbb7d148044bdba0323669434bbeaa7272f84 - ok
fbb2bcaf87b71fefb1a45cb8133fdb26b56122ae

Slow:

18680e55e507984e34c63d26d5bfdd006675d8e2
d3c89fea4f7807c583808a6d0fb2c4922dc74cfa
80fd80a8c988ab3162859cba91fd7d8404ed49f1
d9876be30c21130565792ccd1a6b215561cd73e0
48753144bde1feafc88d5e9158f4d454137238d2
26085d9570e94f2793506f49caafbce4801af0b2
3c0c92ab3f598b3c66edac6ce945ee71294171ae
4278bf5673480effdca3b1e048cecb7a9ab32976 - yaml fix
078be776eecb581255e3e5c552245365e52a5bf4
8658e3fe79472d45548f407b22f63ef2f2183899
f38c4c97ce7b5e7cdfa93739b494bcdb8d44d969
193d6c60df7410a15c7400b07e5b9310647ce2d7
a0851c47aa876b98190b33106d6ca28fb384523c
71b4137d30b790963cc47d6d3361ff13995fd172
6a764c18793fbb314e688c73bb5b3c9f9e6ccfd9
7a0f9215173fc079b8db6825db53f88dd99eb520
950b5c1a7abecadd6cb7381bd754df3bad19bc73
f16e92c000dcaa1350c0f8a5126ec02dd708e371
1efaae59493e191e95bbc68a47e1487b4239f89d
b33dd7ebde837b137db34b31f0281b54a47ce580
9e69c540605517e602170a37cfecb475c129b2f7
22c9646b9702b6e505eb3d09afd2ee88563f1734 - slow
cebe11544ebeaed9a0c9ffd722c1639fa7463cbe tag: zephyr-v1.14.0 - slow
3fa90938fe2c7d1f3c786e8416e84c6f9a99c321 - slow!
~~~

pfalcon, 2019-10-11:
Ok, so this is the worst bisect I've done so far. Partly because I started too wide of course. Had to trace thru multiple local minimums (dumb_http_server for obviously or very subtly broken, but then recovered), thru toolchain and depend libs upgrades, etc.

Anyway, half-day later, fbb2bcaf87b71fefb1a45cb8133fdb26b56122ae is the last commit which serves ~50req/s. 45df09c010d29684a0f2fe7e2b4751c9cea25b48 is the first commit with 200ms delay, dropping to ~4.5req/s.

I wanted to say that I don't recognize that commit, but it's https://github.com/zephyrproject-rtos/zephyr/pull/13545, and here's, my approval. Reading thru comments though, my usual concerns of whether that was tested enough...

I myself definitely didn't test it, and later it wasn't easy to connect performance regressions with these seemingly simple changes.

Ping @dleach02 for you to be in loop of the situation.

pfalcon, 2019-10-12:
> Anyway, half-day later, fbb2bca is the last commit which serves ~50req/s. 45df09c is the first commit with 200ms delay, dropping to ~4.5req/s.

Btw, the tree at these commits won't build as is with the current Zephyr build infra, to make them build, need to pull a fix-up (4278bf5673480effdca3b1e048cecb7a9ab32976):

~~~
git cherry-pick 4278bf5673480effdca3b1e048cecb7a9ab32976
~~~

jukkar, 2019-10-14:
I managed to figure out why we see the RST packets from zephyr. Before the TCP connection is fully closed, we have a time wait delay which by default is 250ms. The net_context is kept referenced that long after connection is closed, before it the connection is fully teared down. If there are lot of incoming connections, then it is better to make this value much shorter. In the tests before, I set CONFIG_NET_TCP_TIME_WAIT_DELAY to 0 (to disable it), other alternative is to increase the value of CONFIG_NET_MAX_CONTEXTS to much higher value. I would recommend setting the CONFIG_NET_TCP_TIME_WAIT_DELAY=0 as waiting after connection is closed has limited use.

After these changes I get very consistent results from ab, below is a run using 2000 requests:
```
$ ab -n2000 http://192.0.2.1:8080/
This is ApacheBench, Version 2.3 <$Revision: 1843412 $>
Copyright 1996 Adam Twiss, Zeus Technology Ltd, http://www.zeustech.net/
Licensed to The Apache Software Foundation, http://www.apache.org/

Benchmarking 192.0.2.1 (be patient)
Completed 200 requests
Completed 400 requests
Completed 600 requests
Completed 800 requests
Completed 1000 requests
Completed 1200 requests
Completed 1400 requests
Completed 1600 requests
Completed 1800 requests
Completed 2000 requests
Finished 2000 requests

Server Software:        
Server Hostname:        192.0.2.1
Server Port:            8080

Document Path:          /
Document Length:        2084 bytes

Concurrency Level:      1
Time taken for tests:   3.161 seconds
Complete requests:      2000
Failed requests:        0
Total transferred:      4280000 bytes
HTML transferred:       4168000 bytes
Requests per second:    632.70 [#/sec] (mean)
Time per request:       1.581 [ms] (mean)
Time per request:       1.581 [ms] (mean, across all concurrent requests)
Transfer rate:          1322.25 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        0    0   0.0      0       1
Processing:     1    1   0.0      1       1
Waiting:        1    1   0.0      1       1
Total:          1    2   0.0      2       2

Percentage of the requests served within a certain time (ms)
  50%      2
  66%      2
  75%      2
  80%      2
  90%      2
  95%      2
  98%      2
  99%      2
 100%      2 (longest request)
```

alemoppo, 2019-10-14:
Hello, that value seems to me already set to 0 in [prj.conf:24](https://github.com/zephyrproject-rtos/zephyr/blob/d7dbb572e3e745c846544d466b74874061d0cf5a/samples/net/sockets/dumb_http_server/prj.conf#L24)
I noticed this problem with just that project.
I also noticed that in version 2.0.99 I can no longer make HTTPS connections after the first one; however in version 2.0.0 it works correctly (HTTPS), even if there is a problem with the delay.

Can be a low level problem with the i.mxrt1050?

jukkar, 2019-10-14:
> Hello, that value seems to me already set to 0 in [prj.conf:24](https://github.com/zephyrproject-rtos/zephyr/blob/d7dbb572e3e745c846544d466b74874061d0cf5a/samples/net/sockets/dumb_http_server/prj.conf#L24)
> I noticed this problem with just that project.

Interesting. I have done all my testing with #19691 and everything seems to work just fine after I set `CONFIG_NET_TCP_TIME_WAIT_DELAY=0`

> Can be a low level problem with the i.mxrt1050?

That is certainly possible. When I was testing with qemu_x86, I noticed that first packet of the HTTP request was lost (Zephyr did not receive it), and Linux host needed to re-send it which caused around 200ms delay. After I noticed it, I have done all my testing runs with sam_e70_xplained board which has worked well.

dleach02, 2019-10-14:
@jukkar, you have seen the problem on qemu_x86?

@agansari, can you look at this from the rt1050?

jukkar, 2019-10-15:
> @jukkar, you have seen the problem on qemu_x86?

I do not know what problem you are referring to.

This PR is getting convoluted with different people using different hw and software in the testing. I decided to get some real numbers with different hw, config options and explain why numbers look like how they do.

Baseline issue in master
-------------------

Platform: **sam-e70-xplained**
Sample: **dump_http_server**
Git: *master*
Results: The request takes 200ms. The reason is that the last FIN message is queued for fixed duration of 200ms instead of sending it asap. This is fixed by https://github.com/zephyrproject-rtos/zephyr/pull/19747
```
$ ab -n30 http://192.0.2.1:8080/
Server Hostname:        192.0.2.1
Server Port:            8080
Document Path:          /
Document Length:        2122 bytes
Concurrency Level:      1
Time taken for tests:   6.137 seconds
Complete requests:      30
Failed requests:        0
Total transferred:      65430 bytes
HTML transferred:       63660 bytes
Requests per second:    4.89 [#/sec] (mean)
Time per request:       204.581 [ms] (mean)
Time per request:       204.581 [ms] (mean, across all concurrent requests)
Transfer rate:          10.41 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        0    0   0.1      0       1
Processing:   204  204   0.1    204     204
Waiting:        4    4   0.1      4       4
Total:        204  204   0.1    205     205
ERROR: The median and mean for the total time are more than twice the standard
       deviation apart. These results are NOT reliable.

Percentage of the requests served within a certain time (ms)
  50%    205
  66%    205
  75%    205
  80%    205
  90%    205
  95%    205
  98%    205
  99%    205
 100%    205 (longest request)
```

Fix Proposal
----------------

Platform: **sam-e70-xplained**
Sample: **dump_http_server**
Git: *master + fix #19747*
Results: The request take about 1000ms. Because we are no longer waiting 200ms, ab gets results quicker and can send new request earlier. The dumb_http_server is a single threaded server which will wait for incoming connections, serve them and then wait another connection. At the moment we do not have any queing for received SYN packets so if the application has not called accept(), the incoming SYN is dropped. This is happening here and I verified it from debug logs. The sample app is a bit too slow to accept another connection attempt and this forces the Linux to re-send a new SYN packet after around 1sec timeout. This is not happening always, sometimes dumb_http_server manages to accept the connection in time, thus the ab reported time is <1sec.
```
$ ab -n10 http://192.0.2.1:8080/
Server Hostname:        192.0.2.1
Server Port:            8080
Document Path:          /
Document Length:        2122 bytes
Concurrency Level:      1
Time taken for tests:   9.230 seconds
Complete requests:      10
Failed requests:        0
Total transferred:      21810 bytes
HTML transferred:       21220 bytes
[... 160 lines cut ...]
Requests per second:    2.54 [#/sec] (mean)
Time per request:       393.401 [ms] (mean)
Time per request:       393.401 [ms] (mean, across all concurrent requests)
Transfer rate:          5.31 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:      390  391   0.8    391     392
Processing:     2    2   0.0      2       2
Waiting:        2    2   0.0      2       2
Total:        392  393   0.8    394     394
WARNING: The median and mean for the total time are not within a normal deviation
        These results are probably not that reliable.

Percentage of the requests served within a certain time (ms)
  50%    394
  66%    394
  75%    394
  80%    394
  90%    394
  95%    394
  98%    394
  99%    394
 100%    394 (longest request)
```

pfalcon, 2019-10-16:
@jukkar: Thanks for the detailed comments here and in #19747. I'm tracing my way thru to make sure I can confirm your findings, and to clarify some points.

You wrote: 

> Because we are no longer waiting 200ms, ab gets results quicker and can send new request earlier.

So, can you elaborate where exactly 200ms delay comes from? In previous comment you wrote:

> and Linux host needed to re-send it which caused around 200ms delay.

So, that's the fine point we should keep in mind (and which I found earlier, keep forgetting, then rediscovering) - that any delay can be reactive on Zephyr side, i.e. caused by peer. But then it's caused by some abnormality in TCP flow. And such abnormalities are what's being the problem on Zephyr side. Except that such abnormalities may be caused not just by the IP stack, but also by "hardware" (which also includes QEMU networking, which is known to have some reliability issues).

pfalcon, 2019-10-16:
https://unix.stackexchange.com/a/210381/29893 talks that the minimum (starting) RTO value for Linux is ~200ms, and how to change it. However, changing it too 100ms, or 300ms, running against master 711c4d8a488808e6bddd71d9194ece6b2aaabca1, mps2_an385+overlay-smsc911x.conf (that's still QEMU target; getting away from SLIP due to points above that it has link stability issues; and then mps2_an385+overlay-smsc911x.conf is the only not obviously broken QEMU ethernet target, per https://github.com/zephyrproject-rtos/zephyr/issues/13943) - I still get ~220ms from `ab`, so that doesn't see to be, or there's no complete understanding of what happens there.

jukkar, 2019-10-16:
Many of the issues in TCP flow are caused because we do not queue SYN packets. That is why if app is not accepting connection, we many drop the initial SYN which causes a delay before a new SYN is sent by the peer. This is perfectly fine behaviour, but of course it looks bad in statistics.

commit db9d66a6 (2019-11-07): net: tcp: When closing the connection send FIN without extra delays

The earlier code was always queuing the FIN that is sent when
connection is closed. This caused long delay (200 ms) before the peer at
the other end noticed that the connection was actually closed.
Now check if there is nothing in the queue, then send the FIN
immediately. If there is some data in the queue, flush it when a valid
ack has been received.

Fixes #19678