zephyrproject-rtos/zephyr #19684: [message_queues.rst] unclear about data_item structure type

sdalu, 2019-10-08:
In documentation: reference/kernel/data_passing/message_queues.html
• `sizeof(data_item_type)` is used instead of `sizeof(struct data_item_type)`
• struct is defined as `data_item_type` and `next data_item_t`
• could be useful to have a note (as in the api) that `struct data_item_type` should have a size multiple of the message queue alignment to keep the alignment. Perhaps propose to use the `aligned` attribute when defining type: `typedef struct { }  __attribute__((aligned(4))) data_item_t` (edited)

aescolar, 2019-10-08:
CC @andrewboie @andyross

jenmwms, 2020-09-16:
> could be useful when talking about defining message queue, to talk again about the message queue alignment and propose to use the `aligned` attribute when defining type: `typedef struct { }  __attribute__((aligned(4))) data_item_t`

I need some help understanding this, please. Where do you suggest we discuss this in relation to the implementation example of [Defining a Message Queue](https://docs.zephyrproject.org/latest/reference/kernel/data_passing/message_queues.html#defining-a-message-queue)?

jenmwms, 2020-09-16:
Perhaps I could add something like this at the end of that section?  
`
typedef struct {
    uint32_t field1;
    uint32_t field2;
    uint32_t field3;
}__attribute__((aligned(4))) data_item_type;
`

jenmwms, 2020-09-16:
I think I understood the suggestions correctly. Please check the PR - https://github.com/zephyrproject-rtos/zephyr/pull/28423

PR #28423 (2020-09-17): docs: ref: kernel: clarify data_item structure in message queues
- use sizeof(struct data_item_type) instead of sizeof(data_item_type)
- use struct data_item_type instead of invalid reference data_item_t
- add aligned typedef struct data_item_type example implemenation
- add note about checking return value for error message

Fixes #19684
commit 65bcba61 (2020-09-17): docs: ref: kernel: clarify data_item structure in message queues

- use sizeof(struct data_item_type) instead of sizeof(data_item_type)
- use struct data_item_type instead of invalid reference data_item_t
- add aligned typedef struct data_item_type example implemenation
- add note about checking return value for error message

Fixes #19684
closed 2020-09-17 by MaureenHelm