zephyrproject-rtos/zephyr #19700: nrfx_uart RX hang on errors

jimparis, 2019-10-08:
I haven't set up a clear way to reproduce this, but I've seen cases where my nrf52832-based board will stop responding to input on the UART.  I noticed it particularly often while switching between a seral terminal and mcumgr, especially with `mcumgr reset`.  In my past experience with Nordic parts, this happens when there are unacknowledged errors like buffer overruns, parity errors, or BREAKs.

If I open up GDB while it's not receiving characters, I can see that `uart0_addr->EVENT_ERROR` is 1, and clearing it with `p uart0_addr->EVENT_ERROR=0` makes things work again.

Looking at the code (v2.0.0), I don't see any handling of `EVENT_ERROR` in the case where `CONFIG_UART_0_INTERRUPT_DRIVEN` is set.  The serial interrupt handler `uart_nrfx_isr` directly invokes the shell's `uart_callback`, but the UART API provides no way for the shell to handle these errors.
https://github.com/zephyrproject-rtos/zephyr/blob/8f1780f6dee29dc2b2db01744e328a1a4a20d0b8/subsys/shell/shell_uart.c#L118-L143

The async API (`CONFIG_UART_0_ASYNC`) has lots of what appears to be correct error handling in `uart_nrfx_uart.c`, but that option doesn't appear compatible with the shell?

A dumb patch like this appears to work:
```c
diff --git a/drivers/serial/uart_nrfx_uart.c b/drivers/serial/uart_nrfx_uart.c
index ec1f7b1db730..bca2bee49658 100644
--- a/drivers/serial/uart_nrfx_uart.c
+++ b/drivers/serial/uart_nrfx_uart.c
@@ -822,6 +822,9 @@ static void uart_nrfx_isr(void *arg)
 {
        ARG_UNUSED(arg);
 
+       if (nrf_uart_event_check(uart0_addr, NRF_UART_EVENT_ERROR))
+               nrf_uart_event_clear(uart0_addr, NRF_UART_EVENT_ERROR);
+
        if (irq_callback) {
                irq_callback(irq_cb_data);
        }
```

aescolar, 2019-10-09:
CC @carlescufi

Mierunski, 2019-10-15:
Interrupt Driven API does not specify a way of handling errors, however it is possible to erase errors using uart_err_check function. It's not documented in API, but whole interrupt driven API seems to be based on 16550 uart handling where reading error register clears it, it seems that other vendors have implemented it in the same way as well.
Disabling error interrupts should also do the trick and fix hangs on errors.
Adding error handling to driver ISR would make it behave in different way than drivers from other vendors, so it's unfavorable solution.

carlescufi, 2019-10-18:
@jimparis could you take a look at the response above and let us know what you think?

ioannisg, 2019-11-11:
@jimparis ping :)

ioannisg, 2019-11-21:
@carlescufi this is effectively blocking the v2.1 release; i suggest to remove the Medium priority bug label, until we have a response from @jimparis and, also, a better understanding on whether this is actually a bug.

Mierunski, 2019-11-21:
uart_err_check clears error state. It's more of API documentation issue than bug in driver. Also its not only an issue in nRF drivers so platform: nRF label should be removed in my opinion

jimparis, 2019-11-21:
Hi,
I'm just now getting back from paternity leave and it will be a little while before I can get back up to speed on this project.
Regarding the undocumented `uart_err_check`: that calls `uart_nrfx_err_check`, which will retrieve and clear the `ERRORSRC` field, but that won't clear the NRF_UART_EVENT_ERROR event, so I think the problem remains.

Mierunski, 2019-11-22:
Right, my bad, looked too fast at this one. Proposed solution is good, created a PR to fix that before release.

jimparis, 2019-11-22:
PR looks good to me, thanks

PR #20935 (2019-11-26): drivers: nrf: Fix UART and UARTE hanging on RX errors
Add clearing error event to UART and UARTE drivers.
Without it driver goes into infinite interrupt loop.

Fixes #19700 

Solution proposed by @jimparis
closed 2019-11-26 by ioannisg