zephyrproject-rtos/zephyr #19701: mem_pool_threadsafe sporadic failures impacting CI

pabigot, 2019-10-09:
```
sanitycheck -p qemu_x86_long -T tests/kernel/mem_pool/mem_pool_threadsafe
```

is failing intermittently but often enough that it failed both retests in
https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/52828/5/console

It can also be reproduced by hand on Ubuntu 19.04 using the above command.

```
ESCcESC[?7lESC[2JESC[0mSeaBIOS (version rel-1.12.1-0-ga5cab58e9a3f-prebuilt.qemu.org)
Booting from ROM..***** Booting Zephyr OS build zephyr-v2.0.0-1080-g0bb8ec86d776 *****
Running test suite mpool_threadsafe
===================================================================
starting test - test_mpool_threadsafe
ASSERTION FAIL [!sys_dnode_is_linked(&to->node)] @ ZEPHYR_BASE/kernel/timeout.c:86
E: ** CPU Exception 6 (code 0/0x0) **

E: RIP=0000000000107a71 RSP=000000000010baa0 RFLAGS=0000000000000017

E: RAX=0000000000105047 RBX=0000000000106230 RCX=00000000000003f8 RDX=00000000000003f8

E: RSI=000000000000000a RDI=0000000000000000 RBP=0000000000000015 RSP=000000000010baa0

E: R8=00000000ffffffff R9=0000000000000001 R10=0000000000000202 R11=000000000010ba88

E: R12=00000000001026ff R13=0000000000000d20 R14=0000000000105b48 R15=0000000000000008

E: >>> ZEPHYR FATAL ERROR 0: CPU exception
E: Current thread: 0x00106200 (unknown)
E: Halting system
PASS - test_mpool_threadsafe
===================================================================
Test suite mpool_threadsafe succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
~
```

pabigot, 2019-10-09:
Further bisecting indicates this is a result of 3eb1a8b59a641666287503ccd9735fb3a1b981d9 from #19458.

npitre, 2019-10-09:
I just pushed #19715 which apparently makes this issue disappear.
I ran 6500 iterations of the test overnight without a single failure.

I'm not convinced the fix in #19715 is actually fixing the issue here
entirely, as I can't explain how that would make a difference. I may 
simply suppose that some race possibility still exists but that #19715 
makes it much less likely.

gnuless, 2019-10-09:
I've also put up #19719, which simply disables the SMP scheduler for `qemu_x86_long`.

There is no reason why the use of SMP scheduler with only 1 CPU should exhibit different behavior than the legacy scheduler does, but tests fail sporadically when it's enabled.  The arch-specific code for x86 changes only trivially when `SMP` is on, so I suspect these are latent problems in the scheduler.

pabigot, 2019-10-09:
OK, both #19715 and #19719 "fix the test".  I think #19715 isn't the correct fix but it narrows down the problem.

`_MPOOL_MINBLK` is 16 on this platform.  The master version produces this parameter set and `struct sys_mem_pool_base` initialization:

    BLK SIZE MIN 8 MAX 32 ALIGN 8
    max_sz = 32,
    n_max = 2,
    n_levels = 1,
    max_inline_level = 0,

where the test fails.

The effect of #19715 is to change it to this initialization:

    BLK SIZE MIN 16 MAX 64 ALIGN 8
    max_sz = 64,
    n_max = 2,
    n_levels = 2,
    max_inline_level = 1

where the test passes.

If I take #19715 and change it like this:

    -#define BLK_SIZE_MAX (4 * BLK_SIZE_MIN)
    +#define BLK_SIZE_MAX (2 * BLK_SIZE_MIN)

I get:

    BLK SIZE MIN 16 MAX 32 ALIGN 8
    max_sz = 32,
    n_max = 2,
    n_levels = 1,
    max_inline_level = 0

and the failure recurs.

So my hypothesis is that something's wrong with the mem_pool infrastructure when `n_levels` is one.

npitre, 2019-10-09:
On Wed, 9 Oct 2019, Peter A. Bigot wrote:

> If I take #19715 and change it like this:
> 
>     -#define BLK_SIZE_MAX (4 * BLK_SIZE_MIN)
>     +#define BLK_SIZE_MAX (2 * BLK_SIZE_MIN)
> 
> I get:
> 
>     BLK SIZE MIN 16 MAX 32 ALIGN 8
>     max_sz = 32,
>     n_max = 2,
>     n_levels = 1,
>     max_inline_level = 0
> 
> and the failure recurs.

Yes, because at that point you basically reverted things to the previous 
state.

The basic thing to understand here is that the mempool allocator code 
splits memory blocks into 4 sub-blocks for small allocations.

The test is built around this mempool internal behavior to allocate 4 
blocks of BLK_SIZE_MIN out of the initial block of BLK_SIZE_MAX. If this 
factor is not respected, then the allocator cannot split the original 
block into 4 sub-blocks, and then it will only allow for a single block 
of BLK_SIZE_MIN, meaning that in the test the allocation will hit the 
wait queue 4 times more often with a timeout of 200.

> So my hypothesis is that something's wrong with the mem_pool infrastructure when `n_levels` is one.

I disagree. There is in fact much more mempool code solicitation when it 
can split its initial block i.e. when n_levels > 1. When it can't, it is 
the scheduler that gets more solicited instead, exposing a race with a 
timeout instance that gets queued twice.

pabigot, 2019-10-09:
OK.  Then I would ask that you update #19715 to make that relationship more clear in the test where the bounds are given, or in the commit description.  I interpreted the change as being related to increasing the minimum supported block size, rather than the need to ensure that more than one level was being provided.

npitre, 2019-10-09:
On Wed, 9 Oct 2019, Peter A. Bigot wrote:

> OK.  Then I would ask that you update #19715 to make that relationship more clear in the test where the bounds are given, or in the commit description.  I interpreted the change as being related to [...]

Sure. OTOH maybe we should keep it as it is for now so that the actual 
race keeps being investigated?

pabigot, 2019-10-09:
> Sure. OTOH maybe we should keep it as it is for now so that the actual race keeps being investigated?

No, because it causes CI to fail for any PR that happens to cause this combination of platform and test to be invoked.  If this is truly due to SMP failures then somebody (@gnuless?) should identify or create an issue to ensure the problem is fixed rather than hiding it.  That issue should have some link back to this.

BTW: When I said "update #19715" I didn't mean add a comment in the PR, I meant make the justification clear in either the code (preferably) or the commit message that makes the change.  Nobody's going to look at the PR after it's closed.  Somebody might look at the commit message.

npitre, 2019-10-09:
> BTW: When I said "update #19715" I didn't mean add a comment in the PR, I meant make the justification clear in either the code (preferably) or the commit message that makes the change.  Nobody's going to look at the PR after it's closed.  Somebody might look at the commit message.

I will. I just replied to your comment before even seeing your request 
for clarification.

npitre, 2019-10-09:
On Wed, 9 Oct 2019, Peter A. Bigot wrote:

> No, because it causes CI to fail for any PR that happens to cause this combination of platform and test to be invoked.  If this is truly due to SMP failures then somebody (@gnuless?) should identify [...]

If that may help investigating:

The wrapper function k_mem_pool_alloc() calls z_pend_curr_unlocked() 
when no memory is available and a timeout is specified. It looks like 
this is the only location in the whole tree z_pend_curr_unlocked() is 
used, and without #19715 applied, this would get used a lot on 64-bit 
targets during this test.

This is a wrapper around z_pend_curr_irqlock(z_arch_irq_lock(), ...)
that calls pend() --> z_add_thread_timeout() --> z_add_timeout().

Concurrently, we have k_mem_pool_free_id() using a local lock with a 
suspicious comment that says: "Note that this spinlock only exists 
because z_unpend_all() * is unsynchronized.  Maybe we want to put the 
lock into the * wait_q instead and make the API safe?". So, after taking 
this lockwith k_spin_lock(), it calls z_unpend_all() which, 
incidentally, is also the only location in the whole tree where it is 
used. This calls z_unpend_thread() and z_ready_thread().

In z_unpend_thread() we have z_unpend_thread_no_timeout(), followed by 
z_abort_thread_timeout() where the timeout instance is unlinked.

So, is there a way for execution to return from z_pend_curr_unlocked(), 
quickly figure out that there is still no memory available, and loop 
back into z_pend_curr_unlocked() while another CPU is in the middle of 
z_unpend_thread() making the first thread runable before it had a chance 
to unlink that first thread's timeout instance?

I'm not familiar enough with Zephyr SMP implementation to 
say for sure, but the above locking scheme looks somewhat iffy to me.

gnuless, 2019-10-09:
> I'm not familiar enough with Zephyr SMP implementation to say for sure, but the above locking scheme looks somewhat iffy to me.

Probably best to go straight to the source, which I believe in this case would be @andyross ...

pabigot, 2019-10-12:
With the merge of #19715 this test no longer fails.  Nonetheless https://github.com/zephyrproject-rtos/zephyr/pull/19715#issuecomment-540056705 specifies that this issue is not fixed.

Since the test passes the criteria of the issue have been addressed.  Another issue should be opened, including reproduction instructions, for any remaining concerns.

closed 2019-10-12 by pabigot

andrewboie, 2020-07-15:
I think I've isolated this. Appears to have something to do with setting CONFIG_TIMESLICE_SIZE. See details in #13813.