zephyrproject-rtos/zephyr #19713: usb: investigate if Network Buffer can be used in USB device stack and USB drivers

jfischer-phytec-iot, 2019-10-09:
Investigate if net_buf can be used in USB device stack and USB drivers.

Reason: The memory should be used more efficiently  and secure. Unnecessary copying should be avoided. Support for multiple device instances should be considered.

There are many locale buffers in different places. Mostly these are only needed for temporary storage, whereby the content is often copied by the driver to another buffer again.

- [ECM class tx_buf, rx_buf](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/netusb/function_ecm.c#L33)
- [EEM class tx_buf, rx_buf](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/netusb/function_eem.c#L26)
- [RNDIS class, tx_buf](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/netusb/function_rndis.c#L216)
- [RNDIS class, bulk out buffer](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/netusb/function_rndis.c#L327)
- [CDC ACM class rx_buf, tx_buf per instance](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/bluetooth.c)
- [CDC ACM class ring-buffers? per instance](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/cdc_acm.c#L186) 
- [loopback function](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/loopback.c#L26)
- [MSC class](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/mass_storage.c#L112)

Worse is the situation with the [standard, class and vendor request buffer](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/usb_device.c#L158). It is only used for the request over Control Pipe and also passed class drivers like hid and dfu. The buffer is usually not or little used during normal function. There is no verification that it is not written beyond the bounds of the buffer.

Classes already uses net_buf only:
- [Bluetooth USB HCI](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/bluetooth.c)

... will be continued ...

finikorg, 2019-10-10:
If usb_transfer() functions could use buffers Ethernet over USB code can be easily transformed.

jli157, 2021-06-30:
@jfischer-no  Any updates for this issue?

jfischer-no, 2021-06-30:
> @jfischer-no Any updates for this issue?

It is part of new USB device controller driver API, RFC PR will be opened soon.

carlescufi, 2022-12-02:
Fixed in the new USB stack, use `CONFIG_USB_DEVICE_STACK_NEXT` to enable it.

PR #38679 (2022-12-02): usb: add new (experimental) USB device support
Add new USB device controller API (UDC API)
Add new USB device controller API and nRF USBD controller driver.
Add new USB device stack implementation using new UDC API

Resolves #19713
Resolves #29132
Resolves #29133
Resolves #29134
Resolves #29135
Resolves #29136
Resolves #30042

_Commits introducing USB host controller (UHC) driver API are here for reasons and will be moved to own PR later._
closed 2022-12-02 by carlescufi