zephyrproject-rtos/zephyr #19721: samples/bluetooth/ipsp does not respond to pings from Linux

cfriedt, 2019-10-09:
**Describe the bug**
I was following the instructions at https://docs.zephyrproject.org/latest/samples/bluetooth/ipsp/README.html and found that I was unable to get an ICMPv6 response from the nRF board when pinging from my Ubuntu 18.04 machine. 

After enabling network logging, I observed the following, which shows ICMPv6 packets getting dropped if they have an invalid checksum.
```
[00:05:42.240,753] <dbg> net_ipv6.net_ipv6_input: (0x20001e48): IPv6 packet len 72 received from fe80::5ef3:70ff:fe92:5c06 to ff02::1:ffe9:6fd0
[00:05:42.240,814] <dbg> net_icmpv6.net_icmpv6_input: (0x20001e48): DROP: invalid checksum
```
After inspecting [the code](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/icmpv6.c#L376), indeed it seems to be the case that if `net_calc_chksum_icmpv6(pkt) != 0U` is not satisfied, then the packet is dropped.

Ideally, there would be an easy way to enable ICMPv6 checksum in Linux via `/proc/sys/net/ipv6/icmp` but the only node that exists there (at least on my desk) is `ratelimit`. If there were an easy way to enable it, then the Zephyr documentation could simply be modified.

It is a bit of a blocker to simply run a demo application. However, as @jukkar pointed out in Slack, it would be desirable to keep the checksum by default.

Therefore, I'm proposing to add a Kconfig variable to disable ICMP checksum verification and to explicitly set it to `n` for `samples/bluetooth/ipsp`.

That would also ensure that both the example and documentation remains consistent as well as the current behaviour of the ICMPv6 code.

**To Reproduce**
Steps to reproduce the behavior:

See [Bluetooth/IPSP](https://docs.zephyrproject.org/latest/samples/bluetooth/ipsp/README.html) sample.

**Expected behavior**
ping6 response 

**Impact**
It's an annoyance, but it is kind of a blocker for those just starting out with Zephyr.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA 60136f00cb8b040458ceed6ed2d542287da95fbc

**Additional context**
Add any other context about the problem here.

carlescufi, 2019-10-09:
@cfriedt would you be willing to send a Pull Request with the proposed Kconfig option to disable checksums?

jukkar, 2019-10-09:
As I mentioned in slack, if we want to have kconfig option to disable icmp checksums, then it needs to be off by default, and there needs to be separate options for ICMPv4 and ICMPv6.

cfriedt, 2019-10-09:
Was addressed in #19723

Closing for now, since the issue seems to have be resolved by rebasing.

closed 2019-10-09 by cfriedt