zephyrproject-rtos/zephyr #19734: "make gdbserver" doesn't work properly for qemu_x86_long

andrewboie, 2019-10-10:
When I run "make debugserver", I expect to be able to attach GDB in another terminal to interactively debug the kernel in the emulator. I do this all the time with qemu_x86.

However, with a build for qemu_x86_long, I am getting:

```
$ /opt/zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gdb zephyr/zephyr.elf -ex "target remote :1234"
GNU gdb (crosstool-NG 1.24.0-rc2-dirty) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=x86_64-zephyr-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from zephyr/zephyr.elf...(no debugging symbols found)...done.
Remote debugging using :1234

Remote 'g' packet reply is too long (expected 308 bytes, got 608 bytes): 0000000000000000000000000000000000000000000000006306000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000f0ff0000000000000200000000f00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010000060000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000007f0300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000801f0000
```

This works as expected for qemu_x86.

Key error message is here:

```
warning: Selected architecture i386 is not compatible with reported target architecture i386:x86-64
warning: Architecture rejected target-supplied description
```

Not sure how "i386" is getting selected.

Impact:
Prevents interactive debugging for qemu_x86_long.

andrewboie, 2019-10-10:
@gnuless have you run into this?

gnuless, 2019-10-10:
> @gnuless have you run into this?

I didn't even know `make gdbserver` was a thing.  Will take a look.  Probably a simple fix.

andrewboie, 2019-10-10:
I found a workaround:

```
set arch i386:x86-64
target remote :1234
```

If I make that set arch call before I connect to the target, it seems to work fine.

andrewboie, 2019-10-10:
This might be something beyond our control: https://wiki.osdev.org/QEMU_and_GDB_in_long_mode

(osdev is such a godsend)

gnuless, 2019-10-10:
> If I make that set arch call before I connect to the target, it seems to work fine.

We munge the ELF64 into an ELF32 before launching QEMU (it won't launch a 64-bit ELF directly), so that's probably why it's confused.

andrewboie, 2019-10-10:
perhaps have the munge not be in-place, but to a separate artifact that gets passed to the emulator

PR #19958 (2019-10-19): qemu_x86_long: don't munge to elf32 in-place
Instead, use the QEMU_KERNEL_FILE facility to perform the
change in a new file zephyr-qemu.elf. This is what will
be sent to the emulator.

Fixes an issue where opening zephyr.elf in GDB would treat
it as a 32-bit binary instead of what it actually is,
forcing the user to override the arch setting.

Fixes: #19734
commit beb81885 (2019-10-19): qemu_x86_long: don't munge to elf32 in-place

Instead, use the QEMU_KERNEL_FILE facility to perform the
change in a new file zephyr-qemu.elf. This is what will
be sent to the emulator.

Fixes an issue where opening zephyr.elf in GDB would treat
it as a 32-bit binary instead of what it actually is,
forcing the user to override the arch setting.

Fixes: #19734
closed 2019-10-19 by andrewboie