zephyrproject-rtos/zephyr #19737: No Function In Zephyr For Reading BLE Channel Map?

BOZHENG001, 2019-10-10:
**Is your feature request related to a problem? Please describe.**
I am now working on BLE development on Zehpyr, I would like to read channel map so that I can add some specific features to my application, but I found no function in Zephyr supporting this. There are only some functions for setting channel map.

**Describe the solution you'd like**
Since BLE channel hopping is based on channel map, and theoretically the channel map should change with surroundings by time. It would be useful if there is a function in Zephyr to read the channel map now.

**Describe alternatives you've considered**
It would be ok if there are some other alternatives/ways/methods to read the channel map.

aescolar, 2019-10-10:
CC @jhedberg @carlescufi

carlescufi, 2019-10-10:
I think we should probably use `bt_conn_info()` again for this one

joerchan, 2019-10-10:
@carlescufi @jhedberg Previously `bt_conn_get_info()` was for fetching connection information that was otherwise known only to the stack. such as addresses and connection parameters.

If we keep adding all of these HCI commands to the command the time for getting the information becomes much longer. Should we really add all of these to the conn info object? It is a bit overkill to read all possible data if you are only interested in a simple data field.

jhedberg, 2019-10-11:
@BOZHENG001 how exactly are you planning to read the channel map? Will you have some trigger to read it, or will you be doing polling at fixed intervals (if so, what kind of intervals)? I assume the controller knows when the map has changed, so it feels a bit inefficient if the host has to poll for it rather than the controller telling the host when a change has occurred.

BOZHENG001, 2019-10-11:
@jhedberg Exactly the point! It would be more efficient that the controller will trigger the host when a change has occurred. So let's say, for normal application development, this would be the best choice. But considering development process, it would be better that we developers could read the channel map by fixed intervals, because I found, even in our "crowded" office, the channel map is not that crowded. Then I suppose that channel map will not be updated frequently in a normal environment, and, in this situation, polling at fixed intervals would be a better choice.
Thanks for your help in advance!

joerchan, 2019-11-11:
@BOZHENG001 

As an alternative solution you can now do this in the application instead.

```
#include <bluetooth/conn.h>
#include <bluetooth/hci.h>

static int get_chan_map(struct bt_conn *conn)
{
	int err;
	struct net_buf *buf, *rsp;
	struct bt_hci_cp_le_read_chan_map *cp;
	struct bt_hci_rp_le_read_chan_map *rp;
	u16_t handle;

	err = bt_hci_get_conn_handle(conn, &handle);
	if (err) {
		printk("Failed to get connection handle\n");
		return err;
	}

	buf = bt_hci_cmd_create(BT_HCI_OP_LE_READ_CHAN_MAP,
				sizeof(*cp));

	if (!buf) {
		return -ENOBUFS;
	}

	cp = net_buf_add(buf, sizeof(*cp));
	cp->handle = sys_cpu_to_le16(handle);

	err = bt_hci_cmd_send_sync(BT_HCI_OP_LE_READ_CHAN_MAP, buf, &rsp);
	if (err) {
		printk("Failed to read channel map %d\n", err);
		return err;
	}

	rp = (struct bt_hci_rp_le_read_chan_map *)rsp->data;
	memcpy(&conn->le.chan_map[0], rp->ch_map, 5);

	net_buf_unref(rsp);
	return 0;
}
```

BOZHENG001, 2019-11-12:
@joerchan 
Thanks for your help, I will have a try!

joerchan, 2019-12-11:
> It would be ok if there are some other alternatives/ways/methods to read the channel map.

Closing this as an alternative method can be used.

closed 2019-12-11 by joerchan

BOZHENG001, 2020-03-03:
@joerchan 
Hi joerchan
I tried with the code in zephyr (both my old version 1.14 and new version 2.0), and i used it in **central_hr main.c**, but it does not work well. It shows some compiling errors as below:

`c:73:8: warning: implicit declaration of function 'bt_hci_get_conn_handle'; did you mean 'bt_br_set_connectable'? [-Wimplicit-function-declaration]
  err = bt_hci_get_conn_handle(conn, &handle);
        ^~~~~~~~~~~~~~~~~~~~~~
        bt_br_set_connectable
/home/bozheng/zephyrproject/zephyr/samples/bluetooth/central_hr/source/src/main.c:96:14: error: dereferencing pointer to incomplete type 'struct bt_conn'
  memcpy(&conn->le.chan_map[0], rp->ch_map, 5);
              ^~
At top level:
/home/bozheng/zephyrproject/zephyr/samples/bluetooth/central_hr/source/src/main.c:65:12: warning: 'get_chan_map' defined but not used [-Wunused-function]
 static int get_chan_map(struct bt_conn *conn)
            ^~~~~~~~~~~~
`

I tried to check `struct bt_conn` in zephyr, and it shows as an **Opaque type representing a connection to a remote device**. Do you have any suggestion for me to solve this?

joerchan, 2020-03-03:
@BOZHENG001 You might have to go to 2.1 to get the missing function. Or maybe cherry-pick the commit (0f06c7d8e4626c6793c2a9f34d17113ba9cb4ec4).

BOZHENG001, 2020-03-03:
@joerchan Thanks for your hints! I only tried the cherry-pick in my zephyr 2.0, and it works! But I found the results from this function is always the default channel map (ff ff ff ff 1f). I suppose it means this function only reads the channel map, but does not sense environment and update channel map from environment.
I am wondering if we have another function to sense environment and update channel map? (I know we can use `bt_le_set_chan_map()` to update channel map, but how should we get the real channel map from environment?)

joerchan, 2020-03-03:
@BOZHENG001 It reads the channel map that is being used in the connection at the moment.
Updating the channel map is a different thing. We don't have anything for that yet.