zephyrproject-rtos/zephyr #19748: k_sleep(K_FOREVER) behavior unexpected

pabigot, 2019-10-10:
> One side comment. During playing with k_sleep and negative numbers I've noticed that `k_sleep(K_FOREVER)` returns immediately. Is K_FOREVER intended to be used for `k_sleep`?

_Originally posted by @nordic-krch in https://github.com/zephyrproject-rtos/zephyr/pull/19591#issuecomment-540554331_

Since we had recently reached some consensus that `k_sleep` should use the standard timeout API it should support `K_FOREVER`.

If it does not a fix might be integrated with rework of `k_poll` since functionally `k_sleep` could be seen as `k_poll` with no events.   (The current implementation has asserts that at least one event is provided, but passing zero events works as expected when asserts are not enabled, which is the default condition.)

aescolar, 2019-10-10:
@pabigot So, an enhacement? ;)

nordic-krch, 2019-11-08:
Note that there are few places in the tree where `k_sleep(FOREVER)` is used even though it expires immediately (probably against user's intention).

PR #20482 (2019-11-08): kernel: allow threads to sleep forever
Previously, passing K_FOREVER to k_sleep()/k_usleep()
would return immediately.

Forever is a long time. Even if woken up at some point,
we still had forever to sleep, so return K_FOREVER in this
case.

Fixes: #19748
closed 2019-11-08 by carlescufi