zephyrproject-rtos/zephyr #19759: z_arch_switch() passed pointer to NULL outgoing switch handle

andrewboie, 2019-10-10:
If CONFIG_SWITCH is enabled, the initial context switch into the main thread (either in prepare_multithreading() or smp_init_top() has a fundamental flaw. It's relying on a partially initialized "dummy_thread" for the outgoing thread information, with dummy_thread->switch_handle being NULL.

This causes a NULL pointer dereference to any arch that uses the outgoing switch handle for anything. When I enable memory protection this causes a fatal page fault on intel64.

Some solutions I've considered:
1.  Add a NULL check in the arch switch code for the outgoing thread handle (slow)
2. Implement CONFIG_ARCH_HAS_CUSTOM_SWAP_TO_MAIN instead of a dummy thread, which would be mandatory for all arches implementing CONFIG_SWITCH, and there is no support for this yet in smp_init_top()
3. Change switch handles to be something typed (like the actual thread objects) so that dummy thread switch handles can be populated in kernel code in an arch-agnostic way

To me option 3 seems preferable

andyross, 2019-10-17:
Not completely understanding the problem.  What's trying to use the switched_from handle of a dummy thread?  Sounds like there's an unrelated bug here, maybe?  I think it's very reasonable to have a requirement that the pointers always be non-null at the arch layer and put the complexity into the kernel, that just isn't the way it works currently because of that dummy thread trick.

As far as changing a switch handle into a thread struct, I guess I don't see how it's related to the NULL issue described here exactly.  But regardless, sure, it can be done.  It will require pretty significant surgery to xtensa to make work, and defeat some relatively useful optimizations (the cross-stack call trick, pushing only caller-save registers on interrupt entry, stuff like that) there to the interrupt entry and context switch code (the same is true for x86_64, but... yeah).   But yeah, that would work just fine too.

I continue to weep internally that I haven't been able to convince anyone of the value of this architecture (which is certainly not unique to Zephyr, FWIW), but it's not like it's the only way to do things.

andyross, 2019-10-17:
Just to clarify the technical question: having a NULL in the switch_handle fields just means that a thread has not been suspended by interrupt entry or a context switch (or thread initialization, of course).  Trying to switch back "to" such a thread simply doesn't make any sense, it doesn't have state to restore.

So we need to figure out where and why that's happening, regardless of what we do regarding the typing of the switch handles or whatever.

andrewboie, 2019-10-17:
@andyross 

z_arch_switch() takes two arguments: switch_to, switched_from.

On the initial context switch from dummy thread, switched from is a pointer to NULL.

> Not completely understanding the problem. What's trying to use the switched_from handle of a dummy thread? Sounds like there's an unrelated bug here, maybe?

The implementation of z_arch_switch()

andrewboie, 2019-10-17:
> defeat some relatively useful optimizations (the cross-stack call trick, pushing only caller-save registers on interrupt entry, stuff like that) there to the interrupt entry and context switch code (the same is true for x86_64, but... yeah). But yeah, that would work just fine too.

Isn't all the data being manipulated at some offset from the thread struct anyway? Not seeing how this is an optimization, and it certainly hasn't been documented.

At the end of the day I don't care how this is gets solved, other than I really don't want to have to pull NULL checks in z_arch_switch() that are only true when a thread is initially started.

andrewboie, 2019-10-17:
> I think it's very reasonable to have a requirement that the pointers always be non-null at the arch layer and put the complexity into the kernel, that just isn't the way it works currently because of that dummy thread trick.

The NULL in dummy_thread->switch_handle gets passed all the way to z_arch_switch()

andyross, 2019-10-17:
> Isn't all the data being manipulated at some offset from the thread struct anyway? 

No, on xtensa and x86_64 the context lives on the process stack.  Docs are at the arch layer obviously, see for example https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/xtensa/include/xtensa-asm2-context.h and https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/x86_64/core/xuk.h#L93  Doing it this way allows the entry/exit code to skip the stack pointer jiggling that would otherwise be required.  For example interrupt exit and the bottom half of switch just pop a bunch of stuff and issue an IRET synchrononously.  Or you can implement the top half of switch with a few pushes of a faked interrupt entry frame followed by a simple CALL to code that gets shared with interrupt entry.  You can also easily implement "don't spill callee-saved registers when invoking interrupt handlers" by just calling out and then pushing after the return.    Xtensa actually goes even farther with this idea and arranges for a "cross stack call" frame that allows us to avoid spilling not just the ABI registers but in fact all the up-stack register windows on the CPU.  Really, it's a good trick.  Linux does it this way too.  But it requires that you pass the resulting stack pointer around as the handle, or else store it somewhere in the thread struct (which is the exact converse of the complaint here, if I'm understanding correctly).

> The NULL in dummy_thread->switch_handle gets passed all the way to z_arch_switch()

We should find out why, then.  Do you have a trace of what is trying to context switch into a dummy?  Dummy threads shouldn't do anything but take interrupts, once any other thread gets scheduled they should never be scheduled again.

andyross, 2019-10-17:
I threw an assertion in trying to catch this and am not having any luck spotting it on the existing platforms in master.  Is there maybe something specific to the memory protection initialization that is trying to switch back into a dummy?

andyross, 2019-10-17:
Still trying to figure out what the actual problem being reported is.  This bit has me confused:

> z_arch_switch() takes two arguments: switch_to, switched_from.  On the initial context switch from dummy thread, switched from is a pointer to NULL.

That's as designed.  The switched_from argument is an output pointer: the switch code is expected to store the resulting switch handle of the current thread into that pointer before begining to execute the "switch_to" context.  Nothing should ever be reading from that pointer, just storing into it.

And FWIW, this quick patch produces working tests/kernel results on qemu_x86_64, qemu_x86_long and qemu_xtensa.  As far as I can tell nothing should ever be seeing this null.  How do I reproduce?

```patch
--- a/kernel/include/kswap.h
+++ b/kernel/include/kswap.h
@@ -78,6 +78,7 @@ static ALWAYS_INLINE unsigned int do_swap(unsigned int key,
                }
 #endif
                _current = new_thread;
+               __ASSERT(new_thread->switch_handle != NULL, "");
                z_arch_switch(new_thread->switch_handle,
                             &old_thread->switch_handle);
        }
```

andrewboie, 2019-10-17:
> Is there maybe something specific to the memory protection initialization that is trying to switch back into a dummy?

No...all I did was install a page table which has the 0 page unmapped and I get a fault. There is no userspace, no stack overflow protection, just you can't access the NULL page

Here, pull this branch and run any test with qemu_x86_long:

https://github.com/andrewboie/zephyr/tree/switch_problem

If you then cherry-pick in https://github.com/andrewboie/zephyr/commit/9e0834d10431f9f362bacd2ede305e3c43fe46a2 you will see that the crash goes away

andrewboie, 2019-10-17:
> And FWIW, this quick patch produces working tests/kernel results on qemu_x86_64, qemu_x86_long and qemu_xtensa. As far as I can tell nothing should ever be seeing this null. How do I reproduce?

Memory protection HW isn't enabled for ANY of these.
You have to pull my branch.

andrewboie, 2019-10-17:
This is the code on intel64:

```

/*
 * void z_x86_switch(void *switch_to, void **switched_from);
 *
 * Note that switch_handle for us is simply a pointer to the containing
 * 'struct k_thread', thus:
 *
 * RDI = (struct k_thread *) switch_to
 * RSI = (struct k_thread **) switched_from
 */

.globl z_x86_switch
z_x86_switch:
	movq (%rsi), %rsi

	andb $~X86_THREAD_FLAG_ALL, _thread_offset_to_flags(%rsi)

	popq %rax
	movq %rax, _thread_offset_to_rip(%rsi)
	movq %rsp, _thread_offset_to_rsp(%rsi)
	movq %rbx, _thread_offset_to_rbx(%rsi)
	movq %rbp, _thread_offset_to_rbp(%rsi)
	movq %r12, _thread_offset_to_r12(%rsi)
	movq %r13, _thread_offset_to_r13(%rsi)
	movq %r14, _thread_offset_to_r14(%rsi)
	movq %r15, _thread_offset_to_r15(%rsi)

	movq %gs:__x86_tss64_t_ist1_OFFSET, %rsp

	/* fall through to __resume */

/*
 * Entry:
 *   RSP = top of CPU interrupt stack
 *   RDI = (struct k_thread *) thread to resume
 */

__resume:
	pushq $X86_KERNEL_DS			/* SS */
	pushq _thread_offset_to_rsp(%rdi)	/* RSP */
	pushq _thread_offset_to_rflags(%rdi)	/* RFLAGS */
	pushq $X86_KERNEL_CS			/* CS */
	pushq _thread_offset_to_rip(%rdi)	/* RIP */

	movq _thread_offset_to_rbx(%rdi), %rbx
	movq _thread_offset_to_rbp(%rdi), %rbp
	movq _thread_offset_to_r12(%rdi), %r12
	movq _thread_offset_to_r13(%rdi), %r13
	movq _thread_offset_to_r14(%rdi), %r14
	movq _thread_offset_to_r15(%rdi), %r15

	testb $X86_THREAD_FLAG_ALL, _thread_offset_to_flags(%rdi)
	jz 1f

	fxrstor _thread_offset_to_sse(%rdi)
	movq _thread_offset_to_rax(%rdi), %rax
	movq _thread_offset_to_rcx(%rdi), %rcx
	movq _thread_offset_to_rdx(%rdi), %rdx
	movq _thread_offset_to_rsi(%rdi), %rsi
	movq _thread_offset_to_r8(%rdi), %r8
	movq _thread_offset_to_r9(%rdi), %r9
	movq _thread_offset_to_r10(%rdi), %r10
	movq _thread_offset_to_r11(%rdi), %r11
	movq _thread_offset_to_rdi(%rdi), %rdi  /* do last :-) */

1:	iretq
```

andrewboie, 2019-10-17:
@andyross 

```
--- a/kernel/include/kswap.h
+++ b/kernel/include/kswap.h
@@ -78,6 +78,7 @@ static ALWAYS_INLINE unsigned int do_swap(unsigned int key,
                }
 #endif
                _current = new_thread;
+               __ASSERT(new_thread->switch_handle != NULL, "");
                z_arch_switch(new_thread->switch_handle,
                             &old_thread->switch_handle);
        }
```

Your assert is incorect
You need

``
__ASSERT(old_thread->switch_handle != NULL, "");
```

It's the OLD thread pointer which is NULL

andyross, 2019-10-17:
OK, thanks.  That's the part I didn't understand.  FWIW: old_thread is just a copy of _current.  It too is never supposed to be null if we're executing code that is capable of swapping.  Will continue digging.

But do note that x86_64 in fact **does** have a 16k null guard, so it's not as simple as the pointer being null as it should trip there too.

andyross, 2019-10-17:
Wait wait, sorry.  One more clarification, is this bit a typo?:

> __ASSERT(old_thread->switch_handle != NULL, "");

Because as mentioned above that's 100% legal and expected.  If "old_thread" itself is null, that's a bug.  But the state of a currently-executing thread's switch_handle field is (for obvious reasons) indeterminate garbage.  You can only restore it after it's been saved, not while it's running.

andrewboie, 2019-10-17:
```
$ git diff
diff --git a/kernel/include/kswap.h b/kernel/include/kswap.h
index d98fa44618..bc50889299 100644
--- a/kernel/include/kswap.h
+++ b/kernel/include/kswap.h
@@ -78,6 +78,7 @@ static ALWAYS_INLINE unsigned int do_swap(unsigned int key,
                }
 #endif
                _current = new_thread;
+               __ASSERT(old_thread->switch_handle != NULL, "");
                z_arch_switch(new_thread->switch_handle,
                             &old_thread->switch_handle);
        }
```

this reproduces the issue in a very simple form

andyross, 2019-10-17:
Whoops, crossed. Yeah, that's not a bug.  We need to locate the code that's dereferencing old_thread->switch_handle and fix it.

andrewboie, 2019-10-17:
> Whoops, crossed. Yeah, that's not a bug. We need to locate the code that's dereferencing old_thread->switch_handle and fix it.

Wait so this is supposed to be NULL?
You are saying we add a NULL check in x86 long's z_arch_switch? For EVERY context switch, even though this just applies to the first switch out of the dummy thread?

andrewboie, 2019-10-17:
The implementation of z_arch_switch() saves state to the old_thread->switch handle (which is expected to be a pointer to old_thread itself) and then pops state off the new_thread->switch_handle (which is a pointer to new_thread)

If x86 long's z_arch_switch() is implemented incorrectly we need help here, the intended design is very unclear

andrewboie, 2019-10-17:
And if the switched_from parameter is supposed to be output-only and its contents are garbage when running, how does the implementation of z_arch_switch() know what outgoing thread to save state to?

andyross, 2019-10-17:
I apologize.  Indeed we never had a single spot for arch docs so the switch implementation docs in xtensa and x86_64 was all we had for a long while.   The current design docs at https://docs.zephyrproject.org/latest/reference/kernel/smp/smp.html#switch-based-context-switching are, I think, really not so terrible though.

> how does the implementation of z_arch_switch() know what outgoing thread to save state to?

It doesn't.  That's the kernel's job.  The scheduler, not the arch, figures out what threads are swapping and just tells the arch layer where to stick the resulting handle (likewise, the kernel decides what handle to return from on interrupt entry and takes the interrupted context handle as a simple function argument).  The intent was that this was low-level code typically written in assembly, often implemented over a non-zephyr HAL, and that we wanted to minimize the amount of Zephyr-knowledge needed to do it.

andrewboie, 2019-10-17:
> It doesn't. That's the kernel's job. The scheduler, not the arch, figures out what threads are swapping and just tells the arch layer where to stick the resulting handle (likewise, the kernel [...]

How do we fix this in terms of how z_arch_switch() is implemented in arch/x86/core/intel64/locore.S?
Because right now it's treating the switched_from parameter as input/output and not output only.

andrewboie, 2019-10-17:
It looks like ARC has the same problem, they added a band-aid NULL check to skip, context, impacting performance:

```

SECTION_FUNC(TEXT, z_arch_switch)

#ifdef CONFIG_EXECUTION_BENCHMARKING
	push_s r0
	push_s r1
	push_s blink

	bl read_timer_start_of_swap

	pop_s blink
	pop_s r1
	pop_s r0
#endif
	/*
	 * r0 = new_thread->switch_handle = switch_to thread,
	 * r1 = &old_thread->switch_handle = &switch_from thread
	 */

	ld_s r2, [r1]
	/*
	 * r2 may be dummy_thread in z_cstart, dummy_thread->switch_handle
	 * must be 0
	 */
	breq r2, 0, _switch_to_target_thread
```

andrewboie, 2019-10-17:
> The intent was that this was low-level code typically written in assembly, often implemented over a non-zephyr HAL, and that we wanted to minimize the amount of Zephyr-knowledge needed to do it.

I'll be blunt this seems like a solution looking for a problem. Every single arch is going to provide their own custom z_arch_switch(), I do not see how some third-party HAL is going to provide a z_arch_switch() with the same semantics Zephyr is expecting

andyross, 2019-10-17:
But as for the intel64 stuff, yeah, it's using the contents of RSI as synonym for _current, which isn't correct.  But it's very close to correct except for the case of the initial dummy thread.  I'd look at solutions that involve stuffing the "correct" arch-specific value into the dummy thread swap_handle field before switch() sees it, instead of testing inside of switch.  Maybe we could stick an arch step in in z_cstart() that allowed the arch layer to init the dummy thread (which is sort of what CUSTOM_SWAP_TO_MAIN does, but maybe simpler), or we could pass the dummy thread as an argument to z_arch_kernel_init()....

And as far as being blunt: not only was this a solution that was already present and working before being replaced with one that doesn't, the bug with stuffing context into into a null pointer went unnoticed because of the removal of the null guard page table in the early implementation that would have caught this mistake long before it merged.  I mean...  this doesn't seem fair to make me responsible for mistakes made when replacing code I wrote that didn't have them.

andrewboie, 2019-10-17:
> not only was this a solution that was already present and working before being replaced with one that doesn't, the bug with stuffing context into into a null pointer went unnoticed because of the [...]

I don't see past history as germane to the discussion, and I think anyone should be able to speak freely about design points regardless of how long a particular implementation has been in place.

We have two platforms, ARC and x86 long which are expecting the switch handles to be valid thread pointers. ARC works around it at a cost of performance, x86 long currently crashes but for both we need a solution.

> Maybe we could stick an arch step in in z_cstart() that allowed the arch layer to init the dummy thread

We would need to add this call to smp_init_top() as well, but this seems a reasonable way to address this for both ARC and x86 long without impacting context switch performance.

andrewboie, 2019-10-17:
> We would need to add this call to smp_init_top() as well, but this seems a reasonable way to address this for both ARC and x86 long without impacting context switch performance.

@andyross actually this still isn't optimal. For these platforms (x86 long, ARC) which store outgoing thread information in thread->arch, we have to take the address of switched_from parameter when calling z_arch_switch(), and this is immediately dereferenced. This is some extra overhead that I do not think is optimal.

AFAICT, switch design seems more oriented towards push all your outgoing context onto whatever the current stack is, and then update switched_from to be the stack pointer after everything has been pushed. In this scenario, switched_from truly is an output-only parameter.

So it seems to me that we should not be stashing context in thread->arch like ARC and x86 long currently are, instead they should just push everything to the current stack and update switched_from to be the stack pointer? This seems the fastest way to do this and makes switched_from truly output-only.

andyross, 2019-10-18:
Well, at least we've moved from "fundamental flaw" to "still isn't optimal", I guess.  The dereference isn't actually needed, it's just the way the code works right now.  You can recover the thread pointer from the field address with a subtract.  I'll submit something later today.

But yes: this was designed with the intent of allowing stack-based context frames to be implemented naturally.  But it really doesn't require that, it just doesn't provide a space on its own to store the spilled registers.

andrewboie, 2019-10-18:
>  The dereference isn't actually needed, it's just the way the code works right now. You can recover the thread pointer from the field address with a subtract. I'll submit something later today.

Thanks.

My understanding at this point is that if the switch handles are stack locations, this all works flawlessly and there are no problems with the design.

However if the handles are something related to the thread struct, or the thread struct itself, I think we have some extra overhead, and these unanticipated problems with dummy thread initialization.

Does it make sense to make a documentation pass on this? We have a prototype for z_arch_switch() in include/sys/arch_interface.h that could have some expansion on its docstring. You do have good overall SMP documentation but given that ARC and x86-long tripped over the details, and I got confused too, I think it's worth doing..

Also at some point I think we should re-do the ARC and x86 long switch implementations so that they are using stack-based handles? This seems to allow for the tightest context switching implementation.

PR #19952 (2019-10-19): arch/x86/intel64: Fix assumption with dummy threads
The intel64 switch implementation doesn't actually use a switch handle
per se, just the raw thread struct pointers which get stored into the
handle field.  This works fine for normally initialized threads, but
when switching out of a dummy thread at initialization, nothing has
initialized that field and the code was dumping registers into the
bottom of memory through the resulting NULL pointer.

Fix this by skipping the load of the field value and just using an
offset instead to get the struct address, which is actually slightly
faster anyway [... cut]
closed 2019-10-19 by andrewboie