zephyrproject-rtos/zephyr #19761: tests/net/ieee802154/fragment failed on reel board.

chen-png, 2019-10-11:
**Describe the bug**
There is a "kernel panic" error when running tests/net/ieee802154/fragment case on reel board.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b=reel_board tests/net/ieee802154/fragment
2. west flash
3. See error

**Screenshots or console output**
I: nRF5 802154 radio initialized
W: You have 1 IPv6 net_if addresses but 2 network interfaces
W: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value.
***** Booting Zephyr OS build zephyr-v2.0.0-1126-g91384ef01eb6 *****
Running test suite ieee802154_fragment

starting test - test_fragment_sam00_dam00
PASS - test_fragment_sam00_dam00

starting test - test_fragment_sam01_dam01
ASSERTION FAIL [net_pkt_lladdr_dst(pkt)->addr] @ ZEPHYR_BASE/subsys/net/ip/6lo.c:511
E: r0/a1:  0x00000004  r1/a2:  0x000001ff  r2/a3:  0x00000001
E: r3/a4:  0x0000f784 r12/ip:  0x00000f01 r14/lr:  0x00003c51
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x0000caf6
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic
E: Current thread: 0x20000aac (unknown)
E: Halting system

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: 91384ef01eb6

jukkar, 2019-10-11:
> W: You have 1 IPv6 net_if addresses but 2 network interfaces
> W: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value.

Have you considered increasing CONFIG_NET_IF_MAX_IPV6_COUNT to 2 as suggested by the output?
Can you try if it fixes the issue?

chen-png, 2019-10-12:
> > W: You have 1 IPv6 net_if addresses but 2 network interfaces
> > W: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value.
> 
> Have you considered increasing CONFIG_NET_IF_MAX_IPV6_COUNT to 2 as suggested by the output?
> Can you try if it fixes the issue?

I added the "CONFIG_NET_IF_MAX_IPV6_COUNT=2" into the prj.conf and ran the test case again, it would not warning "You have 1 IPv6 net_if addresses but 2 network interfaces. Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value", but still reported this error.

jukkar, 2019-10-13:
> but still reported this error.

Ok, thanks for verifying this.

PR #22090 (2020-01-22): net: ieee802154: Sort out fragmentation tests
Fragmentation tests in `tests/net/ieee802154` failed due to asserts on LL addresses and a memory corruption issue. This appears to be happening since the recent 6lowpan rework.

@jukkar It seems that these tests are not automatically executed during the sanity-check, that's why it went so long unnoticed. A short investigation showed up that the ieee802514 tests have the following dependency: `depends_on: ieee802154`, yet no emulated qemu board sets `ieee802154` in their capabilities. IMO we could fix it in this PR as well. Shall we add the capabili [... cut]
commit 8801fd01 (2020-01-22): tests: net: ieee802154: Set LL addresses on a packet

The reworked 6lowpan implementation asserts when no LL source or
destination address is set on a packet. This caused the fragmentation
tests to fail before they completed.

Fix the issue, by setting a dummy address on a packet during a setup.

Fixes #19761
closed 2020-01-22 by MaureenHelm