zephyrproject-rtos/zephyr #19762: tests/net/lib/tls_credentials failed on sam_e70_xplained board.

chen-png, 2019-10-11:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b=sam_e70_xplained tests/net/lib/tls_credentials/
2. west flash
3. See error

**Screenshots or console output**
I: Device I2C_0 initialized
I: MAC: fc:c2:3d:11:fc:09
I: Queue 0 activated
I: Queue 1 set to idle
I: Queue 2 set to idle
I: Soft Reset of ETH PHY
I: PHYID: 0x221561 at addr: 0
E: ETH PHY auto-negotiate sequence failed
E: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E:   Data Access Violation
E:   MMFAR Address: 0x20401a1c
E: r0/a1:  0xaaaaaaaa  r1/a2:  0xaaaaaaaa  r2/a3:  0x20400e2c
E: r3/a4:  0x00408765 r12/ip:  0x00000000 r14/lr:  0x204007bc
E:  xpsr:  0x00408c31
E: Faulting instruction address (r15/pc): 0x00000020
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow
E: Current thread: 0x2040090c (main)
E: Halting system

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: 91384ef01

rlubos, 2019-10-14:
I don't have this specifc board, and could not reproduce the fault on other platforms. It seems pretty straightforward though:
>E: >>> ZEPHYR FATAL ERROR 2: Stack overflow
E: Current thread: 0x2040090c (main)

Shall we just increase the main stack size?

jukkar, 2019-10-14:
> Shall we just increase the main stack size?

+1 to that. ping @chen-png could test that (increasing main stack size)

chen-png, 2019-10-15:
> +1 to that. ping @chen-png could test that (increasing main stack size)

I create a tests/net/lib/tls_credentials/boards/sam_e70_xplained.conf and fill one line "CONFIG_MAIN_STACK_SIZE=1344" in it, but it seems that no work.

jukkar, 2019-10-15:
For testing TLS stack requirements, I would suggest something like 2048 here.

jukkar, 2019-10-15:
I managed to test this with sam-e70, and setting main stack size to 2048 fixed the issue. I will send a PR for this.

PR #19832 (2019-10-17): tests: net: tls: Increasing stack size
The default main stack is too small for some boards so increasing
it to 2048. The issue was seen with sam_e70_xplained.

Fixes #19762
commit d500ac87 (2019-10-17): tests: net: tls: Increasing stack size

The default main stack is too small for some boards so increasing
it to 2048. The issue was seen with sam_e70_xplained.

Fixes #19762
closed 2019-10-17 by jukkar