zephyrproject-rtos/zephyr #19763: tests/subsys/usb/device/ failed on mimxrt1050_evk board.

chen-png, 2019-10-11:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b mimxrt1050_evk tests/subsys/usb/device/
2. west flash
3. See error

**Screenshots or console output**
***** Booting Zephyr OS build zephyr-v2.0.0-1126-g91384ef01eb6 *****
D: __usb_descriptor_start 0x80002b68
D: __usb_descriptor_end 0x80002bc1
D: Configuration descriptor 0x80002b7a
D: Interface descriptor 0x80002b83
D: Endpoint descriptor 0x80002b8c
D: Fixing EP address 81 -> 81
D: endpoint 0x81
D: Now the wTotalLength is 25
D: idx_max 11, ascii_idx_max 5, buf 0x80002b99
D: char R : 52, idx 5 -> 11
D: char Y : 59, idx 4 -> 9
D: char H : 48, idx 3 -> 7
D: char P : 50, idx 2 -> 5
D: char E : 45, idx 1 -> 3
D: char Z : 5a, idx 0 -> 1
D: idx_max 13, ascii_idx_max 6, buf 0x80002ba7
D: char V : 56, idx 6 -> 13
D: char E : 45, idx 5 -> 11
D: char D : 44, idx 4 -> 9
D: char - : 2d, idx 3 -> 7
D: char B : 42, idx 2 -> 5
D: char S : 53, idx 1 -> 3
D: char U : 55, idx 0 -> 1
D: idx_max 7, ascii_idx_max 3, buf 0x80002bb7
D: char 1 : 31, idx 3 -> 7
D: char 0 : 30, idx 2 -> 5
D: char . : 2e, idx 1 -> 3
D: char 0 : 30, idx 0 -> 1
D: attached
D: set cb, ep: 0x81
Running test suite test_device

starting test - test_usb_dc_api_invalid

    Assertion failed at ZEPHYR_BASE/tests/subsys/usb/device/src/main.c:124: test_usb_dc_api_invalid: (usb_dc_ep_set_stall(INVALID_EP) equal to TC_PASS)
usb_dc_ep_set_stall(INVALID_EP)
FAIL - test_usb_dc_api_invalid

starting test - test_usb_dc_api
PASS - test_usb_dc_api

starting test - test_usb_dc_api_read_write
E: Wrong endpoint index/address/direction
E: ***** BUS FAULT *****
E:   Imprecise data bus error
E: r0/a1:  0x800012f8  r1/a2:  0x00000080  r2/a3:  0x40184000
E: r3/a4:  0x00000001 r12/ip:  0x00000000 r14/lr:  0x60004607
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x600055cc
E: >>> ZEPHYR FATAL ERROR 0: CPU exception
E: Current thread: 0x8000136c (unknown)
E: Halting system

**Environment (please complete the following information):**
 - OS: Fedora29
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: 91384ef01

MaureenHelm, 2019-10-14:
@MarkWangChinese please have a look

jfischer-phytec-iot, 2019-11-22:
Lower the priority, it is _just_ a parameter check, will take a look at it next days.

PR #21175 (2020-03-10): drivers: rework usb_dc_mcux_ehci
This is bug fix, rework, and style fix of usb_dc_mcux_ehci

Fixes #19763
commit e0b2227c (2020-03-10): drivers: usb_dc_mcux_ehci: add checks for valid endpoint

Add checks for valid endpoint.
Fix controllerHandle check in usb_dc_detach().

Fixes: #19763
closed 2020-03-10 by carlescufi