zephyrproject-rtos/zephyr #19783: floating point in C++ on x86_64 uses SSE

lowlander, 2019-10-12:
**Describe the bug**
When using floating point in C++ the compiler seems to use SSE and give the following error;
```
/data/opt/zephyr-sdk-0.10.3/x86_64-zephyr-elf/x86_64-zephyr-elf/include/c++/8.3.0/limits: In static member function 'static constexpr float std::numeric_limits<float>::min()':
/data/opt/zephyr-sdk-0.10.3/x86_64-zephyr-elf/x86_64-zephyr-elf/include/c++/8.3.0/limits:1599:35: error: SSE register return with SSE disabled
       min() _GLIBCXX_USE_NOEXCEPT { return __FLT_MIN__; }
                                   ^
```

**To Reproduce**
Using for example `std::numeric_limits<float>::min()` 

**Expected behavior**
It should "just work" and not use SSE

**Impact**
Can't use C++ on x86_64 because std::chrono doesn't work.

**Environment (please complete the following information):**
 - OS: linux 
 - Toolchain Zephyr SDK 0.10.3
 - Commit master / 5881f118c0a4dbef8b6534327a666e4fa156ce65

lowlander, 2019-10-12:
This seems to be a general problem, the x86-64 calling convention uses MMX registers for floating point return values, so without sse no floating point unless something "special" is done when building the compiler.

On godbolt the same test fails with gcc and clang;

https://godbolt.org/z/2nUZnl

jukkar, 2019-10-14:
Also reported at #19066 when using C compiler.

lowlander, 2019-10-15:
@andrewboie x86_long and x86, and all other platforms that are used with sanitycheck.sh work. 

If x86_64 is "deprecated" should I just close this bug? 

PS: would it be an idea to "remove" x86_64 from sanitycheck.sh so not all testst that cause problems have to have an "exclude" ?

gnuless, 2019-10-15:
@lowlander I'll be putting in a PR in a few days to remove x86_64 from the tree.

lowlander, 2019-10-15:
@gnuless leave this bug open, or should I close it ?

andrewboie, 2019-10-31:
arch/x86_64 has been removed
However its replacement, arch/x86/intel64, doesn't have working floating point.

lowlander, 2019-10-31:
@andrewboie no float at all?

andrewboie, 2019-10-31:
@lowlander disregard, it's there, just not lazy fp save/restore (which has security issues on current CPUs anyway).

I think we can close this now.

andrewboie, 2019-10-31:
mis-click, see if you have the same problem with the new x86 64-bit port

lowlander, 2019-10-31:
Is the new x86-64 all just in master ?

lowlander, 2019-11-04:
The SSE problem is not reproducible anymore, closing this bug report.

closed 2019-11-04 by lowlander