zephyrproject-rtos/zephyr #19791: How to use CMSIS DSP Library on nRF52832 running zephyr LTS Version(V1.14) ?

icephyr, 2019-10-14:
I met a question here. Hope anybody can help on it, thanks.

I'm evaluating whether it is feasible to run the FFT algorithm on zephyr LTS project, with the nRF52832 board. I saw some sayings that it can work with the CMSIS DSP library, but I only found the CMSIS-rtos-v1 and CMSIS-rtos-v2 in zephyr source code (LTS version 1.14), not DSP. So I wonder whether the DSP library is available in zephyr. 
Need I download the CMSIS DSP project manually and intergrate into the zephyr project by myself ? Or the library has been integrated into the SDK, just need to include the head files ?

Some samples can be provided is better.

SebastianBoe, 2019-10-15:
> So I think the CMSIS souce code shall be downloaded and included manually

This is correct. I am not aware of anyone that has done this, but if the code is easy
to build it shouldn't be too difficult.

SebastianBoe, 2019-10-15:
> hope anybody knows how to use FFT on zephyr

I am not aware of which FFT implementation other FFT users have used.

icephyr, 2019-10-16:
> > So I think the CMSIS souce code shall be downloaded and included manually
> 
> This is correct. I am not aware of anyone that has done this, but if the code is easy
> to build it shouldn't be too difficult.

Well noted, I will have a try, thanks.

closed 2019-10-16 by SebastianBoe