zephyrproject-rtos/zephyr #19798: Proposal: Modifications to the settings module

Laczen, 2019-10-14:
## Introduction

The settings module provides support for multiple source backends simultaneously. Multiple backends are processed in the order they have been registered. It has been decided that starting from Zephyr 2.1 backends will only return the latest valid information found in the backend and that deleted information will not be returned. This is good as most users were struggling with the fact that the set method was called multiple times.

### Problem description

The proposed change does not address the problem of how to handle multiple backends, backends are processed in their order and this can still result in multiple calls to the set function
for each backend. As deleted items are no longer loaded a "reset" of the information by a later backend is no longer processed.

As the use of multiple backends is rare it will probably not be visible until it is being used.

### Proposed change

There are several possibilities in solving this problem:
a. For each load going through all remaining backends and delaying the loading until the real last entry was found. This change could be challenging and could have a negative effect on loading speed.
b. A much simpler solution could be to drop support of simultaneous multiple backends. This would remove any registering of sources and destinations, remove the list of backends, ... only the currently registered backend is treated as source and destination. Still it would be possible to support multiple backends (e.g. factory defaults, current running data, ...) but they would be user selected and not simultaneous. The user would set (register) the backend and call settings_load().

carlescufi, 2019-10-15:
CC @rakons @nvlsianpu @jhedberg @Vudentz

nvlsianpu, 2019-10-15:
cc @fnde-ot @aescolar

nvlsianpu, 2019-10-15:
Indeed - you spot something I overlooked as I'm concentrated on one back-end settings user case.
Maybe code which will make "loading the last" working on multiple back-end might be optional.

rakons, 2019-10-16:
Well... Then settings should support only one backend, and for those that wish to have more backends, we need special backend - that would be kind of "backend splitter". Then it might work both ways - for example select backend to write by the key.

fnde-ot, 2019-10-17:
I would agree that settings should support only one backend. Note that it is already possible to implement a custom settings backend that can manage multiple data sources.
We actually use this to store settings in multiple locations and formats, so that "fixed" settings ("bt/id", "bt/name", ...) can be stored in rom while dynamic settings ("bt/keys", ...) can be placed in flash or elsewhere.

Laczen, 2019-10-17:
@rakons, @fnde-ot, I agree with only one backend and a "backend splitter", this would put the complexity where it needs to be (in the "backend splitter") and would also allow the removal of the present limitation of only one destination backend, the splitter could provide support for multiple destinations based on the settings name (or the settings value...).

So I think we should go for the simplification of the settings module to remove multiple backends.

closed 2023-04-27 by Laczen