zephyrproject-rtos/zephyr #19802: Zephyr was unable to find the toolchain after update to zephyr version 1.13.0

nanjunneo, 2019-10-14:
**Describe the bug**

Since older version Zephyr 1.13.0 is needed for project, so I get the older version by git checkout xxx to get the version. 

-cd ~/build_folder
-cmake –Gninja /home/usr/zephyr_workspace/project_name

--------------------------------------------------------------------------------------
**Expected behavior**

The build shall succeed

--------------------------------------------------------------------------------------

**Screenshots or console output**

-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3.4") 
-- Selected BOARD stm32f4_disco_bb
Zephyr version: 1.13.0
Parsing Kconfig tree in /home/neo/zephyr/Kconfig
Loading /home/neo/zephyr_workspace/boards/arm/stm32f4_disco_bb/stm32f4_disco_bb_defconfig as base
Merging /home/neo/zephyr_workspace/bme680/prj.conf

warning: TEST_RANDOM_GENERATOR (defined at subsys/random/Kconfig:8) was assigned the value 'y' but
got the value 'n'. You can check symbol information (including dependencies) in the 'menuconfig'
interface (see the Application Development Primer section of the manual), or in the Kconfig
reference at http://docs.zephyrproject.org/reference/kconfig/CONFIG_TEST_RANDOM_GENERATOR.html
(which is updated regularly from the master branch). See the 'Setting configuration values' section
of the Board Porting Guide as well.
CMake Error at /home/neo/zephyr/cmake/compiler/gcc.cmake:18 (message):
  Zephyr was unable to find the toolchain.  Is the environment misconfigured?

  User-configuration:

  ZEPHYR_TOOLCHAIN_VARIANT: zephyr

  Internal variables:

  CROSS_COMPILE:
  /home/neo/zephyr-sdk-0.10.3/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-

  TOOLCHAIN_HOME: /home/neo/zephyr-sdk-0.10.3/sysroots/x86_64-pokysdk-linux

Call Stack (most recent call first):
  /home/neo/zephyr/cmake/toolchain.cmake:57 (include)
  /home/neo/zephyr/cmake/app/boilerplate.cmake:252 (include)
  CMakeLists.txt:6 (include)

**Environment (please complete the following information):**
 - OS: ( Linux 18.04)
 - Toolchain (Zephyr SDK, cmake)
Zephyr SDK  0.10.3
cmake version 3.13.3

After update the west 
west build -b reel_board samples/hello_world
 shows usage: west [-h] [-z ZEPHYR_BASE] [-v] {flash,debug,debugserver} ...
west: error: argument command: invalid choice: 'build' (choose from 'flash', 'debug', 'debugserver')  

does it has effect to cmake build?

thank you guys in advance!

andrewboie, 2019-10-14:
> Since older version Zephyr 1.13.0 is needed for project, so I get the older version by git checkout xxx to get the version.

1.13 is not currently supported, have you considered migrating to the 1.14 LTS release, which is supported for two years?

dleach02, 2019-10-14:
@nanjunneo, I suspect you will need to stick with the zephyr-sdk-0.9.3-setup.run toolchain that was released with 1.13.0 and documented in the 1.13.0 startup guide. As @andrewboie indicates, 1.14 LTS release will be supported for two years from release.

nanjunneo, 2019-10-16:
@andrewboie  Thank you very much!  The version is quite old. I will update it to the new version for further developing.

nanjunneo, 2019-10-16:
@dleach02 thank you so much.  I downloaded the SDK 0.9.3 and it works properly!!

closed 2019-10-16 by nanjunneo