zephyrproject-rtos/zephyr #19811: native_posix stack smashing

mfiumara, 2019-10-14:
**Describe the bug**
When compiling the sample bluetooth/mesh demo with target `native_posix`, the sample crashes when trying to provision it with the following message:
```
WARNING: Using a test - not safe - entropy source
[00:00:00.000,000] <inf> flash_native_posix: Device flash_ctrl initialized
***** Booting Zephyr OS build zephyr-v2.0.0-1058-g288940dbbda5 *****
Initializing...
[00:00:00.540,000] <inf> bt_hci_core: Identity: 60:57:18:3e:41:61 (public)
[00:00:00.540,000] <inf> bt_hci_core: HCI: version 4.2 (0x08) revision 0x1000, manufacturer 0x0002
[00:00:00.540,000] <inf> bt_hci_core: LMP: version 4.2 (0x08) subver 0x1000
Bluetooth initialized
[00:00:00.540,000] <dbg> bt_mesh_settings.mesh_commit: sub[0].net_idx 0xffff
[00:00:00.540,000] <inf> bt_mesh_main: Device UUID: 00000000-0000-0000-0000-00000000dddd
Mesh initialized
*** stack smashing detected ***: <unknown> terminated
```
I have not enough knowledge regarding the native_posix port so am unsure how to proceed.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b native_posix
2. sudo ./build/zephyr/zephyr.exe
3. Use (nRF Mesh App)[https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp] to scan for Mesh devices
4. Connect to `Zephyr` node, Identify and then Provision (either OOB or no OOB)
4. This results in stack smashing in the example app

**Expected behavior**
I expect the example application to go through the full provisioning procedure, as is the case on for example the `nrf52_pca10040` dev kit.

**Impact**
This is a showstopper for the `native_posix` port, at least for this specific example. It could have an impact on other examples as well.

**Environment (please complete the following information):**
 - OS: (Arch Linux)
 - Toolchain (Native Toolchain
 - Used v2.0-branch

jhedberg, 2019-10-15:
Could you try running the binary through valgrind? That might give some more info.

mfiumara, 2019-10-15:
@jhedberg I reran the same thing with valgrind: https://gist.github.com/mfiumara/ced58c214b4aa84917fb8f836ce05dd8

jhedberg, 2019-10-15:
@mfiumara thanks! This in particular caught my eye regarding the backtrace:
```
by 0x135CB5: prov_random (prov.c:1010)
```

There's an open issue and a not yet merged bug fix #19789 for the `PROV_BUF(rnd, 16);` buffer being one byte too small. Could you look up the `prov_random()` function in `subsys/bluetooth/mesh/prov.c` and change the rnd buffer to be 17 instead of 16 bytes, i.e. `PROV_BUF(rnd, 17);`, and see if the issue is still reproducible? Thanks.

mfiumara, 2019-10-15:
@jhedberg: I reran the test and indeed with the 1 byte bigger buffer and the sample application goes through the provisioning steps now and succesfully completes! There are however some other remarks by valgrind (which do not crash the application but maybe worth sharing):
```
[00:00:33.030,000] <dbg> bt_mesh_settings.store_pending: 
[00:00:33.030,000] <dbg> bt_mesh_settings.store_pending_rpl: 
[00:00:33.030,000] <dbg> bt_mesh_settings.store_rpl: src 0x0001 seq 0x000006 old_iv 0
==31821== Thread 5:
==31821== Use of uninitialised value of size 4
==31821==    at 0x113CF3: crc8_ccitt (crc8_sw.c:22)
==31821==    by 0x118331: fcb_elem_crc8 (fcb_elem_info.c:59)
==31821==    by 0x117E23: fcb_append_finish (fcb_append.c:126)
==31821==    by 0x11978D: settings_fcb_save_priv (settings_fcb.c:285)
==31821==    by 0x11978D: settings_fcb_save (settings_fcb.c:313)
==31821==    by 0x1189A3: settings_save_one (settings_store.c:85)
==31821==    by 0x131993: store_rpl (settings.c:984)
==31821==    by 0x131993: store_pending_rpl (settings.c:1029)
==31821==    by 0x131993: store_pending (settings.c:1351)
==31821==    by 0x114285: z_work_q_main (work_q.c:32)
==31821==    by 0x114224: z_thread_entry (thread_entry.c:29)
==31821==    by 0x11A06A: posix_thread_starter (posix_core.c:305)
==31821==    by 0x495C659: start_thread (in /usr/lib32/libpthread-2.30.so)
==31821==    by 0x4A7B919: clone (in /usr/lib32/libc-2.30.so)
```

closed 2019-10-15 by mfiumara