zephyrproject-rtos/zephyr #19813: tests/crypto/rand32 failed on sam_e70 board on v1.14 branch.

chen-png, 2019-10-15:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b=sam_e70_xplained tests/crypto/rand32 
              -DCONF_FILE=prj_hw_random_xoroshiro.conf
2. west flash
3. See error

**Screenshots or console output**

***** USAGE FAULT *****
  Illegal load of EXC_RETURN into PC
***** Hardware exception *****
Current thread ID = 0x00000000
Faulting instruction address = 0x48024001
Fatal fault in essential thread! Spinning...

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: v1.14-branch 9591025

ioannisg, 2019-10-16:
@chen-png could you please, confirm that you do not see this error on current master ?

chen-png, 2019-10-17:
> @chen-png could you please, confirm that you do not see this error on current master ?

Yes, i ran it with the master(commit: 5881f1), it worked well.
***** Booting Zephyr OS build zephyr-v2.0.0-1163-g5881f118c0a4 *****
Running test suite common_test

starting test - test_rand32
Generating random numbers
PASS - test_rand32

Test suite common_test succeeded

PROJECT EXECUTION SUCCESSFUL

ioannisg, 2019-10-17:
Thanks @chen-png 
Then it seems we need to back port the driver fix in #17199  to 1.14.x branch.
Do we want to do this @nashif @dleach02 ?

dleach02, 2019-10-22:
@ioannisg will cherry pick and merge to 1.14

nashif, 2020-03-18:
This was fixed by #22984

PR #22984 (2020-03-15): drivers: sam: entropy: implement get_entropy_isr
The commit contributes the implementation of
get_entropy_isr API function for the SAM entropy
driver. The implementation is similar to get_entropy,
with the difference that it does not invoke k_yield()
when called with the ENTROPY_BUSYWAIT options flag set.
When the function is invoked without the ENTROPY_BUSYWAIT
flag, it simply returns whatever data is available,
without busy waiting on the RNG herdware.

Manually back-porting #17199

This PR fixes #19813
closed 2020-03-18 by nashif