zephyrproject-rtos/zephyr #19848: stm32wb MPU failure

pavlohamov, 2019-10-16:
**Describe the bug**
Flash write operations leads to MPU failure 

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir samples/mpu/mpu_test/build; cd samples/mpu/mpu_test/build
2. cmake -GNinja -DBOARD=nucleo_wb55rg ../ && ninja flash
3. Open serial console
3. mpu run 
4. See exception trace

**Expected behavior**
MPU test must pass without any exception 

**Impact**
No file system could be created ins partition located on the device internal flash due to MPU exceptions

[image]

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 18.04
 - Zephyr
 - master e1ae80c038

pavlohamov, 2019-10-16:
@erwango I've updated description. FS sample is more convenient to troubleshoot this issue.

pavlohamov, 2019-10-17:
Update: ```

+++ b/scripts/kconfig/kconfiglib.py
@@ -1406,10 +1406,6 @@ class Kconfig(object):
                     .format(self.config_prefix, sym.name, escape(val)))
 
             else:  # sym.orig_type in _INT_HEX:
-                if sym.orig_type is HEX and \
-                   not val.startswith(("0x", "0X")):
-                    val = "0x" + val
-
                 add("#define {}{} {}\n"
                     .format(self.config_prefix, sym.name, val))
```
Will fix the issue with sizes

pavlohamov, 2019-10-17:
https://github.com/zephyrproject-rtos/zephyr/issues/19877

closed 2019-10-17 by pavlohamov

ulfalizer, 2019-10-17:
Potential fix: https://github.com/zephyrproject-rtos/zephyr/pull/19891

Note that I didn't write the `kconfigfunctions.py` functions. Just pattern-matching a bit here.