zephyrproject-rtos/zephyr #19850: Bluetooth: Mesh: Modular settings handling

trond-snekvik, 2019-10-16:
The mesh settings.c module manipulates the data of a lot of modules, and this pattern is causing some awkwardness in encapsulation, as experienced in #19799.

To improve encapsulation of module data, the mesh settings module could allow registering of handlers in the different mesh modules, and only take care of the multiplexing and orchestration itself. This'll improve maintainability and modularity, reducing the risk of missing functionality and impact of internal changes.

LingaoM, 2020-12-31:
I think we should solve this ancient problem. I agree with @trond-snekvik point of view, I think each module should manage its own data storage separately.

At present, there are 10 modules that need to be decoupled from setting. The list is as follows:

1. 	BT_MESH_RPL_PENDING
2. 	BT_MESH_KEYS_PENDING
3. 	BT_MESH_NET_PENDING
4. 	BT_MESH_IV_PENDING
5. 	BT_MESH_SEQ_PENDING
6. 	BT_MESH_HB_PUB_PENDING
7. 	BT_MESH_CFG_PENDING
8. 	BT_MESH_MOD_PENDING
9. 	BT_MESH_VA_PENDING
10.	BT_MESH_CDB

My current idea is to create a handler in settings, each module implements this interface separately, as show in below:

``` C
struct bt_mesh_settings {

	/* Settings module name */
	const char *name;

	/* Setting module store flag */
	int bit;
	atomic_t *flags;

	/* Setting module store callback */
	void (*store)(void);

	/* Setting module recovery callback */
	void (*recover)(const char *name, size_t len_rd,
		    settings_read_cb read_cb, void *cb_arg);
};
```
This will completely decouple, and the settings module will not take up additional RAM to store data.

trond-snekvik, 2021-01-05:
@LingaoM is this something you're currently working on? It's approaching the top of our task queue as well

LingaoM, 2021-01-05:
In fact, I have always wanted to do it, but my recent work has nothing to do with it

trond-snekvik, 2021-01-05:
Alright, I'll let you know if we start working on it.

commit 561a8e4f (2021-01-14): Bluetooth: Mesh: Break up mesh settings

The mesh settings.c module is a giant piece of code responsible for
storing the mesh stack configuration. Such approach makes it difficult
to control the data to be stored, breaks the stack modules'
encapsulation by forcing them to reveal the internal kitchen, which
leads to unpleasant issues such as #19799.

This commit moves the responsibility of storing the configuration
to corresponding modules while keeping control of the moment of storing
the configuration and of starting the stack after the settingss loading
is completed.

This doesn't introduce  [... cut]
closed 2021-01-14 by jhedberg