zephyrproject-rtos/zephyr #19851: How to use flash with varying erase block size

erwango, 2019-10-16:
**Is your enhancement proposal related to a problem? Please describe.**
On STM32F4 series, there is no fixed erase block size:
Erase operation are done per sector and sectors have varying block size.
As a consequence `DT_FLASH_ERASE_BLOCK_SIZE` is not available on these SoCs

Issue was already reported in #13328 (dts: erase-block-size should be available by partition
), which proposed to provide per partition erase block size.
It was finally resolved by https://github.com/zephyrproject-rtos/zephyr/issues/13369 (nvs sample should not use DT_FLASH_ERASE_BLOCK_SIZE).

It happens that `DT_FLASH_ERASE_BLOCK_SIZE` is now used in multiple places in code base, which could be understood, as OTOH it is convenient and OTOH how can people know it can't be used though it seems fairly generic.

**Describe the solution you'd like**
Allow flash related code that requires flash erase block size information to be used on STM32F4 despite this lack of fixed size available.
Maybe this can be done by adding a flash api that returns the erase block size for a given sector.
Anyway, chosen solution should allow to get rid of this issue so that it doesn't pops up every 6 months.

**Describe alternatives you've considered**

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

erwango, 2019-10-16:
@Laczen, fyi as I know you're curious of flash related issues and you were involved in previous episode.

ioannisg, 2019-10-16:
FYI @nvlsianpu

nvlsianpu, 2019-10-16:
> Allow flash related code that requires flash erase block size information to be used on STM32F4 despite this lack of fixed size available.
Maybe this can be done by adding a flash api that returns the erase block size for a given sector.

such API exist: at flash_map level`flash_area_get_sectors()` and lower in flash driver `flash_page_foreach()`,  `flash_get_page_count()`, `flash_get_page_info_by_idx()`, `flash_get_page_info_by_offs()`

That is right -should be used instead of direct reference to DT definition.

Other problem is that storage's system we have nowadays in zephyr requires regular layout of erasable block. Problem can by squashed by using a block size which is Least common multiple of HW flash pages size in a partition used for instance.

Laczen, 2019-10-16:
I agree with @nvlsianpu that flash page api should be used instead of DT definition. So the problem is the definition of the default nvs backend, I will have a look how this can be fixed.

erwango, 2019-10-16:
> I agree with @nvlsianpu that flash page api should be used instead of DT definition. So the problem is the definition of the default nvs backend, I will have a look how this can be fixed.

So I agree with you, but how do we ensure that people don't use `DT_FLASH_ERASE_BLOCK_SIZE` directly ?

nvlsianpu, 2019-10-16:
> I agree with @nvlsianpu that flash page api should be used instead of DT definition. So the problem is the definition of the default nvs backend, I will have a look how this can be fixed.

@Laczen Posted patch already
https://github.com/zephyrproject-rtos/zephyr/pull/19858

erwango, 2019-10-16:
> @Laczen Posted patch already #19858

That was fast, thanks for this.
But how do we ensure people don't use it again. 

So one point is that I can try to make sure f4 boards are tested in all flash related samples.. But doesn't prevent people to use it when convenient (For instance there are other occurence (cf subsys/net/lib/openthread/platform/openthread-core-zephyr-config.h).

nvlsianpu, 2019-10-16:
> So I agree with you, but how do we ensure that people don't use DT_FLASH_ERASE_BLOCK_SIZE directly ?

DT_FLASH_ERASE_BLOCK_SIZE is only valid for embedded flash. This might be used for compile time optimization for instance. I see nothing wrong in using it, just need to know when this is not good idea. Maybe this macro should be renamed to something like DT_SOC_FLASH_ERASE_BLOCK_SIZE or DT_EMBEDED_FLASH_ERASE_BLOCK_SIZE for emphasize to which flash it belongs.

erwango, 2019-10-16:
>  I see nothing wrong in using it

When it could not be defined (because it is not fixed for a defined embedded flash, cf stm32f4 case) it is a problem...

nvlsianpu, 2019-10-16:
What I just recognized is that flash API for fetching page layout is someway not documented at the zephyr website.

erwango, 2019-10-16:
Following discussion on slack storage channel, one proposal would be to remove 'erase-block-size' property which is misleading and prevent people to use dedicated flash API. To be discussed in dev-review meeting.

galak, 2019-10-24:
A few ideas related to limiting scope of DT defines:
1. split out 'label' and other connecting defines into a separate header
2. for DT_COMPAT (not sure if that's the correct prefix) only generate defines for 'label' and not other properties that would be driver specific.
3. have some testcase that utilizes a driver that doesn't use the DT defines
4. Can we deprecate DT_FLASH_ defines completely at this point?

Let's try to look at option 4 to start with.

galak, 2019-10-24:
any ideas about the usage here:
```
subsys/net/lib/openthread/platform/openthread-core-zephyr-config.h:#define SETTINGS_CONFIG_PAGE_SIZE          DT_FLASH_ERASE_BLOCK_SIZE
```

```
scripts/west_commands/sign.py:                                   ('DT_FLASH_WRITE_BLOCK_SIZE',
scripts/west_commands/sign.py:                    'DT_FLASH_WRITE_BLOCK_SIZE={} '
subsys/dfu/boot/mcuboot.c:#define FLASH_MIN_WRITE_SIZE DT_FLASH_WRITE_BLOCK_SIZE
subsys/dfu/img_util/flash_img.c:BUILD_ASSERT_MSG((CONFIG_IMG_BLOCK_BUF_SIZE % DT_FLASH_WRITE_BLOCK_SIZE == 0),
subsys/dfu/img_util/flash_img.c:                 "DT_FLASH_WRITE_BLOCK_SIZE");
tests/subsys/settings/fcb_init/src/settings_test_fcb_init.c:__aligned(DT_FLASH_WRITE_BLOCK_SIZE)
tests/subsys/settings/fcb_init/src/settings_test_fcb_init.c:u8_t prepared_mark[DT_FLASH_WRITE_BLOCK_SIZE] = {ERASED_VAL};
tests/subsys/settings/fcb_init/src/settings_test_fcb_init.c:    u8_t new_val[DT_FLASH_WRITE_BLOCK_SIZE];
tests/subsys/settings/fcb_init/src/settings_test_fcb_init.c:                         DT_FLASH_WRITE_BLOCK_SIZE);
```

nvlsianpu, 2019-10-24:
We can introduce a variable which will be initialize at some point using flash_map API instead of `DT_FLASH_ERASE_BLOCK_SIZE` @rlubos can you point where exactly do it?

d3zd3z, 2019-11-25:
One issue we are finding with MCUboot is that some devices are fairly large, and have small erase sectors (say 2MB, with 512 byte sectors). Currently, we build a data structure with one element for each sector. It would actually be handy for us to be able to detect devices that have sectors of the same size (or be configurable at compile time).

I suppose it would still be the correct answer, though, to just iterate through the flash API, and if configured for fixed-size sectors, just verify that the device indeed does report all of the sectors being the same size.