zephyrproject-rtos/zephyr #19852: Add support for GPIO AF remap on STM32F1XX

gmarull, 2019-10-16:
**Is your feature request related to a problem? Please describe.**

STM32F1XX family allows to remap alternate functions on some peripherals. This is done via the `AFIO` registers, a feature not implemented in STM32 GPIO or pinmux drivers.

The STM32 LL HAL has the necessary APIs to control remaps: https://github.com/zephyrproject-rtos/hal_stm32/blob/master/stm32cube/stm32f1xx/drivers/include/stm32f1xx_ll_gpio.h#L898

**Describe the solution you'd like**

- Have driver level support for remap functionality.
- Make it as transparent as possible for the user. For example, if he picks PC10/11 pins for USART3, remap should be automatically configured.

**Describe alternatives you've considered**

Add driver-level support for re-mapping. I'm not sure what is the best approach considering it is only a functionality present on F1XX devices, so I'd like to hear from suggestions before implementing a proper solution.

**Additional context**

I have found a custom board where USART3 is remapped to PC10/11, so changing DTS files was not enough to get the peripheral working. I added the following snipped to the board `pinmux.c` file for now to make it work:

```c
LL_APB2_GRP1_EnableClock(LL_APB2_GRP1_PERIPH_AFIO);
LL_GPIO_AF_RemapPartial_USART3();
```

carlescufi, 2020-03-24:
@gmarull are you planning to submit a PR for this?

gmarull, 2020-03-24:
@carlescufi I faced this problem on a test board which I'm not using anymore, so I do not have plans for a PR. Considering this is a feature present in some older low-end F1 microcontrollers I'm not sure it is worth the effort...

erwango, 2020-03-25:
@carlescufi @gmarull GPIO AF remap is indeed a quite particular way of handling alternate functions amongst the STM32 family. I don't think it has much value to treat this as a complete feature (lot of code, only a few SoCs impacted). So far it has been treated case by case, we can continue this way.

carlescufi, 2020-03-25:
OK, let´s close this then, thanks @erwango and @gmarull. Whenever there's need to implement this the interested party can send a PR.

closed 2020-03-25 by carlescufi