zephyrproject-rtos/zephyr #19867: modem: ublox-sara-r4/u2 build error

WilliamGFish, 2019-10-16:
**Describe the bug**
Modem ublox-sara-r4 driver will not compile.

**To Reproduce**
west build -b particle_boron

**Expected behavior**
Build and compile

**Impact**
Can not build and run with modem 

**Screenshots or console output**
```
drivers/modem/ublox-sara-r4.c:43:12: error: 'DT_INST_0_UBLOX_SARA_R4_MDM_POWER_GPIOS_CONTROLLER' undeclared here (not in a function); did you mean 'DT_INST_0_JEDEC_SPI_NOR_WP_GPIOS_CONTROLLER'?
  MODEM_PIN(DT_INST_0_UBLOX_SARA_R4_MDM_POWER_GPIOS_CONTROLLER,
```

**Additional context**
**commit a0fceff changed dts/bindings/modem/ublox,sara-r4.yaml** 
`compatible: "ubloc,sara-r4"`

needs to be reverted to:
`constraint: "ublox,sara-r4"`

carlescufi, 2019-10-16:
@WilliamGFish would you mind sending a PR?

mike-scott, 2019-10-16:
I suppose, I can also add a test build in the lwm2m sample to use the ublox shield -- to avoid this kind of breakage in the future.

PR #20029 (2019-10-22): dts: bindings: Fix u-blox misspelling
Spell ublox correctly.

Fixes #19867.
commit 485e22b4 (2019-10-22): dts: bindings: Fix u-blox misspelling

Spell ublox correctly.

Fixes #19867.
closed 2019-10-22 by galak