zephyrproject-rtos/zephyr #19882: Add support for multiple channel sampling to STM32 ADC driver

gmarull, 2019-10-17:
**Is your enhancement proposal related to a problem? Please describe.**

Multiple channels can't be sampled using a single `adc_read` call. For example,

```
const struct adc_sequence seq = {
    .channels = BIT(8) | BIT(9),
    .buffer = buffer,
    .buffer_size = sizeof(buffer),
    .resolution = 12,
};
```

will result in channel `8` being sampled after calling `adc_read(dev, &seq)`, but not channel `9`.

The reason for this behavior is that STM32 ADC driver does not support ADC "scan" mode. Only a single acquisition is configured: https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/adc/adc_stm32.c#L273

**Describe the solution you'd like**

Be able to have all channels in the sequence sampled with a single `adc_read` call.

**Describe alternatives you've considered**

Implement scan mode on the STM32 ADC driver. Note that a DMA is required for such mode, not sure how to make the DMA peripheral to be used configurable, Kconfig option?.

**Additional context**

N/A

erwango, 2019-10-17:
@cybertale, if you're interested: DMA + ADC ;-)

Oanerer, 2019-10-17:
Maybe there is an intermediate version possible, similar to what @anangl suggested to me here: https://github.com/zephyrproject-rtos/zephyr/pull/19652#discussion_r335399990 ? I also wanted to use the scan functionality of the Gecko ADC for multiple channels, but his suggestion works just as well.

cybertale, 2019-10-17:
@gmarull @erwango  This is on the to do list, but has to be done after the DMA driver. Here is some explanation: https://github.com/zephyrproject-rtos/zephyr/issues/17375#issuecomment-536519655
The main reason is that without DMA, zephyr is not fast enough to handle every EOC interrupt signals, which will end up with overrun errors. By the way, the DMA driver is very close to the end in #19712 .

gmarull, 2019-10-17:
@cybertale Thanks for the update! I can help with the implementation based on DMA support branch, I'd like to have it for a project I'm working on (F100R8). I have F3XX, F4XX and H7XX boards to perform testing.

cybertale, 2019-10-17:
@gmarull Thanks! drivers of f3 and f4 series SoCs have already been implemented, but I don't have a h7 series board, you can have a look into the driver implemented in #19711 and #19712 to add DMA dts support of h7 and driver implementation if you're interested. :wink: :

windelbouwman, 2021-01-21:
Interesting. At least the adc_read function should return some error code to indicate that this is not implemented.

gmarull, 2021-09-02:
Closing, there's a newer issue describing the same request. #32288

closed 2021-09-02 by gmarull