zephyrproject-rtos/zephyr #19885: SMP doesn't work on ARC any longer

abrodkin, 2019-10-17:
**Describe the bug**
SMP example don't progress.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nsim_hs_smp ../tests/kernel/smp/
3. make && make run
4. See nothing happens after initial application header get printed:
```
***** Booting Zephyr OS build zephyr-v2.0.0-1172-g55183494c2e8 *****
```

**Expected behavior**
```
***** Booting Zephyr OS build zephyr-v2.0.0-754-g6c283ca3d01f *****
Running test suite smp
===================================================================
starting test - test_smp_coop_threads
PASS - test_smp_coop_threads
===================================================================

...
```

**Impact**
IMHO showstopper for ARC.

With `git bisect` I found a commit that introduces the problem - https://github.com/zephyrproject-rtos/zephyr/commit/11bd67db537da97e7d807309af9bbc99e598d7a5.

andyross, 2019-10-17:
Does z_arch_sched_ipi() support work on ARC?  If you turn off CONFIG_SCHED_IPI_SUPPORTED with something like this, does it work?

```patch
--- a/arch/arc/Kconfig
+++ b/arch/arc/Kconfig
@@ -105,7 +105,6 @@ config      ARC_HAS_STACK_CHECKING
 
 config ARC_CONNECT
        bool "ARC has ARC connect"
-       select SCHED_IPI_SUPPORTED
        help
          ARC is configured with ARC CONNECT which is a hardware for connecting
          multi cores.
```

abrodkin, 2019-10-18:
Nope with removed `SCHED_IPI_SUPPORTED` I only see this:
```
***** Booting Zephyr OS build zephyr-v2.0.0-1172-g55183494c2e8 *****
Running test suite smp
===================================================================
starting test - test_smp_coop_threads
```
And no progress further. That said it's more than with `SCHED_IPI_SUPPORTED` but still pretty far from the end :)

@vonhust seems to have something here https://github.com/foss-for-synopsys-dwc-arc-processors/zephyr/tree/topic-fixes-smp which allows me to get to the end of SMP test suite, he'll be publishing it soon as a pull-request.

abrodkin, 2019-10-23:
@vonhust any chance to publish your fixes for SMP anytime soon?

ruuddw, 2019-11-13:
Added the v2.1 milestone to put the PR (ready for merging) on the radar...

PR #20082 (2019-11-13): arch: arc: necessary fixes for smp
After  recent changes in Zephyr 's SMP high-level code and tests cases,  the arc smp doesn't work and failed in the tests related to SMP and multi-core.

This PR fixes all the failures.

Fixes #19885 
Fixes #19599
closed 2019-11-13 by andrewboie