zephyrproject-rtos/zephyr #19899: List of never-enabled Kconfig symbols

ulfalizer, 2019-10-17:
Wrote a script that lists all symbols that are not user-configurable (have no prompt), have no defaults, and that aren't selected/implied. Such symbols can never be anything but `n`.

Could go through it and check if you see something that can be removed.

```
HAS_IMX_RDC
HAS_IMX_CCM
USE_STDC_A3G4250D
USE_STDC_AIS2DW12
USE_STDC_AIS328DQ
USE_STDC_AIS3624DQ
USE_STDC_H3LIS331DL
USE_STDC_HTS221
USE_STDC_I3G4250D
USE_STDC_IIS2DH
USE_STDC_IIS2DLPC
USE_STDC_IIS2MDC
USE_STDC_IIS328DQ
USE_STDC_IIS3DWB
USE_STDC_ISM303DAC
USE_STDC_ISM330DHCX
USE_STDC_ISM330DLC
USE_STDC_L20G20IS
USE_STDC_L3GD20H
USE_STDC_LIS2DE12
USE_STDC_LIS2DH12
USE_STDC_LIS2DS12
USE_STDC_LIS2DTW12
USE_STDC_LIS2HH12
USE_STDC_LIS331DLH
USE_STDC_LIS3DE
USE_STDC_LIS3DHH
USE_STDC_LIS3DH
USE_STDC_LIS3MDL
USE_STDC_LPS22HB
USE_STDC_LPS25HB
USE_STDC_LPS27HHW
USE_STDC_LPS33HW
USE_STDC_LPS33W
USE_STDC_LSM303AGR
USE_STDC_LSM303AH
USE_STDC_LSM6DS3
USE_STDC_LSM6DS3TR
USE_STDC_LSM6DSL
USE_STDC_LSM6DSM
USE_STDC_LSM6DSOX
USE_STDC_LSM6DSR
USE_STDC_LSM9DS1
USE_STM32_HAL_ADC
USE_STM32_HAL_ADC_EX
USE_STM32_HAL_CAN
USE_STM32_HAL_CEC
USE_STM32_HAL_COMP
USE_STM32_HAL_CRC
USE_STM32_HAL_CRC_EX
USE_STM32_HAL_CRYPT
USE_STM32_HAL_CRYPT_EX
USE_STM32_HAL_DAC
USE_STM32_HAL_DAC_EX
USE_STM32_HAL_DCMI
USE_STM32_HAL_DCMI_EX
USE_STM32_HAL_DFSDM
USE_STM32_HAL_DFSDM_EX
USE_STM32_HAL_DMA
USE_STM32_HAL_DMA2D
USE_STM32_HAL_DMA_EX
USE_STM32_HAL_DSI
USE_STM32_HAL_EXTI
USE_STM32_HAL_FDCAN
USE_STM32_HAL_FIREWALL
USE_STM32_HAL_FLASH
USE_STM32_HAL_FLASH_EX
USE_STM32_HAL_FLASH_RAMFUNC
USE_STM32_HAL_FMPI2C
USE_STM32_HAL_FMPI2C_EX
USE_STM32_HAL_GFXMMU
USE_STM32_HAL_GPIO
USE_STM32_HAL_GPIO_EX
USE_STM32_HAL_HASH
USE_STM32_HAL_HASH_EX
USE_STM32_HAL_HCD
USE_STM32_HAL_HRTIM
USE_STM32_HAL_HSEM
USE_STM32_HAL_I2C
USE_STM32_HAL_I2C_EX
USE_STM32_HAL_I2S
USE_STM32_HAL_I2S_EX
USE_STM32_HAL_IPCC
USE_STM32_HAL_IRDA
USE_STM32_HAL_IWDG
USE_STM32_HAL_JPEG
USE_STM32_HAL_LCD
USE_STM32_HAL_LPTIM
USE_STM32_HAL_LTDC
USE_STM32_HAL_LTDC_EX
USE_STM32_HAL_MDIOS
USE_STM32_HAL_MDMA
USE_STM32_HAL_MMC
USE_STM32_HAL_MMC_EX
USE_STM32_HAL_NAND
USE_STM32_HAL_NOR
USE_STM32_HAL_OPAMP
USE_STM32_HAL_OPAMP_EX
USE_STM32_HAL_OSPI
USE_STM32_HAL_PCCARD
USE_STM32_HAL_PWR
USE_STM32_HAL_PWR_EX
USE_STM32_HAL_QSPI
USE_STM32_HAL_RAMECC
USE_STM32_HAL_RCC
USE_STM32_HAL_RNG
USE_STM32_HAL_RTC
USE_STM32_HAL_RTC_EX
USE_STM32_HAL_SAI
USE_STM32_HAL_SAI_EX
USE_STM32_HAL_SD
USE_STM32_HAL_SD_EX
USE_STM32_HAL_SDADC
USE_STM32_HAL_SDRAM
USE_STM32_HAL_SMARTCARD
USE_STM32_HAL_SMARTCARD_EX
USE_STM32_HAL_SMBUS
USE_STM32_HAL_SPDIFRX
USE_STM32_HAL_SPI
USE_STM32_HAL_SPI_EX
USE_STM32_HAL_SRAM
USE_STM32_HAL_SWPMI
USE_STM32_HAL_TIM_EX
USE_STM32_HAL_TSC
USE_STM32_HAL_UART
USE_STM32_HAL_UART_EX
USE_STM32_HAL_USART
USE_STM32_HAL_USART_EX
USE_STM32_HAL_WWDG
USE_STM32_LL_ADC
USE_STM32_LL_BDMA
USE_STM32_LL_COMP
USE_STM32_LL_CRC
USE_STM32_LL_CRS
USE_STM32_LL_DAC
USE_STM32_LL_DELAYBLOCK
USE_STM32_LL_DMA
USE_STM32_LL_DMA2D
USE_STM32_LL_FMC
USE_STM32_LL_FSMC
USE_STM32_LL_GPIO
USE_STM32_LL_HRTIM
USE_STM32_LL_LPTIM
USE_STM32_LL_LPUART
USE_STM32_LL_MDMA
USE_STM32_LL_OPAMP
USE_STM32_LL_SDMMC
USE_STM32_LL_SWPMI
USE_STM32_LL_TIM
USE_STM32_LL_USART
BOARD_DEPRECATED
SOC_PART_NUMBER_MCIMX6X1EVK10AB
SOC_PART_NUMBER_MCIMX6X1EVK10AC
SOC_PART_NUMBER_MCIMX6X3EVK10AB
SOC_PART_NUMBER_MCIMX6X3EVK10AC
SOC_PART_NUMBER_MCIMX6X1EVO10AB
SOC_PART_NUMBER_MCIMX6X1EVO10AC
SOC_PART_NUMBER_MCIMX6X3EVO10AB
SOC_PART_NUMBER_MCIMX6X3EVO10AC
SOC_PART_NUMBER_MCIMX6X2EVN10AB
SOC_PART_NUMBER_MCIMX6X2EVN10AC
SOC_PART_NUMBER_MCIMX6X3EVN10AB
SOC_PART_NUMBER_MCIMX6X3EVN10AC
SOC_PART_NUMBER_MCIMX6X4EVM10AC
SOC_PART_NUMBER_MCIMX7D7DVM10SC
SOC_PART_NUMBER_MIMXRT1015CAF4A
SOC_PART_NUMBER_MIMXRT1021CAF4A
SOC_PART_NUMBER_MIMXRT1021CAG4A
SOC_PART_NUMBER_MIMXRT1021DAF5A
SOC_PART_NUMBER_MIMXRT1051CVL5A
SOC_PART_NUMBER_MIMXRT1051DVL6A
SOC_PART_NUMBER_MIMXRT1052CVJ5B
SOC_PART_NUMBER_MIMXRT1052CVL5A
SOC_PART_NUMBER_MIMXRT1052CVL5B
SOC_PART_NUMBER_MIMXRT1052DVJ6B
SOC_PART_NUMBER_MIMXRT1052DVL6B
SOC_PART_NUMBER_MIMXRT1061CVL5A
SOC_PART_NUMBER_MIMXRT1061DVL6A
SOC_PART_NUMBER_MIMXRT1062CVL5A
SOC_PART_NUMBER_MIMXRT1064CVL5A
SOC_PART_NUMBER_MK22FX512AVLK12
SOC_PART_NUMBER_MK64FN1M0CAJ12
SOC_PART_NUMBER_MK64FN1M0VLQ12
SOC_PART_NUMBER_MK64FN1M0VMD12
SOC_PART_NUMBER_MK64FX512VDC12
SOC_PART_NUMBER_MK64FX512VLL12
SOC_PART_NUMBER_MK64FX512VLQ12
SOC_PART_NUMBER_MK64FX512VMD12
SOC_PART_NUMBER_MK80FN256VDC15
SOC_PART_NUMBER_MK80FN256VLL15
SOC_PART_NUMBER_MK82FN256VDC15
SOC_PART_NUMBER_MKE14F256VLH16
SOC_PART_NUMBER_MKE14F256VLL16
SOC_PART_NUMBER_MKE14F512VLH16
SOC_PART_NUMBER_MKE14F512VLL16
SOC_PART_NUMBER_MKE16F256VLH16
SOC_PART_NUMBER_MKE16F256VLL16
SOC_PART_NUMBER_MKE16F512VLH16
SOC_PART_NUMBER_MKE16F512VLL16
SOC_PART_NUMBER_MKE18F256VLH16
SOC_PART_NUMBER_MKE18F256VLL16
SOC_PART_NUMBER_MKE18F512VLH16
SOC_PART_NUMBER_MKL25Z32VFM4
SOC_PART_NUMBER_MKL25Z64VFM4
SOC_PART_NUMBER_MKL25Z128VFM4
SOC_PART_NUMBER_MKL25Z32VFT4
SOC_PART_NUMBER_MKL25Z64VFT4
SOC_PART_NUMBER_MKL25Z128VFT4
SOC_PART_NUMBER_MKL25Z32VLH4
SOC_PART_NUMBER_MKL25Z64VLH4
SOC_PART_NUMBER_MKL25Z128VLH4
SOC_PART_NUMBER_MKL25Z32VLK4
SOC_PART_NUMBER_MKL25Z64VLK4
SOC_PART_NUMBER_MKW22D512VHA5
SOC_PART_NUMBER_MKW41Z256VHT4
SOC_PART_NUMBER_LPC55S69JET98
SOC_GECKO_CORE
SOC_GECKO_CRYOTIMER
SOC_GECKO_LETIMER
SOC_GECKO_PRS
SOC_GECKO_RMU
SOC_GECKO_RTC
SOC_GECKO_TIMER
SOC_GECKO_WDOG
SOC_NOINIT_LD
SOC_RODATA_LD
SOC_RWDATA_LD
CPU_CORTEX_M23
CPU_CORTEX_R4
X86_NO_SPECULATIVE_VULNERABILITIES
BIG_ENDIAN
HAS_SYS_POWER_STATE_DEEP_SLEEP_2
HAS_SYS_POWER_STATE_DEEP_SLEEP_3
BOOTLOADER_CONTEXT_RESTORE_SUPPORTED
ATOMIC_OPERATIONS_CUSTOM
SPI_DW_PORT_0_CLOCK_GATE_DRV_NAME
SPI_DW_PORT_1_CLOCK_GATE_DRV_NAME
HAS_WDT_MULTISTAGE
CLOCK_CONTROL_NRF_FORCE_ALT
FLASH_NRF_FORCE_ALT
NET_ROUTING
NET_ROUTE_MCAST
NET_HEADERS_ALWAYS_CONTIGUOUS
```

aescolar, 2019-10-21:
@ulfalizer note that BIG_ENDIAN is for off-tree architectures which are bigendian.

ulfalizer, 2019-10-22:
> @ulfalizer note that BIG_ENDIAN is for off-tree architectures which are bigendian.

Thanks, removed that one.

Suspect there's other "false positives" in there too. This is unrelated to the all the unused symbol PRs btw.

Oanerer, 2019-11-28:
PR [#20316](https://github.com/zephyrproject-rtos/zephyr/pull/20316) depends on symbol `SOC_GECKO_WDOG`

erwango, 2019-11-28:
@ulfalizer: `USE_STDC_FOO` and `USE_STM32_BAR` should be kept.

nashif, 2024-05-18:
the info here is most likely outdated and with scripts available to update this 5 years later, only option is to close this and look back if there is an update.

closed 2024-05-18 by nashif