zephyrproject-rtos/zephyr #19916: Limit non-trivial void pointer casting

rakons, 2019-10-18:
## Introduction

This is an example of bad habit pointer casting:
```
static int gatt_read_type(struct bt_conn *conn,
			  struct bt_gatt_discover_params *params)
{
   ...
   // - note: params is casted from bt_gatt_discover_params to void pointer.
   return gatt_send(conn, buf, gatt_read_type_rsp, params, NULL);
}
```

That is used inside gatt_send:
```
static int gatt_send(struct bt_conn *conn, struct net_buf *buf,
		     bt_att_func_t func, void *params,
		     bt_att_destroy_t destroy)
{
	int err;

	if (params) {
		// params is expected to be bt_att_req structure type
		struct bt_att_req *req = params;
...
```

1. There is no real reason for the `param` to be `void` pointer type.
2. we are casting between different structures only relaying on the fact that very first field inside `bt_gatt_discover_params` structure is bt_att_req structure:

```
struct bt_gatt_discover_params {
	struct bt_att_req _req;
	/** Discover UUID type */
	...
};
```

It would work as long as nobody changes bt_gatt_discover_params - then it might silently fail.
We should always get to the point of interest. Even if we leave the void pointer (witch seems pointless), we should call gatt_send like below:

```
return gatt_send(conn, buf, gatt_read_type_rsp, &params->_req, NULL);
```

This would increase code roboustness and make the code easier to analyse.

It is not the first time I have spotted the casting like that - it is just the first time I decided to describe the problem.

### Problem description

Pointer casting is always risky. And used without the additional tests may lead to errors in the future that will be very hard to locate.

### Proposed change

Personally I am used to even add static assertion next to the pointer casting that checks if the key data fields are on expected positions. We should limit them as much as possible. Whenever possible do not expect specific data layout, and if it is expected - always check it using build time static assertions.

## Detailed RFC

The description above seems to explain everything.

### Proposed change (Detailed)

1. Limit the void pointer usage to the places where it is absolutely required (probably callbacks and contexes).
2. Requirement to add static assertion if the specific alignment between different structures is required - this would automatically check the code but will also serve as an documentation.

### Dependencies

The changes proposed here is kind of "good" habit. It would not solve all the problems at once. It would not affect all the modules. The changes may be provided slowly into the modules, even one void pointer casting and dereference fix would increase the code readability.

### Concerns and Unresolved Questions

TBD

## Alternatives

Hard to really say if there are any good alternatives. We cannot forbid sometimes "strange" pointer casting in embedded world. But we should limit it and protect ourself from the code errors that would not be detected during compilation stage.

nvlsianpu, 2019-10-18:
> 2. requirement to add static assertion if the specific alignment between different structures is required - this would automatically check the code but will also serve as an documentation.

For meeting that we would invest lot of efforts from people who are experts in each area.

I would say that what is wrote above should land in the codding standard.

Generally I like the idea - such non-trivial pointer casting usually generates ugly regression bugs, which are impossible to be spotted in code review and also occurred in not modified code.

jhedberg, 2019-10-18:
Changing to use `&params->_req` and using the exact pointer type sounds good to me. What exactly static assertions would look like is a bit unclear to me.

rakons, 2019-10-18:
Just for example, very simple structure layout checking:
```
BUILD_ASSERT(offsetof(struct bt_gatt_discover_params, _req) == 0);
```

closed 2024-12-16 by jhedberg