zephyrproject-rtos/zephyr #19917: Bluetooth: Controller: Missing LL_ENC_RSP after HCI LTK Negative Reply

rugeGerritsen, 2019-10-18:
**Describe the bug**
In the Encryption Start Procedure, for the case when the host does not provide an LTK, the controller is supposed to send an LL_ENC_RSP before sending LL_REJECT_IND or LL_REJECT_EXT_IND. 

**To Reproduce**
1. Modify hci_core.c, `le_ltk_request()` to always call `le_ltk_neg_reply()`:
```
//	if (bt_smp_request_ltk(conn, evt->rand, evt->ediv, ltk)) {
//		le_ltk_reply(handle, ltk);
//	} else {
		le_ltk_neg_reply(handle);
//	}
```
2. Build the peripheral_hr sample
3. Connect and try bonding, expect it to fail.
4. Observe the sniffer log.

**Expected behavior**
Send LL_ENC_RSP before LL_REJECT_IND. See the MSC in Bluetooth Core Specification v5.1, Vol 6, Part D, Section 6.7, or Part B Section 5.1.3.1.
[image]

_If encryption is not supported by the Link Layer of the slave, the Link Layer of
the slave shall send an LL_REJECT_IND or LL_REJECT_EXT_IND PDU with
the ErrorCode set to Unsupported Remote Feature / Unsupported LMP
Feature (0x1A)._ 
…
_The Link Layer of the slave shall then send an LL_ENC_RSP PDU. The Link
Layer of the slave shall then notify the Host with the Rand and EDIV fields._

**Impact**
Possible showstopper. Peer devices may cache that the controller does not support encryption, thereby leaving the link unencrypted.

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
f79fbac101f8febb459acee577e9f6b76e00acc0

rugeGerritsen, 2019-10-18:
@cvinayak, FYI

carlescufi, 2019-10-18:
@rugeGerritsen Please specify which controller you are using, legacy or split

rugeGerritsen, 2019-10-18:
Updated the bug description to state that this bug is present in 2.0.0, both the legacy and split controller

cvinayak, 2019-10-18:
@rugeGerritsen Please try with CONFIG_BT_CTLR_FAST_ENC=y

carlescufi, 2019-10-18:
Thanks @rugeGerritsen, this is indeed a bug. It will only be fixed in the split controller.

rugeGerritsen, 2019-10-18:
> @rugeGerritsen Please try with CONFIG_BT_CTLR_FAST_ENC=y

The bug is not present when CONFIG_BT_CTLR_FAST_ENC=y. This is only enabled in the hci_uart example

PR #20378 (2019-11-08): Bluetooth: controller: split: Fix the slow Enc Setup alternative
Fix the implementation of slow encryption setup design
alternative to send ENC_RSP PDU before sending REJECT_IND
or REJECT_EXT_IND PDU.

Fixes #19917.
commit 48e5d948 (2019-11-08): Bluetooth: controller: split: Fix the slow Enc Setup alternative

Fix the implementation of slow encryption setup design
alternative to send ENC_RSP PDU before sending REJECT_IND
or REJECT_EXT_IND PDU.

Fixes #19917.
PR #19986 (2019-11-19): Bluetooth: controller: Fix the slow Enc Setup alternative
Fix the implementation of slow encryption setup design
alternative to send ENC_RSP PDU before sending REJECT_IND
or REJECT_EXT_IND PDU.

Fixes #19917.
commit 38052900 (2019-11-19): Bluetooth: controller: legacy: Fix the slow Enc Setup alternative

Fix the implementation of slow encryption setup design
alternative to send ENC_RSP PDU before sending REJECT_IND
or REJECT_EXT_IND PDU.

Fixes #19917.
closed 2019-11-08 by carlescufi