zephyrproject-rtos/zephyr #19969: [TOPIC-GPIO] mcux driver problems with pull configuration

pabigot, 2019-10-20:
`gpio_basic_api` currently passes on the topic branch, but #19553 has not yet been merged.  What it does is make sure that `gpio_pin_interrupt_configure()` is not invoked from `gpio_pin_configure()` unless the flags include either enabling or disabling interrupts.

If `gpio_pin_interrupt_configure()` is not invoked from `gpio_pin_configure()` `gpio_mcux` starts to fail `gpio_basic_api` because in some way the pull flags are not being set properly:
```
NOTE: cannot configure pin as disconnected; trying as input

    Assertion failed at ../src/test_gpio_port.c:389: check_pulls: (raw_in() not 
equal to false)
physical pull-down does not read low
FAIL - test_gpio_port
```

This can be demonstrated in the current topic branch by removing the pass-through to `gpio_pin_interrupt_configure()` and running the test (which is not supposed to depend on any legacy processing of interrupt-related flags in `gpio_pin_configure()`.

#19847 does not fix this problem.

mnkp, 2019-10-20:
@pabigot Could you introduce a delay between the call to `gpio_pin_configure` and the moment you attempt to read out pin input level. The pull up resistors have typically large values, if there is any capacitance attached to the pin it will take a moment for the value to change.

pabigot, 2019-10-20:
Thanks; I've verified that the problem does disappear with a delay before reading the signal.

pabigot, 2019-10-22:
Should be resolved.

PR #19971 (2019-10-22): [TOPIC-GPIO] tests/drivers/gpio_basic_api: delay before checking pull effect
When checking the behavior of pull configurations delay long enough for the signal to stabilize.  Checking without a sufficient delay may indicate a false failure.

Closes #19969
closed 2019-10-22 by pabigot