zephyrproject-rtos/zephyr #20007: tests/net/mld failed on mimxrt1050_evk board.

chen-png, 2019-10-22:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b=mimxrt1050_evk -p=auto tests/net/mld
2. west flash
3. See error

**Screenshots or console output**
E: ***** USAGE FAULT *****
E:   Unaligned memory access
E: r0/a1:  0x00000000  r1/a2:  0x6000e041  r2/a3:  0x00000000
E: r3/a4:  0xf88899e5 r12/ip:  0x00000000 r14/lr:  0x60003feb
E:  xpsr:  0x41000000
E: Faulting instruction address (r15/pc): 0x60006c0a
E: >>> ZEPHYR FATAL ERROR 0: CPU exception
E: Current thread: 0x8000152c (sysworkq)
E: Halting system

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: f0fa7b850c

jukkar, 2019-10-22:
I have this board but I cannot flash it in order to try to see what is going on with the test. I will create separate issue for the flashing problems.

jukkar, 2019-10-22:
Update: I have two mimxrt1050_evk boards. The other one is using OpenSDA J-Link Onboard Debug Probe and that I cannot flash with "make flash" command. The other board is in "usb mass storage" mode and that I managed to flash by copying the zephyr.bin to it (flashing the device this way is quite cumbersome). Anyway, the mld test program does not crash in my test so I need to figure out what is going on.

jukkar, 2019-10-22:
Just tried this with latest upstream commit 1f97946358f048ca7201d27b298b234cb29af3ab and I am not seeing the issue. Flashed the device with commit f0fa7b8 and the issue is present. I am assuming here that the issue has been already fixed so closing this one. Please try with later commit and re-open if you see the issue again.

closed 2019-10-22 by jukkar

chen-png, 2019-10-23:
>  Please try with later commit and re-open if you see the issue again.

yes, for the new commit, this issue isn't present.