zephyrproject-rtos/zephyr #20010: Cannot flash mimxrt1050_evk board

jukkar, 2019-10-22:
I have a imxrt1050-evkb board but I cannot flash it. I have followed the instructions at https://docs.zephyrproject.org/latest/boards/arm/mimxrt1050_evk/doc/index.html#option-1-opensda-jlink-onboard-debug-probe-recommended but no luck. This happens with Fedora 30.

Is there anything I am doing wrong here or any hints how to solve this issue.

```
$ cmake -Bbuild -DBOARD=mimxrt1050_evk .
...
$ make -C build flash
...
[100%] Flashing mimxrt1050_evk
-- west flash: using runner jlink
-- runners.jlink: Flashing file: zephyr/tests/net/mld/build/zephyr/zephyr.bin
SEGGER J-Link Commander V6.46g (Compiled Jun 14 2019 19:36:40)
DLL version V6.46g, compiled Jun 14 2019 19:36:30

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Firmware: J-Link OpenSDA 2 compiled May 27 2019 11:01:03
Hardware version: V1.00
S/N: 621000000
VTref=3.300V
Target connection not established yet but required for command.
Device "MCIMXRT1052" selected.

Connecting to target via SWD
Found SW-DP with ID 0x0BD11477
Scanning AP map to find all available APs
AP[1]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x04770041)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FD000
CPUID register: 0x411FC271. Implementer code: 0x41 (ARM)
Found Cortex-M7 r1p1, Little endian.
FPUnit: 8 code (BP) slots and 0 literal slots
CoreSight components:
ROMTbl[0] @ E00FD000
ROMTbl[0][0]: E00FE000, CID: B105100D, PID: 000BB4C8 ROM Table
ROMTbl[1] @ E00FE000
ROMTbl[1][0]: E00FF000, CID: B105100D, PID: 000BB4C7 ROM Table
ROMTbl[2] @ E00FF000
ROMTbl[2][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS-M7
ROMTbl[2][1]: E0001000, CID: B105E00D, PID: 000BB002 DWT
ROMTbl[2][2]: E0002000, CID: B105E00D, PID: 000BB00E FPB-M7
ROMTbl[2][3]: E0000000, CID: B105E00D, PID: 000BB001 ITM
ROMTbl[1][1]: E0041000, CID: B105900D, PID: 001BB975 ETM-M7
ROMTbl[1][2]: E0042000, CID: B105900D, PID: 004BB906 CTI
ROMTbl[0][1]: E0040000, CID: B105900D, PID: 000BB9A9 TPIU-M7
ROMTbl[0][2]: E0043000, CID: B105F00D, PID: 001BB101 TSG
Cache: Separate I- and D-cache.
I-Cache L1: 32 KB, 512 Sets, 32 Bytes/Line, 2-Way
D-Cache L1: 32 KB, 256 Sets, 32 Bytes/Line, 4-Way
Cortex-M7 identified.
Reset delay: 0 ms
Reset type NORMAL: Resets core & peripherals via SYSRESETREQ & VECTRESET bit.
ResetTarget() start

 Performing Core Reset over SRC
Core did not halt after reset, manually halting CPU...

 Performing reset over JTAG/SWD AP

J-Link: Target is halted.
 
J-Link: Reset 

 Performing Reset over reset line.
ResetTarget() end

Downloading file [zephyr/tests/net/mld/build/zephyr/zephyr.bin]...

**************************
WARNING: CPU could not be halted
**************************

****** Error: Failed to prepare for programming.
Failed to download RAMCode!
Can not read register 16 (XPSR) while CPU is running
Can not read register 20 (CFBP) while CPU is running
Can not read register 0 (R0) while CPU is running
Can not read register 1 (R1) while CPU is running
Can not read register 2 (R2) while CPU is running
Can not read register 3 (R3) while CPU is running
Can not read register 4 (R4) while CPU is running
Can not read register 5 (R5) while CPU is running
Can not read register 6 (R6) while CPU is running
Can not read register 7 (R7) while CPU is running
Can not read register 8 (R8) while CPU is running
Can not read register 9 (R9) while CPU is running
Can not read register 10 (R10) while CPU is running
Can not read register 11 (R11) while CPU is running
Can not read register 12 (R12) while CPU is running
Can not read register 14 (R14) while CPU is running
Can not read register 15 (R15) while CPU is running
Can not read register 17 (MSP) while CPU is running
Can not read register 18 (PSP) while CPU i
Unspecified error -1

****** Error: CPU is not halted

Script processing completed.
```

agansari, 2019-10-22:
Hello @jukkar from the information listed above, i think your core is stuck with some bad code and can't be flashed.

Do the following:
 - unplug the board
 - move dip switch from 0110 -> 1110 (this will prevent CPU booting from external flash)
 - use J-Link commander (JLinkExe command on Linux) and run the following commands
         - connect
         - MIMXRT1052CVL5B
         - S (SWD)
         - h (CPU should halt now, this is just to check the dip switch)
         - exec EnableEraseAllFlashBanks
         - erase
 - all flash areas should be clean now
 - switch back the dip switch to 0110 and flash via west/make/ninja

jukkar, 2019-10-22:
Hi @agansari done that
```
$ JLinkExe 
SEGGER J-Link Commander V6.46g (Compiled Jun 14 2019 19:36:40)
DLL version V6.46g, compiled Jun 14 2019 19:36:30

Connecting to J-Link via USB...O.K.
Firmware: J-Link OpenSDA 2 compiled May 27 2019 11:01:03
Hardware version: V1.00
S/N: 621000000
VTref=3.300V

Type "connect" to establish a target connection, '?' for help
J-Link>connect
Please specify device / core. <Default>: MCIMXRT1052
Type '?' for selection dialog
Device>MCIMXRT1052
Please specify target interface:
  J) JTAG (Default)
  S) SWD
  T) cJTAG
TIF>s
Specify target interface speed [kHz]. <Default>: 4000 kHz
Speed>
Device "MCIMXRT1052" selected.

Connecting to target via SWD
Found SW-DP with ID 0x0BD11477
Scanning AP map to find all available APs
AP[1]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x04770041)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FD000
CPUID register: 0x411FC271. Implementer code: 0x41 (ARM)
Found Cortex-M7 r1p1, Little endian.
FPUnit: 8 code (BP) slots and 0 literal slots
CoreSight components:
ROMTbl[0] @ E00FD000
ROMTbl[0][0]: E00FE000, CID: B105100D, PID: 000BB4C8 ROM Table
ROMTbl[1] @ E00FE000
ROMTbl[1][0]: E00FF000, CID: B105100D, PID: 000BB4C7 ROM Table
ROMTbl[2] @ E00FF000
ROMTbl[2][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS-M7
ROMTbl[2][1]: E0001000, CID: B105E00D, PID: 000BB002 DWT
ROMTbl[2][2]: E0002000, CID: B105E00D, PID: 000BB00E FPB-M7
ROMTbl[2][3]: E0000000, CID: B105E00D, PID: 000BB001 ITM
ROMTbl[1][1]: E0041000, CID: B105900D, PID: 001BB975 ETM-M7
ROMTbl[1][2]: E0042000, CID: B105900D, PID: 004BB906 CTI
ROMTbl[0][1]: E0040000, CID: B105900D, PID: 000BB9A9 TPIU-M7
ROMTbl[0][2]: E0043000, CID: B105F00D, PID: 001BB101 TSG
Cache: Separate I- and D-cache.
I-Cache L1: 32 KB, 512 Sets, 32 Bytes/Line, 2-Way
D-Cache L1: 32 KB, 256 Sets, 32 Bytes/Line, 4-Way
Cortex-M7 identified.
J-Link>halt
PC = 00215F26, CycleCnt = 87131ABB
R0 = 00000000, R1 = 00000001, R2 = 00000001, R3 = 202052C8
R4 = 20200F38, R5 = 20200F60, R6 = 20205179, R7 = CADF7C50
R8 = 00000002, R9 = 00000000, R10= CADF7C50, R11= 00000000
R12= 00000004
SP(R13)= 20200F30, MSP= 20200F30, PSP= 00000000, R14(LR) = 0021394F
XPSR = 61000000: APSR = nZCvq, EPSR = 01000000, IPSR = 000 (NoException)
CFBP = 00000000, CONTROL = 00, FAULTMASK = 00, BASEPRI = 00, PRIMASK = 00

FPS0 = 00000000, FPS1 = 00000000, FPS2 = 00000000, FPS3 = 00000000
FPS4 = 00000000, FPS5 = 00000000, FPS6 = 00000000, FPS7 = 00000000
FPS8 = 00000000, FPS9 = 00000000, FPS10= 00000000, FPS11= 00000000
FPS12= 00000000, FPS13= 00000000, FPS14= 00000000, FPS15= FFFFFFFF
FPS16= 00000000, FPS17= 00000000, FPS18= 00000000, FPS19= 00000000
FPS20= 00000000, FPS21= 00000000, FPS22= 00000000, FPS23= 00000000
FPS24= 00000000, FPS25= 00000000, FPS26= 00000000, FPS27= 00000000
FPS28= 00000000, FPS29= 00000000, FPS30= 00000000, FPS31= FFFFFFFF
FPSCR= 00000000
J-Link>exec EnableEraseAllFlashBanks
J-Link>erase
Erasing device...
Comparing flash   [100%] Done.
Erasing flash     [--------------------------------------------------------------------------------------------------------025%]
J-Link: Flash download: Total time needed: 5.349s (Prepare: 0.062s, Compare: 0.000s, Erase: 5.278s, Program: 0.000s, Verify: 0.000s, Restore: 0.008s)
Erasing done.
J-Link>
```
Did not change the outcome, flashing fails. Verified that the flash output log is identical as before.

agansari, 2019-10-22:
[image]

Can you please check your jumpers?
I attached an image with my board.

agansari, 2019-10-22:
[image]

@jukkar can you check the jumpers on your board? I attached an image of mine.

jukkar, 2019-10-22:
The J27 (upper right corner) was set differently in my board, but changing it like you have, did not make any difference in flashing.

agansari, 2019-10-22:
@MaureenHelm do you have any ideea, and do we support daplink debug methods on this part?

jukkar, 2019-10-22:
Is it possible that the board is just broken if the flashing fails like this?

agansari, 2019-10-23:
That is a possibility and a probability.
CPU should not be bricked unless there is a bit of code that does something to lockup the CPU, or some fuse or configuration preventing it from being flashed.

I can only point you to NXP's support pages such as this:
https://community.nxp.com/thread/503292

jukkar, 2019-10-23:
@agansari thanks for the link, it contained useful instructions. I managed to flash the board by "make flash" and then same time resetting the board. It takes usually several times but eventually the board is flashed.
Pressing reset while flashing is a bit cumbersome way.

jukkar, 2019-10-23:
Just a reference log when the flashing succeeds
```
-- west flash: using runner jlink
-- runners.jlink: Flashing file: zephyr/samples/net/telnet/build/zephyr/zephyr.bin
SEGGER J-Link Commander V6.46g (Compiled Jun 14 2019 19:36:40)
DLL version V6.46g, compiled Jun 14 2019 19:36:30

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Firmware: J-Link OpenSDA 2 compiled May 27 2019 11:01:03
Hardware version: V1.00
S/N: 621000000
VTref=3.300V
Target connection not established yet but required for command.
Device "MCIMXRT1052" selected.

Connecting to target via SWD
Cannot connect to target.

Target connection not established yet but required for command.
Device "MCIMXRT1052" selected.

Connecting to target via SWD
Found SW-DP with ID 0x0BD11477
Scanning AP map to find all available APs
AP[1]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x04770041)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FD000
CPUID register: 0x411FC271. Implementer code: 0x41 (ARM)
Found Cortex-M7 r1p1, Little endian.
FPUnit: 8 code (BP) slots and 0 literal slots
CoreSight components:
ROMTbl[0] @ E00FD000
ROMTbl[0][0]: E00FE000, CID: B105100D, PID: 000BB4C8 ROM Table
ROMTbl[1] @ E00FE000
ROMTbl[1][0]: E00FF000, CID: B105100D, PID: 000BB4C7 ROM Table
ROMTbl[2] @ E00FF000
ROMTbl[2][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS-M7
ROMTbl[2][1]: E0001000, CID: B105E00D, PID: 000BB002 DWT
ROMTbl[2][2]: E0002000, CID: B105E00D, PID: 000BB00E FPB-M7
ROMTbl[2][3]: E0000000, CID: B105E00D, PID: 000BB001 ITM
ROMTbl[1][1]: E0041000, CID: B105900D, PID: 001BB975 ETM-M7
ROMTbl[1][2]: E0042000, CID: B105900D, PID: 004BB906 CTI
ROMTbl[0][1]: E0040000, CID: B105900D, PID: 000BB9A9 TPIU-M7
ROMTbl[0][2]: E0043000, CID: B105F00D, PID: 001BB101 TSG
Cache: Separate I- and D-cache.
I-Cache L1: 32 KB, 512 Sets, 32 Bytes/Line, 2-Way
D-Cache L1: 32 KB, 256 Sets, 32 Bytes/Line, 4-Way
Cortex-M7 identified.
Downloading file [zephyr/samples/net/telnet/build/zephyr/zephyr.bin]...
Comparing flash   [100%] Done.
Erasing flash     [100%] Done.
Programming flash [100%] Done.
Verifying flash   [100%] Done.
J-Link: Flash download: Bank 0 @ 0x60000000: 1 range affected (262144 bytes)
J-Link: Flash download: Total time needed: 9.138s (Prepare: 0.061s, Compare: 3.240s, Erase: 1.052s, Program: 1.454s, Verify: 3.219s, Restore: 0.110s)
O.K.

Script processing completed.

```

agansari, 2019-10-23:
@jukkar glad to hear you unbricked your board.

Solution: flash several times while holding reset.

jukkar, 2019-10-23:
I am not sure if this can be called unbricking :) but anyway the board is somewhat usable but in a very cumbersome way.

closed 2019-10-23 by jukkar