zephyrproject-rtos/zephyr #20016: STM32F4: cannot erase sectors from bank2

vaussard, 2019-10-22:
**Describe the bug**
I am using a STM32F479 (with 2MB of Flash, 1MB in each bank). Calling `flash_erase()` on the 1st sector of the 2nd bank returns -EIO (but the problem affects all sectors of the 2nd bank). This is caused by the Flash setting the FLASH_SR.WRPERR flag.

The STM32F4 flash driver assumes that the sector index can be computed as `regs->cr |= FLASH_CR_SER | (sector << 3);` for any sector. This is not correct, at least on STM32F479. The documentation for FLASH_CR.SNB says:

> These bits select the sector to erase.
> 0000: sector 0
> 0001: sector 1
> ...
> 01011: sector 11
> 01100: not allowed
> 01101: not allowed
> 01110: not allowed
> 01111: not allowed
> 10000: section 12
> 10001: section 13
> ...
> 11011 sector 23
> 11100: not allowed
> 11101: not allowed
> 11110: not allowed
> 11111: not allowed

Shifting the sector by 3 bits will result in a wrong SNB for sectors in the 2nd bank.

I workarounded this issue on STM32F479 with the following patch:

```
diff --git a/drivers/flash/flash_stm32f4x.c b/drivers/flash/flash_stm32f4x.c
index 7e37b229ab..0911593d16 100644
--- a/drivers/flash/flash_stm32f4x.c
+++ b/drivers/flash/flash_stm32f4x.c
@@ -70,6 +70,15 @@ static int erase_sector(struct device *dev, u32_t sector)
                return rc;
        }
 
+#if FLASH_SECTOR_TOTAL == 24
+       if (sector >= 12) {
+               /* SNB is 0b10000 for sector 12 */
+               sector += 4;
+       }
+#else
+#error "Please add support for this geometry"
+#endif
+
        regs->cr &= STM32F4X_SECTOR_MASK;
        regs->cr |= FLASH_CR_SER | (sector << 3);
        regs->cr |= FLASH_CR_STRT;
```

**To Reproduce**

Here is a minimal reproducer (assuming 2MB of Flash).

```
#include <flash.h>
#include <logging/sys_log.h>
#include <zephyr.h>

void main(void)
{
        struct device *stm32_dev;
        int err;

        stm32_dev = device_get_binding(FLASH_DEV_NAME);
        if (stm32_dev == NULL) {
                SYS_LOG_ERR("STM32 flash driver was not found!\n");
                return;
        }   

        err = flash_write_protection_set(stm32_dev, false);
        if (err) {
                SYS_LOG_ERR("Unable to disable Flash protection (%d)", err);
                return;
        }   

        err = flash_erase(stm32_dev, KB(1024), KB(16));
        if (err) {
                SYS_LOG_ERR("Unable to erase (%d)", err);
        }   

        err = flash_write_protection_set(stm32_dev, true);
        if (err) {
                SYS_LOG_ERR("Unable to enable Flash protection (%d)", err);
        }   
}
```

Without the workaround, this gives:

```
[2019-10-22 11:12:44] [general] [ERR] main: Unable to erase (-5)
[2019-10-22 11:12:44] [general] [ERR] main: Unable to enable Flash protection (-5)
```

**Expected behavior**
Erasing the correct sector without errors.

**Impact**
Cannot erase sectors in the 2nd bank (or worst, erase the wrong sector).

**Screenshots or console output**
See above.

**Environment (please complete the following information):**
 - Linux
 - Zephyr SDK v0.10.0
 - Applies to any Zephyr version from 1.13 up to latest upstream

vaussard, 2019-10-22:
FYI @erwango

erwango, 2019-10-22:
Thanks @vaussard.
@superna9999, fyi.

dleach02, 2019-10-22:
@erwango , I've assigned this to you. Please modify assignment where appropriate. We have also given it a medium priority.

PR #20113 (2019-11-05): flash: Fixes on stm32f4 driver
commit 5200669
```
Author: Erwan Gouriou <erwan.gouriou@linaro.org>
Date:   Thu Oct 24 15:10:38 2019 +0200

    drivers/flash: stm32f4: Return error on Dual Bank configuration
    
    Some stm32f4 1MB SoCs support optional Dual Bank configuration.
    This is not yet supported by stm32f4 driver, so report an
    error when configuration is detected
    
    Signed-off-by: Erwan Gouriou <erwan.gouriou@linaro.org>
```

commit 543b3cc
```
Author: Erwan Gouriou <erwan.gouriou@linaro.org>
Date:   Thu Oct 24 15:15:53 2019 +0200

    drivers/flash: stm32f4: Offset SNB  [... cut]
commit fb979e7c (2019-11-05): drivers/flash: stm32f4: Offset SNB from sector 12 on 2MB parts

On 2MB parts, on sector 12 and above SNB is offset by 4.

Fixes #20016
closed 2019-11-05 by MaureenHelm