zephyrproject-rtos/zephyr #20017: Convert GPIO users to new GPIO API

carlescufi, 2019-10-22:
List of users to convert:

```
boards/arm/nrf52840_pca10090
drivers/bluetooth/hci
drivers/can
drivers/ethernet
drivers/ethernet
drivers/ieee802154
drivers/ieee802154
drivers/ieee802154
drivers/led
drivers/sensor/adt7420
drivers/sensor/adxl362
drivers/sensor/adxl372
drivers/sensor/amg88xx
drivers/sensor/apds9960
drivers/sensor/apds9960
drivers/sensor/bma280
drivers/sensor/bmc150_magn
drivers/sensor/bmg160
drivers/sensor/bmi160
drivers/sensor/fxas21002
drivers/sensor/fxos8700
drivers/sensor/hmc5883l
drivers/sensor/hts221
drivers/sensor/iis3dhhc
drivers/sensor/isl29035
drivers/sensor/lis2dh
drivers/sensor/lis2ds12
drivers/sensor/lis2dw12
drivers/sensor/lis2mdl
drivers/sensor/lis3mdl
drivers/sensor/lps22hh
drivers/sensor/lsm6dsl
drivers/sensor/lsm6dso
drivers/sensor/lsm9ds0_gyro
drivers/sensor/mcp9808
drivers/sensor/mpu6050
drivers/sensor/sht3xd
drivers/sensor/stts751
drivers/sensor/sx9500
drivers/sensor/ti_hdc
drivers/sensor/tmp007
drivers/wifi/winc1500
include/drivers
samples/basic/button
samples/bluetooth/mesh
samples/bluetooth/mesh_demo
samples/boards/bbc_microbit/line_follower_robot
samples/boards/bbc_microbit/pong
samples/boards/bbc_microbit/sound
samples/boards/nrf52/mesh/onoff-app
samples/boards/nrf52/mesh/onoff_level_lighting_vnd_app
samples/boards/nrf52/power_mgr
samples/boards/reel_board/mesh_badge
samples/boards/up_squared/gpio_counter
samples/drivers/gpio
samples/sensor/fxos8700-hid
samples/subsys/usb/hid-cdc
samples/subsys/usb/hid-mouse
subsys/disk
tests/bluetooth/mesh
tests/boards/intel_s1000_crb
tests/drivers/gpio/gpio_basic_api
tests/drivers/gpio/gpio_basic_api
tests/drivers/gpio/gpio_basic_api
tests/drivers/pinmux/pinmux_basic_api
```

vikrant8051, 2019-10-23:
@carlescufi 
Is that mean I have to update said Apps (via submitting new PRs) with latest GPIO APIs ?

carlescufi, 2019-10-23:
> @carlescufi
> Is that mean I have to update said Apps (via submitting new PRs) with latest GPIO APIs ?

Yes, please. Send a Pull Request against the `topic-gpio` branch, since that is the branch we are targeting, not `master`.

vikrant8051, 2019-10-23:
@carlescufi 

https://github.com/zephyrproject-rtos/zephyr/pull/16648/commits/ff873b8a49b30cc7b20c91d7fee058d30d060d33 & we have to take reference from this link for modifications (for nrfx). right ?

carlescufi, 2019-10-23:
> @carlescufi
> 
> [ff873b8](https://github.com/zephyrproject-rtos/zephyr/commit/ff873b8a49b30cc7b20c91d7fee058d30d060d33) & we have to take reference from this link for modifications (for nrfx). right ?

That is modifying an actual GPIO driver, not a user of it. Take a look at the one of the PRs listed in the table in the main description of the issue to get a better feeling.

pabigot, 2019-10-23:
@vikrant8051 Specifically you should look at the commits with these subjects:
* 2c6af1f36d [Mon Jun 3 12:29:18 2019 +0200] samples: button: update to use new GPIO API
* 254e3324e1 [Tue Jan 29 00:33:10 2019 +0100] samples: blinky: update to use new GPIO API

since the only GPIO use in the samples you're responsible for are toggling LEDs and responding to button presses.  The only significant change in API will be in button handling.

(Note: The SHA1s above will change when the topic-branch is updated, so look for the commit message if you can't find them.)

pabigot, 2019-11-04:
Some notes on converting GPIO users.  If this is useful it may be edited for clarifications/extensions.

* If you capture devicetree `GPIOS_PIN` information in a configuration structure, you should now also capture devicetree `GPIOS_FLAGS` information so it can be passed to `gpio_pin_configure` along with any direction/pull/other flags.  Both devicetree values should fit in a `u8_t`.

* Don't make people search out the datasheet to figure out what the active level for a signal should be.  Document in the device bindings the active level of each signal as produced/consumed by the device, and add a note that the flags value in the devicetree node should reflect any inversion performed between the device and the GPIO line.

* Ensure that all devicetree nodes include `GPIO_ACTIVE_HIGH` or `GPIO_ACTIVE_LOW` as appropriate for the GPIO signal path.

* `sensors/drivers/hts221` (merged) provides a basic example of how to handle data-ready interrupts in the new infrastructure.  Several platforms don't support level triggers, so the best practice is to use edge triggers.  Some sensors produce level signals which require an operation (like reading the data) to clear the signal.

  Be aware that on board reset a level ready signal may already be asserted and if so no observation reports will be triggered.  Thus it's important to check the signal state and trigger a handle operation if already signalled.

* `sensors/drivers/sht3xd` (#20295) shows an example of configuring triggers.  It's mostly the same as for data-ready, but isn't enabled until the trigger is configured.  Be aware that just as with data-ready device reset may have left the signal asserted so handling it when enabling is required here too.

A working idiom to handle interrupts is exemplified in the [converted driver for st,hts221](https://github.com/zephyrproject-rtos/zephyr/blob/topic-gpio/drivers/sensor/hts221/hts221_trigger.c):
  * Provide a `setup_signal(dev, enable)` function that invokes `gpio_pin_interrupt_configure()` with `GPIO_INT_DISABLE` or (usually) `GPIO_INT_EDGE_TO_ACTIVE` based on the value of the passed `enable` flag.
  * Provide a `handle_signal(dev)` function that invokes `setup_signal()` to disable the interrupt, then processes the interrupt (signals a semaphore, queues a work item)
  * Use `gpio_pin_configure()` and `gpio_add_callback()` for the signal initializing the trigger infrastructure.
  * On trigger changes first disable the interrupt with `setup_signal(dev, false)`
  * When enabling the trigger use `setup_signal(dev, true)` then read the signal and if it's asserted invoke `handle_signal(dev)`.

jenmwms, 2019-12-18:
Can someone please explain the reasoning to remove tests/boards/intel_s1000_crb? I need to include rationale in the commit for #21471 . Any thoughts, @dcpleung, @carlescufi?

avisconti, 2020-01-17:
@carlescufi 
I am still owner of the following 3 items: lis2dh, lis2ds12, lsm9ds0_gyro.
I am not sure I'll be able to complete them in the timeframe, as I have other urgencies till end of
January. Is there anyone who can volunteer on those tasks?

avisconti, 2020-01-20:
> @carlescufi
> I am still owner of the following 3 items: lis2dh, lis2ds12, lsm9ds0_gyro.
> I am not sure I'll be able to complete them in the timeframe, as I have other urgencies till end of
> January. Is there anyone who can volunteer on those tasks?

OK, I  changed also lis2ds12 (see #22040).
I think I'll not be able to proceed with the remaining two (lis2dh and  lsm9ds0_gyro).

pabigot, 2020-01-20:
I think I'm tapped to handle anything that's left over, which I will do but I can't test them unless somebody can identify a shield or inexpensive supported dev board that has them.

avisconti, 2020-01-20:
> I think I'm tapped to handle anything that's left over, which I will do but I can't test them unless somebody can identify a shield or inexpensive supported dev board that has them.

Maybe we can just test the compilation for the moment? Let see what @erwango and @carlescufi  have to say.

pabigot, 2020-01-20:
Yes, we have agreement that build-test is sufficient so we can make progress, but we're also supposed to annotate devices that have not been verified on real hardware somehow.  AFAIK there is no documentation of how that's to be done.

carlescufi, 2020-01-20:
@avisconti yes, that is the agreement for this particular branch at least.
@pabigot This should come together with the discussion about how to document stability per-API. It can either be done in Doxygen or else in the upcoming maintainers file.

avisconti, 2020-01-22:
@carlescufi  @pabigot 
I removed my ownership for lis2dh and lsm9ds0_gyro.

carlescufi, 2020-01-22:
> @carlescufi @pabigot
> I removed my ownership for lis2dh and lsm9ds0_gyro.

OK, thanks. @pabigot will take those two.

pabigot, 2020-02-03:
All identified conversions have been complete.  Any more that are discovered will be addressed when topic-gpio merges to master, or in master.

closed 2020-02-03 by pabigot