zephyrproject-rtos/zephyr #20021: Add a module to Zephyr to include TF-M project and it's related repos

karl-zh, 2019-10-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

Current we want Zephyr to pass [PSA certification ](https://www.arm.com/why-arm/architecture/platform-security-architecture) for security. And TF-M is an implementation that we want to merge. Currently, I have a [PR19985](https://github.com/zephyrproject-rtos/zephyr/pull/19985) which build TFM in for secure execution and  Zephyr for nonsecure execution. But the current solution will be pull all repos after build start.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Add a module for TF-M and it's related in Zephyr, and link them in west.yml finally.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

The name for the repository could be: psa_tfm

carlescufi, 2019-10-22:
@karl-zh thanks. Can you add a link to the tf-m repo and its license?

karl-zh, 2019-10-22:
> @karl-zh thanks. Can you add a link to the tf-m repo and its license?

https://git.trustedfirmware.org/trusted-firmware-m.git/

BSD-3

karl-zh, 2019-11-22:
@carlescufi Hi, could you please give some help on this ticket to further?
Thanks

carlescufi, 2019-11-22:
@karl-zh https://github.com/zephyrproject-rtos/trusted-firmware-m

karl-zh, 2019-11-25:
Hi
@carlescufi 
TFM build needs include other projects, shall we add 3 more projects in zephyr as it's modules?
The projects details:
https://git.trustedfirmware.org/trusted-firmware-m.git/tree/docs/user_guides/tfm_build_instruction.rst#n90

carlescufi, 2019-11-25:
> Hi
> @carlescufi
> TFM build needs include other projects, shall we add 3 more projects in zephyr as it's modules?
> The projects details:
> https://git.trustedfirmware.org/trusted-firmware-m.git/tree/docs/user_guides/tfm_build_instruction.rst#n90

Well we already have mbedtls, is it possible to use the one we currently have? The two others are going to have to be dependencies, yes. That said this is getting into an area we haven't touched before, so can you please raise this as a TSC meeting topic for this week?

carlescufi, 2019-11-27:
As per the TSC meeting with @microbuilder and @galak, the single repository created at https://github.com/zephyrproject-rtos/trusted-firmware-m will hold all the required dependencies, including of course tf-m itself.

carlescufi, 2019-11-27:
Closing the issue since the repo is created and it will contain all dependencies.

closed 2019-11-27 by carlescufi