zephyrproject-rtos/zephyr #20026: sanitycheck corrupts stty in some cases

pabigot, 2019-10-22:
Manual runs of sanitycheck since 83fc06a8feb6074d1baaa0e93c56e4de76c578bb periodically complete with the following stty settings changed:

    -brkint -icrnl -ixon -icanon -echo

This is something like what I would expect if `stty sane` was not invoked at a necessary time.

I can reproduce this with some regularity with:

    sanitycheck -p mps2_an385 -T tests/kernel/

but it doesn't always leave the terminal corrupted.  There's some evidence it can also affect a subset of the flags.

nashif, 2019-11-05:
issue confirmed.

aescolar, 2019-11-06:
Note for whoever faces it, you can always write
``stty sane`` 
and press enter (even if you dont get the echo)
or in the worst case
Ctrl+J `stty sane`Ctrl+J

nashif, 2020-01-22:
a fix was added in 1fe1f3a23d31efa99ef2be65dd38a6be868e1a2a

closed 2020-01-22 by nashif