zephyrproject-rtos/zephyr #20030: stm32 can: zcan_frame from fifo uninitialized

maxvankessel, 2019-10-22:
**Describe the bug**
`struct zcan_frame` is uninitialized at entering the `can_stm32_rx_isr_handler`. 
It's filled by `can_stm32_get_msg_fifo`. If a standard frame is provided, the `ext_id` part of the reserved union of the `zcan_frame` might contain invalid data. This seems not to be a problem because the `id_type` is set to standard, which tells us it's a standard frame. 

But when the socketcan driver is enabled, it converts a received `zcan_frame` to a socketcan frame. Without masking the `ext_id`, which could lead to invalid data in the `can_frame`

**To Reproduce**
This problem has occurred to me on a stm32f091 board
And it can be reproduced by building the socketcan sample for a stm32f0 and sending standard frames to the board : `cangen can0 -I r -L i -D r`

**Expected behavior**
Or the `can_copy_zframe_to_frame` should mask the `id` or the stm32 can driver should set the content of the union to zero before writing the standard id:

```
if (mbox->RIR & CAN_RI0R_IDE) {
	msg->ext_id = mbox->RIR >> CAN_RI0R_EXID_Pos;
	msg->id_type = CAN_EXTENDED_IDENTIFIER;
} else {
+	msg->ext_id = 0;
	msg->std_id =  mbox->RIR >> CAN_RI0R_STID_Pos;
	msg->id_type = CAN_STANDARD_IDENTIFIER;
}
```

alexanderwachter, 2019-10-24:
@maxvankessel can you try this one?
```C
static inline void can_copy_zframe_to_frame(const struct zcan_frame *zframe,
					    struct can_frame *frame)
{
	frame->can_id = (zframe->id_type << 31) | (zframe->rtr << 30) |
		(zframe->id_type == CAN_STANDARD_IDENTIFIER zframe->std_id :
				    zframe->ext_id);
	frame->can_dlc = zframe->dlc;
	memcpy(frame->data, zframe->data, sizeof(frame->data));
}
```

alexanderwachter, 2019-10-24:
@maxvankessel I just made a PR for that. #20102

maxvankessel, 2019-10-24:
I will check this out this afternoon, but by the looks of it, this should fix it.

alexanderwachter, 2019-10-25:
Fixed by #20102.

closed 2019-10-25 by alexanderwachter