zephyrproject-rtos/zephyr #20033: Thread suspend only works if followed by k_sleep in thread that is performing the suspension

albertofloyd, 2019-10-22:
In our Zephyr app we have a task handler that suspends all threads used by the app.
If k_thread_suspend is not followed by k_sleep() threads do not get suspended.

Same behavior can be reproduced by modifying the following test code 
zephyr\tests\kernel\threads\thread_apis\src\test_threads_suspend_resume.c

	/* checkpoint: suspend current thread */
	k_thread_suspend(tid);
	/* k_sleep(K_MSEC(100)); */
	k_busy_wait(100*1000);

**To Reproduce**
Steps to reproduce the behavior:
0. cd zephyr\tests\kernel\threads\thread_apis
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
starting test - test_threads_suspend_resume_cooperative
    Assertion failed at ZEPHYR_BASE/tests/kernel/threads/thread_apis/src/test_threads_suspend_resume.c:41: threads_suspend_resume: last_prio == create_prio is false

**Expected behavior**
Thread suspension should occur without sleeping in the thread that performs the suspension

**Impact**
This adds latency to enter low power management.

albertofloyd, 2019-10-26:
@andrewboie after some additional debug we found that apart from test issue mentioned above there is an additional odd behavior with thread suspension.

Child thread function
while (1) 
       ... some activity
      k_sleep(1000);
       ... some other activity

Parent thread
     k_thread_supend(child)

In this this point the child thread is removed from ready queue and marked as suspend, but if time is right when calling k_thread_suspend() and thread did perform k_sleep(), after sleep period timeout is complete thread is re-added to ready queue. 

Is this expected?

albertofloyd, 2019-10-30:
@andyross @nashif could you comment on this ? 
Is there a recommended why to use k_sleep() inside a thread that is intended to be suspended?

andyross, 2019-10-30:
A suspended thread should never be scheduled, no.  In the first example, you're trying to suspend the **current** thread, though?  That's not really the intent of the API, it's an asynchronous thing intended to suspend other threads.  So what happens is that it returns synchronously, as the API itself is not a schedule point.  It could be made to be so without difficulty, though.

As to the k_sleep() timeout "unsuspending" a suspending thread, that sounds like a bug. The code looks OK to me at a quick glance.  Do you have some sample code for that?

andrewboie, 2019-10-31:
I think k_thread_suspend(k_current_get()) is pathological but we could easily make this a schedule point as Andy says.

however, if thread A calls k_sleep(), and while sleeping, another thread suspends it, the expiration of the timeout should not wake up the thread. @andyross do we have any tests that exercise this?

@albertofloyd which of these cases is the one concerning you, such that you marked this as high priority?

albertofloyd, 2019-11-01:
There are 2 problems we are facing when suspending a thread from another thread.
The first one is minor regarding blocking the thread that performs the suspension of child thread to be suspended as in the test case.
Second is the one mentioned by @andyross where k_sleep before suspend results in wake from the thread.  This is the reason why I marked this as high. 

I'm suspending the child threads in the main app thread only when there is an external event.
Main app thread is never suspended. See code snippet below, also some log from kernel where I do see when the thread is re-add to the ready queue.

#define EC_TASK_PRIORITY K_PRIO_COOP(5)
#define MAX_EC_TASKS 5

Main thread 
============
//  Create all task, each task is a thread
void main(void) 
{
create_task(postcode_task_init, postcode_thread, 125, true, "POST");
create_task(task2, task2_thread, 1000, false, "ETC");
create_task(task3, task3_thread, 200, false, "ETC2");
... 
... 
while(1)
       k_sleep(5000);
}

void suspend_btn_callback(void)
{
     suspend_all_tasks();
}

Task handler 
============ 
int create_task(task_init_t init, task_thread_t thread_func, u32_t period, bool can_suspend
		char *tag)
{
	task->init = init;
	task->thread_func = thread_func;
	task->period = period;
	sys_slist_prepend(&ec_tasks, &task->node);
	ret = task->init();
....

	k_thread_create(&task->thread, stack,
				EC_TASK_STACK_SIZE, thread_func,
				(void *)&task->period,
				NULL, NULL, EC_TASK_PRIORITY, K_INHERIT_PERMS,
				K_FOREVER);
		k_thread_name_set(&task->thread, tag);
....
}

void suspend_all_tasks(void)
{
	struct task_info *task, *tmp;

	SYS_SLIST_FOR_EACH_CONTAINER_SAFE(&ec_tasks, task, tmp, node) {
		k_thread_suspend(&task->thread);
	}
}

Child threads follow the follow this  pattern
==============
void postcode_thread(void *p1, void *p2, void *p3)
{
.... 
	while (true) {
		k_sleep(period);
                 /* Check for if need to perform any activity */
                printk("Nothing to do");
         }
}

[2019-10-25 17:45:13.876] P sleep 125
[2019-10-25 17:45:13.876] z_remove_thread_from_ready_q remove 0x00118014
[2019-10-25 17:45:13.876] z_remove_thread_from_ready_q remove 0x001180b4
[2019-10-25 17:45:13.876] z_remove_thread_from_ready_q remove 0x00118154
[2019-10-25 17:45:13.888] z_add_thread_to_ready_q add 0x001180b4 
[2019-10-25 17:45:13.888] z_remove_thread_from_ready_q remove 0x001180b4
[2019-10-25 17:45:13.888] z_add_thread_to_ready_q add 0x00118154 
[2019-10-25 17:45:13.888] z_remove_thread_from_ready_q remove 0x00118154
[2019-10-25 17:45:13.888] z_add_thread_to_ready_q add 0x001180b4 
[2019-10-25 17:45:13.903] z_remove_thread_from_ready_q remove 0x001180b4
[2019-10-25 17:45:13.903] z_add_thread_to_ready_q add 0x001181f4 
[2019-10-25 17:45:13.903] z_add_thread_to_ready_q add 0x00118154 
[2019-10-25 17:45:13.920] z_add_thread_to_ready_q add 0x001180b4 
[2019-10-25 17:45:13.920] Conditions met...
[2019-10-25 17:45:13.920] do_enter_idle : deep sleep 1
[2019-10-25 17:45:13.920] About to suspend 0x00118154
[2019-10-25 17:45:13.920] z_remove_thread_from_ready_q remove 0x001181f4
[2019-10-25 17:45:13.927] z_remove_thread_from_ready_q remove 0x00118154
[2019-10-25 17:45:13.927] z_remove_thread_from_ready_q remove 0x001180b4
[2019-10-25 17:45:13.927] z_add_thread_to_ready_q add 0x001180b4 
[2019-10-25 17:45:13.955] z_remove_thread_from_ready_q remove 0x001180b4
[2019-10-25 17:45:13.955] z_add_thread_to_ready_q add 0x00118154

andrewboie, 2019-11-05:
@albertofloyd thanks for the additional detail, we'll take a look.

andyross, 2019-11-05:
Heh, crossed with @andrewboie (this came up in a meeting as a P1 bug status).  Just took another more careful look.  In lieu of a test case I can run directly, can you try the diff below with your rig?

The first bit is for sure a bugfix for k_thread_suspend().  It shouldn't be leaving live timeouts if one happens to be active (thread timeouts are used for all of pending/sleeping/init-startup, so any of those would have likely had the same symptom.   In theory we could also check for schedulability in the thread timeout too, though that would be putting the overhead in arguably the wrong place.

The second chunk just implements the reschedule point mentioned above so you get the behavior you expect when you suspend the current thread.

```diff
--- a/kernel/thread.c
+++ b/kernel/thread.c
@@ -616,7 +616,13 @@ void z_thread_single_suspend(struct k_thread *thread)
                z_remove_thread_from_ready_q(thread);
        }
 
+       (void)z_abort_thread_timeout(thread);
+
        z_mark_thread_as_suspended(thread);
+
+       if (thread == _current) {
+               z_reschedule_unlocked();
+       }
 }
```

albertofloyd, 2019-11-08:
Thanks @andyross  with suggested change, suspending threads works.

PR #20947 (2019-11-26): kernel/thread: Cancel timeouts on k_thread_suspend(), make schedule p…
…oint

When suspending a thread, cancel any pending timeouts which might wake
it up unexpectedly.  Also, make suspending the current thread
(specifically) a schedule point, as callers are clearly going to
expect that to be synchronous.

Also fix a documentation weirdness.  The phrasing in the earlier docs
for k_thread_suspend() was confusing: it could be interpreted as
either document the current (essentially buggy) behavior that threads
will "wake up" due to preexisting timeouts, OR to mean that thread
timeouts will contin [... cut]
commit 50d0942f (2019-11-26): kernel/thread: Cancel timeouts on k_thread_suspend(), make schedule point

When suspending a thread, cancel any pending timeouts which might wake
it up unexpectedly.  Also, make suspending the current thread
(specifically) a schedule point, as callers are clearly going to
expect that to be synchronous.

Also fix a documentation weirdness.  The phrasing in the earlier docs
for k_thread_suspend() was confusing: it could be interpreted as
either document the current (essentially buggy) behavior that threads
will "wake up" due to preexisting timeouts, OR to mean that thread
timeouts will continue  [... cut]
closed 2019-11-26 by nashif