zephyrproject-rtos/zephyr #20068: Application doesn't start when SHELL-UART is enabled and UART is not connected on STM32F0

victorrsoliveira, 2019-10-23:
**Describe the bug**
I am doing a project with the MCU **STM32F091RC** and whenever I enable SHELL-UART through `CONFIG_SHELL=y` in the **prj.conf** file, the application stops working.

The application I am testing is below, just printing a `Hello World` and blinking an LED. Note that I do not create shell commands. I've just enabled it in the **prj.conf** file.

```
/*
 * Copyright (c) 2012-2014 Wind River Systems, Inc.
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <device.h>
#include <dfu/mcuboot.h>
#include <gpio.h>
#include <misc/printk.h>
#include <misc/reboot.h>
#include <zephyr.h>

#define LED_PORT DT_GPIO_LEDS_LED_GREEN_GPIO_CONTROLLER
#define LED DT_GPIO_LEDS_LED_GREEN_GPIO_PIN

void main(void)
{
    struct device *dev;
    u32_t led_value = 0;

    dev = device_get_binding(LED_PORT);
    /* Set LED pin as output */
    gpio_pin_configure(dev, LED, GPIO_DIR_OUT);

    while (1) {
        printk("Hello World! %s\n", CONFIG_BOARD);
        gpio_pin_write(dev, LED, led_value);
        led_value = !led_value;
        k_sleep(K_MSEC(200));
    }
}
```

The file **prj.conf** is like this:

```
CONFIG_PRINTK=y
CONFIG_SHELL=y
CONFIG_REBOOT=y
```

In the .**dts** file I have used this to choose UART console and shell:

``` 
chosen {
		zephyr,console = &usart2;
		zephyr,shell-uart = &usart2;
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
		zephyr,code-partition = &slot0_partition;
	};
```

I have uploaded the same code to **nRF52840_pca10056** board and everything works fine. Another board with **STM32F4** had the same result as the nRF board, application runs OK.

The only difference that breaks the normal execution of the application in the **STM32F0** is the change from `CONFIG_SHELL=y` to `CONFIG_SHELL=n`. It's like the application does not start and LED does not blink.

Does anyone have any idea on this?

jmbrederveld, 2019-11-12:
I came here to report the exact same issue, but (un)fortunately found yours instead.

The board I am testing is STM32F091CC, with Shell and Console on USART3, which I manually added to dts_fixup.h and included in my board .dts and Kconfig respectively.
Using GDB I find the board enters arch_system_halt(), reason 0.

Looking at the backtrace, I find that the function *sys_dlist_get(sys_dlist_t *list) in dlist.h tries to remove the node object (sys_dlist_remove), which in my case is NULL, and triggers the fault.

erwango, 2019-11-12:
@victorrsoliveira, @jmbrederveld , I've been able to run the proposed piece of code, with provided `prj.conf` on a `nucleo_f091rc`. I am still not getting the reported issue.

Is there any piece of information I am missing to reproduce this issue?

jmbrederveld, 2019-11-12:
Thank you for investigating. 
Can you please try the shell on an external UART that is not connected ? Any UART other than USART2 (which by default is connected to the ST-Link MCU).
Please disregard this if that is already what you tested.

erwango, 2019-11-22:
@jmbrederveld, thanks  for these indications, I've managed to reproduce the issue and found a fix.

Issue lies in `boards/arm/nucleo_f091rc/Kconfig.defconfig` where USART_1 (or X depending on your configuration) is not enabled if UART is enabled

The following fixed the issue on my side:
```
CONFIG_UART_X=y
```

While fixing the bug is easy, we'd need a way not to enter this kind of issue or detect it easily.
I'll think about it this week end.

PR #20971 (2020-05-25): drivers: Check errors on devices selected using dt macro
Some Kconfig defined devices may be defined using dt_chosen_label
function. Since there is no way to ensure a device enabled in dts
is also defined in Kconfig, it may happen that instance is not
actually defined.
In this case device_get_binding might return 0, leading to undefined
behavior in the function that calls it.
When not already done, systematically check return of function
device_get_binding on devices defined through dt_chosen_label macro.
Trigger ASSERT when required and return error when possible.

Fixes #20068
commit 2716cbca (2020-05-25): drivers: susbsys: Check errors on devices selected using dt macro

Some Kconfig defined devices may be defined using dt_chosen_label
function. Since there is no way to ensure a device enabled in dts
is also defined in Kconfig, it may happen that instance is not
actually defined.
In this case device_get_binding might return 0, leading to undefined
behavior in the function that calls it.
When not already done, systematically check return of function
device_get_binding on devices defined through dt_chosen_label macro.
Trigger ASSERT when required and return error when possible.

Fixes #20068
closed 2020-05-25 by carlescufi